adafruit/circuitpython #4209: adafruit_minimqtt library issues on circuitpython 6.2.0-beta.2

iakovos-t, 2021-02-16:
I am running circuitpython 6.2.0-beta.2 on a Feather S2 (ESP32S2). For the libraries adafruit-circuitpython-bundle-6.x-mpy-20210214

I have tried using the guide from adafruit under https://learn.adafruit.com/mqtt-in-circuitpython/circuitpython-wifi-usage as well as the examples included with the lib.

The code is returning an error:

```
Connecting to Adafruit IO...
Traceback (most recent call last):
  File "code.py", line 77, in <module>
  File "adafruit_minimqtt/adafruit_minimqtt.py", line 467, in connect
  File "adafruit_minimqtt/adafruit_minimqtt.py", line 263, in _get_socket
TypeError: can't convert str to int

Code done running.
```

The specific line causing the error is:

`mqtt_client.connect()`

The code in its entirety:

```
import time
import ssl
import socketpool
import wifi
import adafruit_minimqtt.adafruit_minimqtt as MQTT
 
# Add a secrets.py to your filesystem that has a dictionary called secrets with "ssid" and
# "password" keys with your WiFi credentials. DO NOT share that file or commit it into Git or other
# source control.
# pylint: disable=no-name-in-module,wrong-import-order
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise
 
# Set your Adafruit IO Username and Key in secrets.py
# (visit io.adafruit.com if you need to create an account,
# or if you need your Adafruit IO key.)
aio_username = secrets["aio_username"]
aio_key = secrets["aio_key"]
 
print("Connecting to %s" % secrets["ssid"])
wifi.radio.connect(secrets["ssid"], secrets["password"])
print("Connected to %s!" % secrets["ssid"])
### Feeds ###
 
# Setup a feed named 'photocell' for publishing to a feed
photocell_feed = secrets["aio_username"] + "/feeds/photocell"
 
# Setup a feed named 'onoff' for subscribing to changes
onoff_feed = secrets["aio_username"] + "/feeds/onoff"
 
### Code ###
 
# Define callback methods which are called when events occur
# pylint: disable=unused-argument, redefined-outer-name
def connected(client, userdata, flags, rc):
    # This function will be called when the client is connected
    # successfully to the broker.
    print("Connected to Adafruit IO! Listening for topic changes on %s" % onoff_feed)
    # Subscribe to all changes on the onoff_feed.
    client.subscribe(onoff_feed)
 
 
def disconnected(client, userdata, rc):
    # This method is called when the client is disconnected
    print("Disconnected from Adafruit IO!")
 
 
def message(client, topic, message):
    # This method is called when a topic the client is subscribed to
    # has a new message.
    print("New message on topic {0}: {1}".format(topic, message))
 
 
# Create a socket pool
pool = socketpool.SocketPool(wifi.radio)
 
# Set up a MiniMQTT Client
mqtt_client = MQTT.MQTT(
    broker=secrets["broker"],
    port=secrets["port"],
    username=secrets["aio_username"],
    password=secrets["aio_key"],
    socket_pool=pool,
    ssl_context=ssl.create_default_context(),
)
 
# Setup the callback methods above
mqtt_client.on_connect = connected
mqtt_client.on_disconnect = disconnected
mqtt_client.on_message = message
 
# Connect the client to the MQTT broker.
print("Connecting to Adafruit IO...")
mqtt_client.connect()
 
photocell_val = 0
while True:
    # Poll the message queue
    mqtt_client.loop()
 
    # Send a new message
    print("Sending photocell value: %d..." % photocell_val)
    mqtt_client.publish(photocell_feed, photocell_val)
    print("Sent!")
    photocell_val += 1
    time.sleep(5)
```

brentru, 2021-02-19:
@iakovos-t  I'm not able to replicate this error using the latest `master` MiniMQTT branch. Could you try downloading from https://github.com/adafruit/Adafruit_CircuitPython_MiniMQTT/releases directly?

Tested with `Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Adafruit MagTag with ESP32S2` and latest MiniMQTT.

iakovos-t, 2021-02-19:
@brentru Unfortunately the problem persists with the latest release downloaded directly from the link you provided.

Tested with Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; FeatherS2 with ESP32S2 and latest MiniMQTT

Is there anything I can do to help troubleshoot further - seems device specific?

brentru, 2021-02-19:
I'm not sure if it's device specific or not, just tried on my Metro ESP32-S2 and wasnt able to replicate either..

Could you attach a logger to MiniMQTT so I can see what's going on? The logging library is here: https://github.com/adafruit/Adafruit_CircuitPython_Logging

to do this - add these lines after the `mqtt_client` initialization in your code:
```
import adafruit_logging as logging
mqtt_client.enable_logger(logging,logging.DEBUG)
```

Then, please re-run your code and copy/paste the output here.

iakovos-t, 2021-02-19:
Connecting to Adafruit IO...
905.876: DEBUG - Attempting to establish MQTT connection...
Traceback (most recent call last):
  File "code.py", line 81, in <module>
  File "adafruit_minimqtt/adafruit_minimqtt.py", line 467, in connect
  File "adafruit_minimqtt/adafruit_minimqtt.py", line 263, in _get_socket
TypeError: can't convert str to int

Code done running.

Error is consistent with both the adafruit example (as run in this instance) and my own code.

iakovos-t, 2021-02-23:
@brentru following a long and unproductive troubleshooting session, a complete reinstall of circuitpython/libs on the device seems to have sorted the issue (same versions as previously). I am uncertain as to what caused it but I can confirm this is now working as expected. Please accept my apologies for time wasted.

closed 2021-02-23 by iakovos-t

brentru, 2021-02-24:
Thank you for updating me

devansau, 2021-06-05:
I had the same issue and found that I had mistakenly used ' 's around the port number in the secrets file.
Removed the quotes and it works fine now, featherS2 to Mosquitto broker up and running.

brentru, 2021-06-07:
That should not be occurring, MiniMQTT validates if the port is an integer prior to connecting.

https://github.com/adafruit/Adafruit_CircuitPython_MiniMQTT/pull/78