adafruit/circuitpython #4210: Raspberry Pi Pico PWM only works for 8 pins on 6.2.0-beta.2 absolute latest

kevinjwalters, 2021-02-16:
This only works on even pins, the rest are left low judging by LEDs on a Cytron Maker Pi Pico I've been using:

```
Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Raspberry Pi Pico with rp2040
>>> import pwmio
>>> import board
>>> pins = (board.GP0, board.GP1, board.GP2, board.GP3,
... board.GP4, board.GP5, board.GP6, board.GP7,
... board.GP8, board.GP9, board.GP10, board.GP11,
... board.GP12, board.GP13, board.GP14, board.GP15)
>>> pwms = [pwmio.PWMOut(p, frequency=2000, duty_cycle=3456) for p in pins]
```

If the odd pins are initialised first then only those work.

```
>>> pwms_o = [pwmio.PWMOut(pins[p_idx], frequency=2000, duty_cycle=3456) for p_idx in range(1, len(pins), 2)]
>>> pwms_e = [pwmio.PWMOut(pins[p_idx], frequency=2000, duty_cycle=3456) for p_idx in range(0, len(pins), 2)]
```

Based on some discussions it sounds like the hardware is capable of 16 especially if the frequency values are all the same.

Tested on `6.2.0-beta.2` and `6.2.0-beta.2-18-g2a467f137 on 2021-02-15`.

Originally mentioned in #4167 as I wasn't sure if this was a bug although having said that CP normally gives an instantiation error if it's run out of PWM things.

Curiously, #4208 shows that the consecutive pair GP18 and GP19 can both do PWM using `PWMAudioOut`.

tannewt, 2021-02-17:
No errors are thrown right? What does the non-working signal look like?

kevinjwalters, 2021-02-17:
I've not looked at signal, the relevant LEDs on the pins appear to be off so it's likely to just be held low.

kevinjwalters, 2021-02-17:
I slowed it down so I can see what's going on, `GP0` flashes as expected, `GP1` doesn't flash and appear low for this:

```
Adafruit CircuitPython 6.2.0-beta.2-18-g2a467f137 on 2021-02-15; Raspberry Pi Pico with rp2040
>>> import pwmio
>>> import board
>>> p0 = pwmio.PWMOut(board.GP0, frequency=2, duty_cycle=32768)
>>> p1 = pwmio.PWMOut(board.GP1, frequency=2, duty_cycle=32768)
>>> p0.duty_cycle
32768
>>> p1.duty_cycle
32768
```

kevinjwalters, 2021-02-22:
I don't think this is particularly significant but worth checking this works on any bugfix

```
Adafruit CircuitPython 6.2.0-beta.2-18-g2a467f137 on 2021-02-15; Raspberry Pi Pico with rp2040
>>> import pwmio
>>> import board
>>> pwm0 = pwmio.PWMOut(board.GP0, frequency=5000, variable_frequency=False, duty_cycle=0)
>>> pwm1 = pwmio.PWMOut(board.GP1, frequency=5000, variable_frequency=False, duty_cycle=0)
>>> pwm0.frequency  ### working output
5000
>>> pwm1.frequency  ### broken output
0
```

kevinjwalters, 2021-02-22:
I had a read of the code. This might be caused by a failure to initialise `self->top` for the second PWM in a channel's slice. The duty cycle code _depends_ on this

https://github.com/adafruit/circuitpython/blob/0ec99b37e0cb906d86590687ed06013516823cbe/ports/raspberrypi/common-hal/pwmio/PWMOut.c#L177-L190
 
but it's only set in `common_hal_pwmio_pwmout_set_frequency` and `pwmio_pwmout_set_top`. The latter is not called in PWMOut.c and the former is only called _once per slice_. This would also explain the frequency attribute being 0 as `actual_frequency` is set in those two functions.

BTW, why does `common_hal_pwmio_pwmout_set_frequency` call `common_hal_pwmio_pwmout_set_duty_cycle`? If that's necessary a comment justifying it would be useful. 

https://github.com/adafruit/circuitpython/blob/0ec99b37e0cb906d86590687ed06013516823cbe/ports/raspberrypi/common-hal/pwmio/PWMOut.c#L237

It also may make https://github.com/adafruit/circuitpython/blob/0ec99b37e0cb906d86590687ed06013516823cbe/ports/raspberrypi/common-hal/pwmio/PWMOut.c#L142 if ordering is not important.

kevinjwalters, 2021-02-22:
On reflection, there's a bit of an issue here with having independent values of `top` per `PWMOut` when it's a per slice value...

tannewt, 2021-02-23:
> This might be caused by a failure to initialise `self->top` for the second PWM in a channel's slice

This definitely seems like a bug! Mind making a PR for it?

> BTW, why does `common_hal_pwmio_pwmout_set_frequency` call `common_hal_pwmio_pwmout_set_duty_cycle`?

duty_cycle is called again because the duty_cycle is scaled by the top value (which set_frequency changes.)

> On reflection, there's a bit of an issue here with having independent values of `top` per `PWMOut` when it's a per slice value...

`top` shouldn't change if two PWMOuts are on the same slice because they are fixed frequency. (The internal `set_top` violates this for AudioPWMOut but only on its internal PWMOuts.)

kevinjwalters, 2021-02-23:
There's a single execution of `pwmio_pwmout_set_top` lurking in the mono/stereo PWM audio code:

https://github.com/adafruit/circuitpython/blob/191b143e7ba3b9f5786996cbffb42e2460f1c14d/ports/raspberrypi/common-hal/audiopwmio/PWMAudioOut.c#L150

closed 2021-02-26 by tannewt

kevinjwalters, 2021-02-26:
I tested `adafruit-circuitpython-raspberry_pi_pico-en_US-20210226-2e6e91d.uf2` and this looks good wrt this `PWMOut` issue.

That version doesn't present `CIRCUITPY` though. I went back to `adafruit-circuitpython-raspberry_pi_pico-en_US-20210215-2a467f1.uf2` and `CIRCUITPY` reappeared. I'm sure it's not related to this and I've just mentioned it in discord in case it's not an intentional interim state.

kevinjwalters, 2021-02-28:
Visual test results here: https://www.youtube.com/watch?v=7nukcyJWZzE

PR #4253 (2021-02-26): Fix second shared PWM
Fixes #4210
commit 1b00d94b (2021-02-26): Fix second shared PWM

Fixes #4210