adafruit/circuitpython #4211: Why are the ESP32S2 JTAG pins not reset on the OR of these conditions? #if defined(DEBUG) || defined(ENABLE_JTAG)

skieast, 2021-02-16:
I would have thought ENABLE_JTAG would be sufficient. When i compile with DEBUG i lose the I2C pins on the board I am using.
I may be completely misunderstanding the reasoning.
Thanks

anecdata, 2021-02-16:
I (also) hope there's no lurking reason not to remove DEBUG from that logic https://github.com/adafruit/circuitpython/pull/4131#issuecomment-778834022

tannewt, 2021-02-17:
We usually assume that you want SWD or JTAG access. We usually will reset pins back to their debug state but allow you to use them if you want.

skieast, 2021-02-17:
So if I compile with DEBUG=1 and ENABLE_JTAG not defined I should be able to use the pins in question for something other than SWD/JTAG? If so this doesnt happen as I cannot define an I2C bus on two of the pins.  Pin in use error I believe. 

thanks

tannewt, 2021-02-18:
I was saying how it usually works. It sounds like the ESP doesn't manage pin sharing for these like other ports do. So, it's something to fix.

tannewt, 2021-02-18:
(If you are compiling yourself then I'd just hack it.) Most folks won't hit this.

skieast, 2021-02-18:
What I did. May look at the problem later. Thanks.

dhalbert, 2026-11-13:
Claude wrote this:

This was fixed in 9.0.0. Until then, `ENABLE_JTAG` defaulted to `DEBUG`, so a `DEBUG=1` build reserved the JTAG pins. #9008 changed the default to off. Now only `make ... ENABLE_JTAG=1` reserves them.

Claude tested `main` (`ab606c9484`, 11.0.0-alpha.3-8) by trying `digitalio.DigitalInOut()` on each JTAG pin after boot, after a reload, and after a USB power cycle:

| Board | JTAG pins | `DEBUG=1` | `ENABLE_JTAG=1` |
|---|---|---|---|
| Metro ESP32-S3 | GPIO39–42 | free | in use |
| Metro ESP32-P4 | GPIO2–5 | free | in use |

The results were the same at all three points.

closed 2026-11-13 by dhalbert