adafruit/circuitpython #4212: Provide a way to set os.environ outside of code.py

tannewt, 2021-02-17:
Once we have a way, we can replace `secrets.py` use with it. `os.environ` is the common way for sharing secrets in CPython. For example, GitHub Actions secrets are provided as environment variables.

tannewt, 2022-03-23:
It'll be simpler to support this through `os.getenv()`: https://docs.python.org/3/library/os.html#os.getenv Having it be a function will give us more flexibility to run code on access.

Also note the `dotenv` Python library looks like a really good model. https://github.com/theskumar/python-dotenv It is used to manage environment variables for projects in regular Python with `.env` files.

askpatrickw, 2022-03-24:
I had some notes on this, looked at other approaches, etc... might be helpful to look at before coding starts.

https://gist.github.com/askpatrickw/610a83c489c6588055f6d1d3fd933fa7

gitcnd, 2024-05-09:
For anyone else arriving here, this appears to be replaced by settings.toml now - see https://learn.adafruit.com/scrolling-countdown-timer/create-your-settings-toml-file  and  https://docs.circuitpython.org/en/latest/docs/environment.html

PR #6407 (2022-05-26): Add dotenv read support
os.getenv() will use it (when available) to load variables from
/.env

This will also be useful when we need secrets or config for
CircuitPython outside of the VM (like WiFi credentials.)

Fixes #4212