adafruit/circuitpython #4214: how to install Debouncer module?

InvisibleRasta, 2021-02-18:
Hello guys i am working on a macro keyboard project on my pico and I cant figur eout how to get the Decouncer module.
i did download adafruit-circuitpython-debouncer-6.x-mpy-1.3.9.zip and palced the mpy file in the lib/adafruit_deboucner folder but i cant load the module.
Here is my script:
`# Imports
import time
import board
import usb_hid
import digitalio
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
from adafruit_debouncer import Debouncer

# Define Pins
btn1_Pin = board.GP28
btn2_Pin = board.GP27
btn3_Pin = board.GP26
btn4_Pin = board.GP22
btn5_Pin = board.GP21
btn6_Pin = board.GP20
btn7_Pin = board.GP19
btn8_Pin = board.GP18
btn9_Pin = board.GP17
btn10_Pin = board.GP16

# Define HID Key Outputs
kbd = Keyboard(usb_hid.devices)

# Onboard LED
led = digitalio.DigitalInOut(board.GP25)
led.direction = digitalio.Direction.OUTPUT

# Setup all Buttons as Inputs with PullDown
btn1 = digitalio.DigitalInOut(btn1_Pin)
btn1.direction = digitalio.Direction.INPUT
btn1.pull = digitalio.Pull.DOWN

btn2 = digitalio.DigitalInOut(btn2_Pin)
btn2.direction = digitalio.Direction.INPUT
btn2.pull = digitalio.Pull.DOWN

btn3 = digitalio.DigitalInOut(btn3_Pin)
btn3.direction = digitalio.Direction.INPUT
btn3.pull = digitalio.Pull.DOWN

btn4 = digitalio.DigitalInOut(btn4_Pin)
btn4.direction = digitalio.Direction.INPUT
btn4.pull = digitalio.Pull.DOWN

btn5 = digitalio.DigitalInOut(btn5_Pin)
btn5.direction = digitalio.Direction.INPUT
btn5.pull = digitalio.Pull.DOWN

btn6 = digitalio.DigitalInOut(btn6_Pin)
btn6.direction = digitalio.Direction.INPUT
btn6.pull = digitalio.Pull.DOWN

btn7 = digitalio.DigitalInOut(btn7_Pin)
btn7.direction = digitalio.Direction.INPUT
btn7.pull = digitalio.Pull.DOWN

btn8 = digitalio.DigitalInOut(btn8_Pin)
btn8.direction = digitalio.Direction.INPUT
btn8.pull = digitalio.Pull.DOWN

btn9 = digitalio.DigitalInOut(btn9_Pin)
btn9.direction = digitalio.Direction.INPUT
btn9.pull = digitalio.Pull.DOWN

btn10 = digitalio.DigitalInOut(btn10_Pin)
btn10.direction = digitalio.Direction.INPUT
btn10.pull = digitalio.Pull.DOWN

led_blinked = time.monotonic_ns()
bounce1 = Debouncer(btn1, interval=0.2)
bounce2 = Debouncer(btn2, interval=0.2)
bounce3 = Debouncer(btn3, interval=0.2)
bounce4 = Debouncer(btn4, interval=0.2)
bounce5 = Debouncer(btn5, interval=0.2)
bounce6 = Debouncer(btn6, interval=0.2)
bounce7 = Debouncer(btn7, interval=0.2)
bounce8 = Debouncer(btn8, interval=0.2)
bounce9 = Debouncer(btn9, interval=0.2)
bounce10 = Debouncer(btn10, interval=0.2)

print('%s Starting v0.5' % time.monotonic_ns())

while True:
    bounce1.update()
    if bounce1.rose:
        print('%s Button 1 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce1.fell:
        print('%s Button 1 released' % time.monotonic_ns())
    bounce2.update()
    if bounce2.rose:
        print('%s Button 2 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce1.fell:
        print('%s Button 2 released' % time.monotonic_ns())
    bounce3.update()
    if bounce3.rose:
        print('%s Button 3 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce3.fell:
        print('%s Button 3 released' % time.monotonic_ns())
    bounce4.update()
    if bounce4.rose:
        print('%s Button 4 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce4.fell:
        print('%s Button 4 released' % time.monotonic_ns())
    bounce5.update()
    if bounce5.rose:
        print('%s Button 5 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce5.fell:
        print('%s Button 5 released' % time.monotonic_ns())
    bounce6.update()
    if bounce6.rose:
        print('%s Button 6 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce6.fell:
        print('%s Button 6 released' % time.monotonic_ns())
    bounce7.update()
    if bounce7.rose:
        print('%s Button 7 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce7.fell:
        print('%s Button 7 released' % time.monotonic_ns())
    bounce8.update()
    if bounce8.rose:
        print('%s Button 8 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce8.fell:
        print('%s Button 8 released' % time.monotonic_ns()) 
    bounce9.update()
    if bounce9.rose:
        print('%s Button 9 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce1.fell:
        print('%s Button 9 released' % time.monotonic_ns())           
    bounce10.update()
    if bounce10.rose:
        print('%s Button 10 pressed' % time.monotonic_ns())
        kbd.send(Keycode.WINDOWS, Keycode.TAB)
    if bounce10.fell:
        print('%s Button 10 released' % time.monotonic_ns())    
# Blink onboard LED every half second
    if time.monotonic_ns() - led_blinked > 500000000:
        #print('%s Blinking led' % time.monotonic_ns())
        led_blinked = time.monotonic_ns()
        led.value = not led.value

    time.sleep(0.001)

`

tannewt, 2021-02-18:
Try placing the mpy in `lib` directly. You only sometimes need the folder.

tannewt, 2021-02-18:
(Specifically you need a folder when a library has multiple mpy files in it. The bundle will have the folder too if you need it.)

dhalbert, 2021-02-20:
Coding note: you might want to consider storing the pins, buttons, and debouncers in lists or tuples, so you can reference them in a loop and avoid repeated code.

Closing for now. Please reopen if you are still having trouble.

closed 2021-02-20 by dhalbert