adafruit/circuitpython #4216: Serial Communication via USB only

arm358, 2021-02-18:
I'm trying to connect the raspberry pi PICO to the Raspberry Pi 4 and send data via Serial from the PICO to the Pi4.

Every example I see for Circuitpython uses either I2C or UART (i.e. using the pins on the board for Serial communication).

Is serial communication solely via USB possible? I want to only connect the Pico to the Pi4 with the USB cable and no other wires.

microDev1, 2021-02-18:
Yup! it is possible... take a look at #4215.

arm358, 2021-02-18:
> 
> 
> Yup! it is possible... take a look at #4215.

Awesome, good to hear. Unfortunately looks like it's pretty new? I'm a pretty basic user so much easier for me to follow a guide and adapt it to what I want to do...

Maybe there will eventually be some guides I can utilize! Thanks for the info.

microDev1, 2021-02-18:
When the PR is merged the documentation will be available [here](https://circuitpython.readthedocs.io/en/latest/README.html).

If you want to get a head start at this... you can grab a `uf2` from the [artifacts](https://github.com/adafruit/circuitpython/actions/runs/577067290) page.
The functionality is similar to `busio.UART` so most of the existing examples will work with some modification like :-

```python
import time
import usb_cdc
# import busio

uart = usb_cdc.serials[1]
# uart = busio.UART(board.GP0, board.GP1, baudrate=115200)

s = bytearray("hello\n\r".encode())

while True:
    uart.write(s)
    time.sleep(1)
```

dhalbert, 2021-02-18:
> > Yup! it is possible... take a look at #4215.
> 
> Awesome, good to hear. Unfortunately looks like it's pretty new? I'm a pretty basic user so much easier for me to follow a guide and adapt it to what I want to do...
> 
> Maybe there will eventually be some guides I can utilize! Thanks for the info.

Yes, it is pretty new (last night :slightly_smiling_face: ). I will be writing a guide soon.

closed 2021-02-20 by tannewt