adafruit/circuitpython #4218: neopixel: calling show() after deinit() doesn't raise an exception

jepler, 2021-02-18:
I was surprised at this:
```python
>>> n = neopixel.NeoPixel(board.GP0, 1, auto_write=False)
>>> n.deinit()
>>> n.show()
>>> 
```

Expected behavior: `n.show()` raises an exception since the related object has been deinitialized
Actual behavior: `n.show()` does not raise an exception

I only tested on Pico, so this may be a port-specific problem.  However, I didn't label it as such initially.

Beau28713, 2021-09-13:
```python
>>> import board
>>> import neopixel
>>> pixel = neopixel.NeoPixel(board.NEOPIXEL, 10, auto_write=False)
>>> pixel.deinit()
>>> pixel.show()
>>> 
```

Running the above example on circuit playground express crashed the core code and rebooted the hardware into safe mode. 

[image]

tannewt, 2021-09-13:
I think the fix is to check if the digitalio is deinited here: https://github.com/adafruit/circuitpython/blob/main/shared-bindings/neopixel_write/__init__.c#L63

Beau28713, 2021-09-14:
I'll take a look at it. Thanks

PR #5946 (2022-01-29): Fixes neopixel show() after deinit() not raising an exception
Previously, the calling neopixel.show() after calling neopixel.deinit() would not raise an exception. Now it does.

Fixes #4218, but I'm not sure how to link them.
closed 2022-01-29 by dhalbert