adafruit/circuitpython #4221: RP2040: bitbangio.I2C not working properly

dhalbert, 2021-02-18:
While trying to use `bitbangio.I2C` to fix #4204 and potentially #4082, I tried using `bitbangio.I2C` for short I2C writes. However, it wasn't working, and further trials showed `bitbangio.I2C` was just not working well.

Examples:
```python
>>> import adafruit_tcs34725, board, busio, bitbangio, adafruit_lis3dh, sht31d
>>> i2c = bitbangio.I2C(board.GP1, board.GP0)
>>> l = adafruit_lis3dh.LIS3DH_I2C(i2c)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lis3dh.py", line 334, in __init__
  File "adafruit_lis3dh.py", line 94, in __init__
RuntimeError: Failed to find LIS3DH!
>>> s = adafruit_sht31d.SHT31D(i2c)
>>> s.temperature
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_sht31d.py", line 336, in temperature
  File "adafruit_sht31d.py", line 232, in _read
  File "adafruit_sht31d.py", line 204, in _data
  File "adafruit_sht31d.py", line 168, in _command
  File "adafruit_sht31d.py", line 168, in _command
OSError: [Errno 19] Unsupported operation
>>> 
```

(this particular error occurred only once):
```python
>>> import adafruit_tcs34725, board, busio, bitbangio, adafruit_sht31d
>>> i2c = bitbangio.I2C(board.GP1, board.GP0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TimeoutError: Clock stretch too long
```

dhalbert, 2021-02-22:
I moved this to 6.2.0 because it's a blocker for #4082. The problems seem more widespread than that single device, though they are not pervasive.

dhalbert, 2021-02-24:
Bitbangio I2C seems to go off the rails when it's trying to do a read. The writes look OK. Here are some Saleae screenshots. The clock pulses after the "Setup Read" are extra-long (9 us), and then something starts clocking SDA every clock cycle. I'm not sure if it's the RP2040 or the device.

[image]

By comparison, here is the same read when done from a Metro M4:

[image]

@ladyada @tannewt @jepler if you recognize this kind of I2C behavior let me know. Since the same bitbangio I2C code is shared among all ports, I think there is something else different, perhaps in the `digitalio` pin management. It could be in the delay timing too, but that looks similar, though not identical.

ladyada, 2021-02-24:
it could be something to do with how we switch from digital output (to pull down the pin) to floating (to allow the resistor to pull up) to input (to read whether its pulled) thats the 'weirdest' part of i2c because we mimic an open drain bidirectional

tannewt, 2021-02-25:
I don't recognize it but I suspect @ladyada is right. The open drain behavior of DigitalInOut may be wrong. I'd be surprised if the bitbangio code was wrong since its old and shared amongst ports.

dhalbert, 2021-02-25:
I did fix some things in `DigitalInOut` which fixed `bitbangio`. I am still trying to get `bitbangio.I2C` to be used internally in `busio.I2C` for short writes without glitches. I think I will submit the first fix as a separate PR so it doesn't get held up by the second.

closed 2021-03-01 by tannewt

sgbaird, 2022-10-06:
FYI https://github.com/ArduCAM/PICO_SPI_CAM/issues/8#issuecomment-1268603755 also produces a `"Clock stretch too long"` error (this is for Pico W).