adafruit/circuitpython #4222: frequency of rp2pio.StateMachine looks broken on 6.2.0-beta.2-18-g2a467f137

kevinjwalters, 2021-02-18:
I've been experimenting with code derived from the demo code on https://github.com/adafruit/Adafruit_CircuitPython_PIOASM - what duty_cycle and frequency square wave is that supposed to create?

I've found that a frequency of `16` is faster than `1` but `64` is slower than `1`. If I print `sm.frequency` it does show correct number. Is 1 supposed to execute at 1Hz? The units aren't documented at the moment in https://github.com/adafruit/circuitpython/blob/ae262cde55669c354fa2422097581b4ac4afaf88/ports/raspberrypi/bindings/rp2pio/StateMachine.c#L85

Can the hardware support below 1Hz?

kevinjwalters, 2021-02-18:
I've only had a quick look but I think the PIO PC wrapping functionality introduces no delay so the example code

```
squarewave = """
.program squarewave
    set pins 1 [1]  ; Drive pin high and then delay for one cycle
    set pins 0      ; Drive pin low
"""
```

would produce a 66.7% duty cycle rather than 50% which is probably what was intended. This was probably based on code that had an explicit `jmp`, cf https://github.com/raspberrypi/pico-examples/blob/master/pio/squarewave/squarewave.pio#L7-L12 vs https://github.com/raspberrypi/pico-examples/blob/master/pio/squarewave/squarewave_wrap.pio#L11-L19

tannewt, 2021-02-18:
Correct, that will be 2/3 duty cycle and the frequency will be the state machine frequency / 3 because the program is 3 cycles long.

The slowest it can run depends on the PIO divisor which is 16 bit int + 8 bit fractional. So whatever 125 mhz / (2 ** 16 + 255 / 256) is.

tannewt, 2021-02-18:
Oh, yes. I think our API for frequency is integer-only now so the slowest will be 1hz.

kevinjwalters, 2021-02-18:
It's broken though, 64Hz shouldn't be slower blinking (i'm using LED) than 1Hz and 1Hz does not look like 1Hz.

tannewt, 2021-02-20:
Ah, ya. That 80 is confusing. I must not have checked the frequency. I think 1907.35 is the minimum.

The TODO item here is to raise an exception when given a frequency below what is possible.

tannewt, 2021-02-20:
I'll update and verify the example too.

commit abbbb91f (2021-02-25): Add state machine divisor check

This causes an exception when setting a state machine too slow or
too fast.

Fixes #4222
closed 2021-02-25 by jepler