adafruit/circuitpython #4226: rp2040: implement rotaryio

dhalbert, 2021-02-19:
I may look at this in conjunction with #3895. If I can get a good reference implementation going, then it will be easy to port.

tannewt, 2021-02-20:
This is similar to the approach I was thinking: https://github.com/micropython/micropython/pull/6894/files

Use PIO to debounce and deduplicate pin values. Push the value out and trigger an interrupt on it.

I want to do this soon so moving to 6.2.0 milestone. Its a natural fit with usb_hid projects.

jepler, 2021-02-20:
The micropython state decoder looks like a solid design for "x4" quadrature decoding that avoids some problems that more naive implementations have.  Crucially, 
 * the data checked by the decoder is exactly the seen edges, not a sample of the inputs at a later time
 * every state transition is handled, rather than just handling certain transitions

Tracking the previous direction (positive or negative) in an attempt to decode changes in both bits simultaneously I think is of dubious value but it doesn't seem to hurt (doesn't affect the behavior when only one bit transition at a time is ever seen) and could make some designs that run just at the margin of the maximum work better, as long as there's enough "physicality" to the quadrature inputs that they don't instantly reverse from "+ludicrous_speed" to "-ludicrous_speed" instantly.

wildestpixel, 2021-02-26:
Found this quite quickly having wired up to test 
[image]

tannewt, 2021-02-26:
Sorry! I'm hoping to get to this very soon! Hopefully today or Monday.

jepler, 2021-03-04:
@tannewt shared with me some initial code related to rotaryio, but it's incomplete: https://github.com/tannewt/circuitpython/tree/rp2040_rotaryio

dhalbert, 2021-03-12:
Fixed by #4329.

closed 2021-03-12 by dhalbert

wildestpixel, 2021-03-16:
Can confirm this now working in latest nightly builds (cannot build myself due to M1 silicon config error). Might be that assignation of encoder pins A & B is flipped. I have : 

`encoder = rotaryio.IncrementalEncoder(board.GP3, board.GP4)`

working in correct orientation (clockwise increment, counter-clockwise decrement), however,

GP3 is connected to B and GP is connected to A.

Switch the wires, the encoder goes the wrong way, or code as 

`encoder = rotaryio.IncrementalEncoder(board.GP4, board.GP3)`

to receive error : 

`Traceback (most recent call last):
  File "code.py", line 12, in <module>
RuntimeError: Pins must be sequential`

tannewt, 2021-03-16:
@jepler want to support descending pins too?