adafruit/circuitpython #4235: RasPi Pico: Often (but not always) Errno 19 on i2c writes

ajs256, 2021-02-21:
I'm sometimes getting "OSError: [Errno 19] Unsupported operation" when attempting to write to an i2c device (a SparkFun Serial LCD) using a RasPi Pico. 

code.py:
```py
import board
import busio

i2c = busio.I2C(board.GP17, board.GP16)

while not i2c.try_lock():
    print("i2c not locked, retrying...")

print(i2c.scan())
i2c.writeto(114, "|-") # Clear
i2c.writeto(114, "Hello world!") # Show some text. Error happens here.
```
The error appears to be happening on line 11, according to the output. Am I trying to send an illegal character? Or is it a bug in CircuitPython?

dhalbert, 2021-02-21:
There are issues with some I2C devices. The I2C peripherals on the RP2040 do not allow zero-length writes, which are the typical way to do scanning. We are working on a workaround for that.

If you remove the `i2c.scan()`, does the problem go away?

ajs256, 2021-02-21:
Hmm. I commented out the scan call, and I now get the same error every time! I don't think the `scan()` is what's causing it.

dhalbert, 2021-02-21:
Could you try some other I2C device? What is the part number of the serial LCD you are using?

ajs256, 2021-02-21:
The LCD is SparkFun ID [LCD-14073](https://sfe.io/p14073). Give me a bit to scrounge up another device.

ajs256, 2021-02-21:
I don't get the error at all using a 8x8 bicolor LED matrix backpack (Adafruit PID 902). I am using the `adafruit_ht16k33` library and not writing raw bytes though.

ajs256, 2021-02-23:
Just tested it using the absolute newest build as of https://github.com/adafruit/circuitpython/commit/b19e7c914844634306745ff6858f52df308f85ff, and the error still happens.

ajs256, 2021-03-03:
Tested again as of https://github.com/adafruit/circuitpython/commit/15615effc2f8a9f991ace105f8a9f7bb51982059, and it still happens. 

Sorry, but because I do not have an SWD debugger or a logic analyzer, I cannot provide any more info or debugging help.

ajs256, 2021-03-03:
Tested using `bus_device`, and it appears to hang when attempting to init the device, unless I pass in `probe=False`. Then it gives me the `OSError` I was getting before. Here's the code I'm using:

```py
import board
import busio
from adafruit_bus_device import i2c_device

i2c = busio.I2C(board.GP17, board.GP16)

while not i2c.try_lock():
    pass # Interestingly, it doesn't work without locking the bus. bus_device examples don't seem to require this. 
print("i2c locked.")

lcd = i2c_device.I2CDevice(i2c, 114, False)
print("Device initted.")

lcd.write("|-") # Clear display
lcd.write("Hello, world!") # Print text
print("Text shown.")
```

dhalbert, 2021-03-03:
Try using `bitbangio.I2C()` on absolute latest. I am still working on getting `busio.I2C()` to work.

ajs256, 2021-03-03:
`bitbangio.I2C` works perfectly, thank you.

ajs256, 2021-03-03:
Well... not exactly perfectly. I have to pass in `probe=False`, or else it just infinitely hangs.

dhalbert, 2021-03-04:
Moving to a 6.x.x milestone since there's now a workaround.

dhalbert, 2021-03-31:
This is now fixed by #4499 and #4508.

closed 2021-03-31 by dhalbert

ajs256, 2021-04-16:
This doesn't appear to be fixed in
```py
Adafruit CircuitPython 6.2.0 on 2021-04-05; Raspberry Pi Pico with rp2040
```
I still need to use `bitbangio`, otherwise I occasionally get `OSError: [Errno 19] Unsupported operation` when writing a string to an `I2CDevice`.

dhalbert, 2021-04-16:
Which device(s) in particular? I tested a bunch.

ajs256, 2021-04-16:
The SparkFun [SerLCD](https://www.sparkfun.com/products/16397), which is ATmega328-based, and the firmware is open source [here](https://github.com/sparkfun/OpenLCD/).

dhalbert, 2021-04-16:
Could you open a specific issue for that device? It must depend on how I2C peripheral is implemented there. There are further minor fixes to I2C that are in the works upstream, but I'm not sure they would fix this. Does it work some of the time with the native I2C?

ajs256, 2021-04-16:
> Could you open a specific issue for that device?
Give me a bit to get a minimal example working.

> Does it work some of the time with the native I2C?
Sometimes with `busio.I2C` and all the time with `bitbangio.I2C`.

ajs256, 2021-04-17:
Just opened #4621.

richteel, 2021-10-30:
I have a Keyboard FeatherWing - QWERTY Keyboard + 2.6" LCD (Adafruit PRODUCT ID: 4818) and a Adafruit Feather RP2040
(Adafruit PRODUCT ID: 4884). I have Circuit Python 7 (adafruit-circuitpython-adafruit_feather_rp2040-en_US-7.0.0.uf2) running on the feather and running the example code from Solder Party to test the keyboard and receiving the same error.

`from bbq10keyboard import BBQ10Keyboard
import board

i2c = board.I2C()
kbd = BBQ10Keyboard(i2c)

while kbd.key_count == 0:
    pass

print(kbd.keys)`

The result is:
`Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 8, in <module>
  File "/lib/bbq10keyboard.py", line 194, in key_count
  File "/lib/bbq10keyboard.py", line 190, in status
  File "/lib/bbq10keyboard.py", line 256, in _read_register
  File "/lib/adafruit_bus_device/i2c_device.py", line 84, in write
OSError: [Errno 19] Unsupported operation

Code done running.`

Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit Feather RP2040 with rp2040