adafruit/circuitpython #4237: All state machines in use for more than two PIO on Pi Pico

kevinjwalters, 2021-02-21:
On `Adafruit CircuitPython 6.2.0-beta.2-18-g2a467f137 on 2021-02-15` this code spits a `RuntimeError: All state machines in use`

```
import time

import rp2pio
import adafruit_pioasm

print("15 second pause in case anything bad happens")
time.sleep(15)

flash_pio_code_2pin = """
.program flash2
    set x, 31
onloop:
    set pins, 3  [5]
    jmp x-- onloop

    set y, 31
offloop:  
    set pins, 0  [5]
    jmp y-- offloop
"""

flash_pio_code_3pin = """
.program flash3
    set x, 31
onloop:
    set pins, 7  [5]
    jmp x-- onloop
    set y, 31
offloop:  
    set pins, 0  [5]
    jmp y-- offloop
"""

flash_pio_code_4pin = """
.program flash4
    set x, 31
onloop:
    set pins, 15  [5]
    jmp x-- onloop

    set y, 31
offloop:  
    set pins, 0  [5]
    jmp y-- offloop
"""

KILLER_THIRD = True
KILLER_FOURTH = False
MIN_FREQUENCY=1908

sm1 = rp2pio.StateMachine(
    adafruit_pioasm.assemble(flash_pio_code_2pin),
    init=adafruit_pioasm.assemble("set pindirs {:d}".format(2**2 - 1)),
    frequency=MIN_FREQUENCY,
    first_set_pin=board.GP0,
    set_pin_count=2,
)

sm2 = rp2pio.StateMachine(
    adafruit_pioasm.assemble(flash_pio_code_2pin),
    init=adafruit_pioasm.assemble("set pindirs {:d}".format(2**2 - 1)),
    frequency=MIN_FREQUENCY,
    first_set_pin=board.GP3,
    set_pin_count=2,
)

if KILLER_THIRD:
    sm3 = rp2pio.StateMachine(
        adafruit_pioasm.assemble(flash_pio_code_3pin),
        init=adafruit_pioasm.assemble("set pindirs {:d}".format(2**3 - 1)),
        frequency=MIN_FREQUENCY,
        first_set_pin=board.GP8,
        set_pin_count=3,
    )

if KILLER_FOURTH:
    sm4 = rp2pio.StateMachine(
        adafruit_pioasm.assemble(flash_pio_code_4pin),
        init=adafruit_pioasm.assemble("set pindirs {:d}".format(2**4 - 1)),
        frequency=MIN_FREQUENCY,
        first_set_pin=board.GP11,
        set_pin_count=4,
    )

time.sleep(30)
```

Based on discussion in [Adafruit Forums: Can RP2040 PIO PULL be used in CircuitPython?](https://forums.adafruit.com/viewtopic.php?f=60&t=175662) this should work either due to code sharing or packing code into shared memory.

kevinjwalters, 2021-03-16:
FYI, I think code sharing kicks in based on the assembed code object, `sm1` and `sm2` would need to refer to same assembled code to do this.

Unrelated, elite users with 32 bytes of pio code may get a disappointment: https://github.com/raspberrypi/pico-sdk/issues/261

dhalbert, 2021-03-18:
I'm closing this because #4405 does fix it. I ran the test program above without error.

closed 2021-03-18 by dhalbert