adafruit/circuitpython #4240: Error 127 - Mac Big Sur 11.2.1 Install Issue

Isaac-Tait, 2021-02-22:
I am following [this](https://learn.adafruit.com/building-circuitpython/build-circuitpython) tutorial. Everything was going well up to the point where I ran `make BOARD=circuitplayground_express`. An alert popped up on my screen that arm-gcc was not a verified developer and so the download would not be processed. I have run into this quite a bit with Apple (I wish there was a way to turn this annoying feature off...) so, I opened up my *System Preferences* and then *Security & Privacy* and on the *General* tab I approved arm-gcc and then attempted to re-run `make BOARD=circuitplayground_express`. Now I am getting this error:

```
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
/bin/sh: arm-none-eabi-gcc: command not found
make: *** [build-circuitplayground_express/py/mpstate.o] Error 127
```

I am not quite sure what to do now. Any suggestions would be appreciated 😸 

Side question: I have a *Circuit Playground Classic* not the *Express*. However, if I run the command `make BOARD=circuitplayground_classic` I get the error:

```
Makefile:30: *** Invalid BOARD specified.  Stop.
```

Can I run a classic using the express "board"?

Cheers.

jepler, 2021-02-22:
CircuitPlayground "classic", with a 32u4 microcontroller, is not compatible with CircuitPython.

closed 2021-02-22 by jepler

Isaac-Tait, 2021-02-22:
Okay, good to know @jepler. I will purchase an express.

I am getting a new error now.

```
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[... 134 lines cut ...]
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
[Errno 2] No such file or directory: 'arm-none-eabi-gcc'
QSTR updated
/bin/sh: arm-none-eabi-gcc: command not found
/bin/sh: arm-none-eabi-gcc: command not found
make: *** [build-circuitplayground_express/py/mpstate.o] Error 127
```

Isaac-Tait, 2021-02-22:
`git version` 2.24.3
`python3 version` 3.9.2
`gettext version` 0.21
`gcc-arm-embedded` is installed (Confirmed this when I ran `--version` it returns `zsh: command not found: gcc-arm-embedded`. So, I ran `brew install gcc-arm-embedded` and it returned `already installed`).

tannewt, 2021-02-22:
> Okay, good to know @jepler. I will purchase an express.

I'd recommend the CircuitPlayground Bluefruit over the Express. It has more RAM and bluetooth.

Can you try `brew list gcc-arm-embedded` (IIRC) to list the files that brew installed? Perhaps it isn't in your path.

Isaac-Tait, 2021-02-27:
When I run the command `brew list gcc-arm-embedded` it returns this error:

`Error: No such keg: /usr/local/Cellar/gcc-arm-embedded`