adafruit/circuitpython #4250: matrix vector multiplication broken in ulab

kevin-tracy, 2021-02-23:
Matrix vector multiplication isn't working in ulab on the following version:
`Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit Grand Central M4 Express with samd51p20`

Here is an example of it's current functionality:
```
import ulab as np
A = np.eye(3)
B = np.array([1,2,3.0])

# the following two lines don't work 
C = np.linalg.dot(A,B)                    # ValueError: matrix dimensions do not match
C = np.linalg.dot(A,B.transpose())        # ValueError: matrix dimensions do not match

# this does work, but the resulting array C is a 2D column vector 
C = np.linalg.dot(A,B.reshape((3,1)))
C = C.flatten()                           # flatten to 1D vector
```

Ideally, this would work similarly to NumPy where a 2D array multiplied by a 1D array results in a 1D array.  Does this functionality exist under a different function? Thank you.

jepler, 2021-02-23:
@v923z I reproduced the same in the legacy branch of ulab as well as in master branch. Can you please look into it? If possible we'd like it fixed in legacy so we can take the fix in 6.2.

v923z, 2021-02-23:
@jepler Sure! What is the deadline? I should be able to fix it tomorrow.

v923z, 2021-02-23:
@jepler @kevin-tracy https://github.com/v923z/micropython-ulab/pull/339 should hopefully fix the issue.

kevin-tracy, 2021-02-23:
@v923z thank you very much! Out of curiosity, how come that change was made to the [micropython-ulab](https://github.com/v923z/micropython-ulab) repo instead of [circuitpython-ulab](https://github.com/adafruit/circuitpython-ulab)?

v923z, 2021-02-23:
@kevin-tracy Because @jepler simply links to micropython-ulab, so it makes sense to implement all fixes there. I think, having the same thing at two different places is generally not a sound idea.

jepler, 2021-02-24:
Initially we thought we might have to make local changes to allow ulab to work with circuitpython but v923z has been very responsive to all our needs, keeping the changes in his repo.  It is working out very well!

v923z, 2021-02-24:
@kevin-tracy I forgot to ask, do you think you could write a test script for `linalg.dot`? We should have been able to catch this error earlier, and the fact that we didn't means that the test was not quite up to the task... In case you are interested, here is what you would need: https://github.com/v923z/micropython-ulab/#testing

kevin-tracy, 2021-02-24:
@v923z I would be happy to! I'll add them sometime this week.

v923z, 2021-02-24:
@kevin-tracy Many thanks! @jepler Jeff, I hope that my appeal for testing doesn't qualify as hijacking the thread.

kevin-tracy, 2021-02-26:
@v923z this might not be the right place to ask this question, but in writing tests I'm having an issue generating the .exp file for a new test. In the readme of the repo it says:

> In case you have a new snippet, i.e., you have no expected results file, or if the results differ from those in the expected file, a new expected file will be generated in the root directory.

But when I do this for a new `new_tests_2.py,` I get the following error:
```
1 tests failed: new_tests_2
-e 
FAILURE *
diff: *.exp: No such file or directory
diff: *.out: No such file or directory
```
Am I misunderstanding something or should I make an issue for this? Thanks.

kevin-tracy, 2021-02-26:
Two quick updates to the above comment:

1. It looks like when a test script results in a different output than expected, it creates the appropriate .exp file in the root directory. However, when there is no expected output .exp file, it **does not** produce the appropriate .exp file in the root directory. 
2. It also looks as if the matrix-vector multiplication is still not working. I tried both with the testing tool on my computer as well as the most recent version on my board:

```
Adafruit CircuitPython 6.2.0-beta.2-160-g5158aec8c on 2021-02-26; Adafruit Grand Central M4 Express with samd51p20
>>> import ulab as np
>>> A = np.eye(3)
>>> b = np.array([1,2,3])
>>> np.linalg.dot(A,b)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: matrix dimensions do not match
```
On my board I am running `20210226-5158aec.uf2` from [here](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/grandcentral_m4_express/en_US/).

v923z, 2021-02-27:
@kevin-tracy Sorry, I didn't spell this out, but anything written on `micropython-ulab` is for `micropython`.

But more importantly, the fix is in a still-open PR, https://github.com/v923z/micropython-ulab/pull/339, so Jeff hasn't yet pulled in the changes, because the PR is still not resolved. Hence, loading an adafruit-compiled piece of firmware is not going to make the cut here. 

What you can do, however, is simply clone https://github.com/v923z/micropython-ulab/, switch to the `legacy-dot` branch, and then run the build script. That will clone `micropython`, and run the tests. If you are happy with the test results, then you can PR against that branch.

closed 2021-03-16 by tannewt

amitkrbhuyan, 2021-05-25:
Is the matrix multiplication issue fixed for circuitpython?

jepler, 2021-05-25:
@amitkrbhuyan we believe it is fixed in the current stable release, 6.2.0.

PR #4410 (2021-03-16): Update ulab
Closes #4250.