adafruit/circuitpython #4259: Speaker can only play so many tones

KTibow, 2021-02-24:
## Problem
I found when I use `play_tone`, I can only play so many separate tones. So I can play 500 as many times as I want, at any speed. But if then I play 490, then 480, and so on, eventually it stops working.
## Explanation
The library uses `simpleio.tone`. It gives the error
```python3
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "simpleio.py", line 65, in tone
NameError: name 'AudioOut' is not defined
```
eventually. This is caused by a `ValueError`, which is catched, and causes it to proceed to the line
```
        with AudioOut(pin) as dac:
```
This fails, because the only working solution is `pwmio`. Now, here's the exact `ValueError`:
```python3
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 2, in beep
ValueError: No more timers available
```
I believe this is caused by not cleaning up after the PWM instance once used.
## Suggested solution
In the MagTag library, import pwmio instead of simpleio. Allocate a variable frequency PWM. Replace the function
```python3
    def play_tone(self, frequency, duration):
        """Automatically Enable/Disable the speaker and play
        a tone at the specified frequency for the specified duration
        It will attempt to play the sound up to 3 times in the case of
        an error.
        """
        if frequency <= 0:
            raise ValueError("The frequency has to be greater than 0.")
        self._speaker_enable.value = True
        attempt = 0
        # Try up to 3 times to play the sound
        while attempt < 3:
            try:
                simpleio.tone(board.SPEAKER, frequency, duration)
                break
            except NameError:
                pass
            attempt += 1
        self._speaker_enable.value = False
```
with
```python3
    def play_tone(self, frequency, duration):
        """Automatically Enable/Disable the speaker and play
        a tone at the specified frequency for the specified duration
        It will attempt to play the sound up to 3 times in the case of
        an error.
        """
        if frequency <= 0:
            raise ValueError("The frequency has to be greater than 0.")
        self._speaker_enable.value = True
        self._speaker.duty_cycle = 0x8000
        self._speaker.frequency = frequency
        time.sleep(duration)
        self._speaker.duty_cycle = 0
        self._speaker_enable.value = False
```
## Alternative solutions
Fix the bug where it doesn't free up the timer <s>in `simpleio`</s> in [the ESP32S2 pwmio port](https://github.com/adafruit/circuitpython/blob/main/ports/esp32s2/common-hal/pwmio/PWMOut.c).

Feel free to correct me or to suggest edits in any of this issue.

todbot, 2021-07-11:
Verified something similar still happens in latest build `Adafruit CircuitPython 7.0.0-alpha.4-15-g22e8a5058 on 2021-07-11; Adafruit FunHouse with ESP32S2` with this code:
```py
import time, board
import pwmio
while True:
    for f in (262, 294, 330, 349, 392, 440, 494, 523):
        with pwmio.PWMOut(board.SPEAKER, frequency=f, duty_cycle=2**15) as pwm:
            time.sleep(0.25)
    time.sleep(1)
```
The above will result in 3 tones being played and then the error:
```
Traceback (most recent call last):
  File "code.py", line 6, in <module>
RuntimeError: All timers in use
```
This is the same technique being used by `simpleio` (which in turn is used by `adafruit_funhouse`).

hierophect, 2021-07-13:
My testing for #4999 resolves your test case for my Wroom but if you can, please give that PR a spin and let me know if you encounter any additional problems.

todbot, 2021-07-14:
Yep seems to work. I tested this on a MagTag with this CI build (`Adafruit CircuitPython 7.0.0-alpha.4-65-ge2888cba6 on 2021-07-13; Adafruit MagTag with ESP32S2`)

PR #4999 (2021-07-14): ESP32S2: Fix PWM timer leak and variable frequency conflicts
Resolves a number of issues within the ESP32S2 PWMIO module, including issues with the deinit function not releasing system timers properly, various unchecked issues with variable frequency channel assignment, and duty cycle config recalculation when the frequency is changed.

Resolves #4259
closed 2021-07-14 by hierophect