adafruit/circuitpython #4260: RP2040 - RTC - time_source not working

jerryneedell, 2021-02-24:
The rtc.set_time_source() does not appear to be working properly on the RP204 

```
Adafruit CircuitPython 6.2.0-beta.2-48-g4cac412fa-dirty on 2021-02-24; Raspberry Pi Pico with rp2040
>>> 
>>> import gps_time_source
Set GPS as time source
Fix timestamp: 00/00/0 17:36:55
RTC timestamp: 00/32/1985 32:08:16
Local time: 00/00/0 17:36:55
Fix timestamp: 02/24/2021 17:36:56
RTC timestamp: 00/32/1985 32:08:16
Local time: 02/24/2021 17:36:56
Fix timestamp: 02/24/2021 17:36:57
RTC timestamp: 00/00/16 32:08:16
Local time: 02/24/2021 17:36:57
```

here is the code --- gsp_time_source.py from the bundel examples
```
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

# Simple script using GPS timestamps as RTC time source
# The GPS timestamps are available without a fix and keep the track of
# time while there is powersource (ie coin cell battery)

import time
import board
import busio
import rtc
import adafruit_gps

uart = busio.UART(board.GP12, board.GP13, baudrate=9600, timeout=10)
# i2c = busio.I2C(board.SCL, board.SDA)

gps = adafruit_gps.GPS(uart, debug=True)
# gps = adafruit_gps.GPS_GtopI2C(i2c, debug=False)  # Use I2C interface

gps.send_command(b"PMTK314,0,1,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0")
gps.send_command(b"PMTK220,1000")

print("Set GPS as time source")
rtc.set_time_source(gps)
the_rtc = rtc.RTC()

last_print = time.monotonic()
while True:

    gps.update()
    # Every second print out current time from GPS, RTC and time.localtime()
    current = time.monotonic()
    if current - last_print >= 1.0:
        last_print = current
        if not gps.timestamp_utc:
            print("No time data from GPS yet")
            continue
        # Time & date from GPS informations
        print(
            "Fix timestamp: {:02}/{:02}/{} {:02}:{:02}:{:02}".format(
                gps.timestamp_utc.tm_mon,  # Grab parts of the time from the
                gps.timestamp_utc.tm_mday,  # struct_time object that holds
                gps.timestamp_utc.tm_year,  # the fix time.  Note you might
                gps.timestamp_utc.tm_hour,  # not get all data like year, day,
                gps.timestamp_utc.tm_min,  # month!
                gps.timestamp_utc.tm_sec,
            )
        )

        # Time & date from internal RTC
        print(
            "RTC timestamp: {:02}/{:02}/{} {:02}:{:02}:{:02}".format(
                the_rtc.datetime.tm_mon,
                the_rtc.datetime.tm_mday,
                the_rtc.datetime.tm_year,
                the_rtc.datetime.tm_hour,
                the_rtc.datetime.tm_min,
                the_rtc.datetime.tm_sec,
            )
        )

        # Time & date from time.localtime() function
        local_time = time.localtime()

        print(
            "Local time: {:02}/{:02}/{} {:02}:{:02}:{:02}".format(
                local_time.tm_mon,
                local_time.tm_mday,
                local_time.tm_year,
                local_time.tm_hour,
                local_time.tm_min,
                local_time.tm_sec,
            )
        )
```
a basic test of the RTC works OK

```
Adafruit CircuitPython 6.2.0-beta.2-48-g4cac412fa-dirty on 2021-02-24; Raspberry Pi Pico with rp2040
>>> 
>>> import rtc
>>> import time
>>> r = rtc.RTC()
>>> r.datetime = time.struct_time((2021, 2, 24, 15, 14, 15, 0, -1, -1))
>>> while True:
...       print(r.datetime)
...       time.sleep(1)
...     
...     
... 
struct_time(tm_year=2021, tm_mon=2, tm_mday=24, tm_hour=15, tm_min=14, tm_sec=18, tm_wday=2, tm_yday=55, tm_isdst=-1)
struct_time(tm_year=2021, tm_mon=2, tm_mday=24, tm_hour=15, tm_min=14, tm_sec=19, tm_wday=2, tm_yday=55, tm_isdst=-1)
struct_time(tm_year=2021, tm_mon=2, tm_mday=24, tm_hour=15, tm_min=14, tm_sec=20, tm_wday=2, tm_yday=55, tm_isdst=-1)
struct_time(tm_year=2021, tm_mon=2, tm_mday=24, tm_hour=15, tm_min=14, tm_sec=21, tm_wday=2, tm_yday=55, tm_isdst=-1)
struct_time(tm_year=2021, tm_mon=2, tm_mday=24, tm_hour=15, tm_min=14, tm_sec=22, tm_wday=2, tm_yday=55, tm_isdst=-1)
struct_time(tm_year=2021, tm_mon=2, tm_mday=24, tm_hour=15, tm_min=14, tm_sec=23, tm_wday=2, tm_yday=55, tm_isdst=-1)
```

here is another test run -- RTC is "close"

```

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.2-48-g4cac412fa-dirty on 2021-02-24; Raspberry Pi Pico with rp2040
>>>
>>>
>>> import rtc_time_source
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'rtc_time_source'
>>> import gps_time_source
Set GPS as time source
Fix timestamp: 02/24/2021 18:09:51
RTC timestamp: 02/24/2021 15:28:55
Local time: 02/24/2021 18:09:51
Fix timestamp: 02/24/2021 18:09:52
RTC timestamp: 02/24/2021 15:28:56
Local time: 02/24/2021 18:09:52
Fix timestamp: 02/24/2021 18:09:53
RTC timestamp: 02/24/2021 15:28:57
Local time: 02/24/2021 18:09:53
Fix timestamp: 02/24/2021 18:09:54
RTC timestamp: 02/24/2021 15:28:58
Local time: 02/24/2021 18:09:54
Fix timestamp: 02/24/2021 18:09:55
RTC timestamp: 02/24/2021 15:28:59
Local time: 02/24/2021 18:09:55
Fix timestamp: 02/24/2021 18:09:57
RTC timestamp: 02/24/2021 15:29:00
Local time: 02/24/2021 18:09:57

```

jerryneedell, 2021-02-24:
just to clarify -- here is a clean start after a power cycle and with a good GPS fix
```

Adafruit CircuitPython 6.2.0-beta.2-48-g4cac412fa-dirty on 2021-02-24; Raspberry Pi Pico with rp2040
>>> import gps_time_source
Set GPS as time source
Fix timestamp: 02/24/2021 18:13:10
RTC timestamp: 00/00/16 32:08:16
Local time: 02/24/2021 18:13:10
Fix timestamp: 02/24/2021 18:13:11
RTC timestamp: 16/80/512 32:08:16
Local time: 02/24/2021 18:13:11
Fix timestamp: 02/24/2021 18:13:12
RTC timestamp: 07/16/18016 32:08:16
Local time: 02/24/2021 18:13:12
Fix timestamp: 02/24/2021 18:13:14
RTC timestamp: 00/00/16 32:08:16
Local time: 02/24/2021 18:13:14
Fix timestamp: 02/24/2021 18:13:15
RTC timestamp: 07/16/15940 32:08:16
Local time: 02/24/2021 18:13:15
Fix timestamp: 02/24/2021 18:13:16
RTC timestamp: 00/32/1985 32:08:16
Local time: 02/24/2021 18:13:16
Fix timestamp: 02/24/2021 18:13:17
RTC timestamp: 00/00/16 32:08:16
Local time: 02/24/2021 18:13:17
Fix timestamp: 02/24/2021 18:13:18
RTC timestamp: 00/00/16 32:08:16
Local time: 02/24/2021 18:13:18
Fix timestamp: 02/24/2021 18:13:19
RTC timestamp: 00/00/1 32:08:16
Local time: 02/24/2021 18:13:19
```

jepler, 2021-03-12:
@jerryneedell can you test this again?   I think maybe this _was_ #4379 but I didn't closely read this issue.

I think there may be a misunderstanding or an opportunity to clarify the documentation.  `rtc.RTC()` is always the on-chip RTC, so using `set_time_source()` doesn't change it.  However, it does change what `time.time()` and `time.localtime()` (with 0 args) use as their source of the time.

So it looks like in all your prints, what you saw was only `rtc.RTC()` misbehaving, and possibly a misunderstanding of how `rtc.RTC()` and `rtc.set_time_source()` are related (they aren't, except that you could hypothetically `rtc.set_time_source(rtc.RTC())`).  However, I still want to hear your interpretation of things before I just close the issue.

jerryneedell, 2021-03-12:
@jepler This si what I get now -- is it the expected result?

```
Adafruit CircuitPython 6.2.0-beta.4-78-g8d3217cbf on 2021-03-12; Raspberry Pi Pico with rp2040
>>> import gps_time_source
Set GPS as time source
No time data from GPS yet
Fix timestamp: 03/12/2021 10:39:18
RTC timestamp: 01/01/2020 00:04:20
Local time: 03/12/2021 10:39:18
Fix timestamp: 03/12/2021 10:39:19
RTC timestamp: 01/01/2020 00:04:21
Local time: 03/12/2021 10:39:19
Fix timestamp: 03/12/2021 10:39:20
RTC timestamp: 01/01/2020 00:04:22
Local time: 03/12/2021 10:39:20
Fix timestamp: 03/12/2021 10:39:21
RTC timestamp: 01/01/2020 00:04:23
Local time: 03/12/2021 10:39:21
Fix timestamp: 03/12/2021 10:39:22
RTC timestamp: 01/01/2020 00:04:24
Local time: 03/12/2021 10:39:22

```

jepler, 2021-03-12:
The "RTC timestamp" line is now consistently advancing by about 1 second per second.  I don't see where your program set the current time of the `rtc.RTC()` object by assigning its `datetime` property, so it's also expected that it starts counting from some arbitrary time in the past.  So, yeah, it looks right to me.  Is there something we can improve about the docs so that you think others would understand why this behavior is the anticipated behavior?

jerryneedell, 2021-03-12:
I'm fine with your explanation. If anything, the example in the gps library may need to be updated to set the RTC. Thanks.

closed 2021-03-12 by jerryneedell