adafruit/circuitpython #4261: audiobusio mic.record maxing out at 32 samples on SAMD21

kattni, 2021-02-24:
Tested on 6.1.0 stable and 6.2.0-beta.2. I ran into this trying to write up a sound-reactive LED example for the BLM Badge. I distilled the code down to what's necessary to reproduce. It fails the same way on the Circuit Playground Express. This code works properly on Circuit Playground Bluefruit.

When the code runs, `mic.record` only records 32 samples, the rest of the 160 are 0. Below is an example. (All of the examples I can find of using sound to get levels uses 160 samples. That is where I got that number.)

```
array('H', [33445, 33360, 33397, 33451, 33380, 33401, 33402, 33420, 33466, 33399, 33382, 33373, 33410, 33440, 33427, 33334, 33398, 33402, 33329, 33335, 33420, 33384, 33466, 33378, 33411, 33440, 33361, 33376, 33436, 33444, 33386, 33321, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])
```

Here's the catch. If I update the file and save, _right before_ the board resets, it fills the array. So without a `time.sleep(5)` in it, the last few arrays are filled. With the `time.sleep(5)` in it, sometimes it fills one array, but often misses it. Below is an example from right before the board resets. I changed it to `time.sleep(0.5)` and it filled the last array before reset.

```
array('H', [33384, 33238, 33257, 33224, 33226, 33252, 33233, 33290, 33212, 33225, 33133, 33172, 33192, 33190, 33243, 33230, 33241, 33285, 33235, 33255, 33332, 33173, 33251, 33289, 33256, 33294, 33195, 33311, 33261, 33263, 33272, 33193, 33249, 33259, 33235, 33290, 33262, 33302, 33196, 33251, 33202, 33231, 33266, 33157, 33172, 33205, 33281, 33225, 33222, 33205, 33215, 33255, 33298, 33261, 33208, 33230, 33293, 33222, 33257, 33245, 33264, 33202, 33244, 33273, 33219, 33198, 33209, 33278, 33267, 33278, 33117, 33201, 33228, 33252, 33347, 33296, 33264, 33239, 33270, 33277, 33237, 33220, 33269, 33293, 33253, 33199, 33291, 33238, 33256, 33207, 33208, 33203, 33292, 33262, 33212, 33221, 33189, 33171, 33241, 33269, 33194, 33247, 33255, 33179, 33307, 33225, 33217, 33299, 33210, 33196, 33296, 33232, 33263, 33183, 33263, 33172, 33141, 33194, 33202, 33232, 33240, 33237, 33280, 33192, 33161, 33234, 33229, 33263, 33216, 33222, 33224, 33231, 33233, 33282, 33223, 33180, 33192, 33257, 33219, 33288, 33243, 33252, 33250, 33321, 33226, 33222, 33310, 33253, 33242, 33246, 33248, 33263, 33202, 33286, 33162, 33271, 33230, 33268, 33234, 33227])
```

Here is the code to reproduce.

```
import time
import array
import math
import board
import audiobusio

mic = audiobusio.PDMIn(board.MICROPHONE_CLOCK, board.MICROPHONE_DATA, sample_rate=16000, bit_depth=16)

while True:
    samples = array.array("H", [0] * 160)
    mic.record(samples, len(samples))
    print(samples)
    time.sleep(5)
```

tannewt, 2021-02-24:
Is record returning the correct number of samples back to you? It will return less than what you give when it can't keep up with the data rate.

kattni, 2021-02-24:
@tannewt It appears to be? Should I be using a different sample rate?

tannewt, 2021-02-25:
Here is what I just did in my volume demo:

```python
while True:
    count = pdm.record(s16, len(s16))
    magnitude = normalized_rms(memoryview(s16)[:count])
    diff = max(0, magnitude - 50)
    led.duty_cycle = min(int(diff) * 256, 2 ** 16 - 1)
    np.brightness = diff / 500
```

dhalbert, 2022-03-02:
Thanks for testing!

closed 2022-03-02 by dhalbert