adafruit/circuitpython #4271: audiobusio.I2SOut on RP2040 playing wav from file: i2s.playing False to early and random [audio][RP2][BUG]

dglaude, 2021-02-25:
Testing I2SOut on pico + "pico base" with 6.2.0-beta.2-154-gd4bf0d5e2

("pico base" use PCM5100A DAC for line out audio over I2S, just like in Raspberry Pi reverence design)

Playing "StreetChicken.wav" from flash on I2S, the file and code inspiration come from the learn guide: https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/circuitpython-wiring-test

The sound is metalic (randomly, sometime it is good) and i2s.playing return False "randomly" long before the end of the sound file.

Code:
```
import board
import audiocore
import audiobusio
import time

print("Playing 10 times StreetChicken.wav")

f = open("StreetChicken.wav", "rb")
wav = audiocore.WaveFile(f)
i2s = audiobusio.I2SOut(board.GP27, board.GP28, board.GP26)

for i in range(10):
    start=time.monotonic()
    i2s.play(wav)
    while i2s.playing:
        pass
    stop = time.monotonic()
    print("Total lenght = ", stop, "-", start, "=", stop-start)

```

Expected result:

Perfect sound, the wave file should play 10 times and last more or less exactly the same length.

Effective result:

The sound is unperfect (metalic scratch) and the timing is random

> Playing 10 times StreetChicken.wav
> Total lenght =  972.984 - 972.949 = 0.0349121
> Total lenght =  973.148 - 972.985 = 0.163086
> Total lenght =  974.415 - 973.149 = 1.26587
> Total lenght =  974.985 - 974.416 = 0.568848
> Total lenght =  976.125 - 974.987 = 1.13818
> Total lenght =  980.144 - 976.126 = 4.01782
> Total lenght =  981.144 - 980.145 = 0.998779
> Total lenght =  981.947 - 981.145 = 0.802002
> Total lenght =  983.191 - 981.948 = 1.24292
> Total lenght =  983.877 - 983.192 = 0.685059

dglaude, 2021-02-25:
Replacing
```
while i2s.playing:
    pass

```

```
by `time.sleep(15)`

Make a much better sound (except the first play) and let you hear the full sound.

> Playing 10 times StreetChicken.wav
> Total lenght =  258.805 - 243.804 = 15.001
> Total lenght =  273.807 - 258.806 = 15.0011
> Total lenght =  288.809 - 273.808 = 15.001
> Total lenght =  303.811 - 288.81 = 15.001
> Total lenght =  318.813 - 303.812 = 15.001
> Total lenght =  333.815 - 318.814 = 15.001
> Total lenght =  348.817 - 333.816 = 15.001
> Total lenght =  363.819 - 348.818 = 15.001
> Total lenght =  378.821 - 363.82 = 15.001
> Total lenght =  393.823 - 378.822 = 15.0011

dglaude, 2021-02-25:
This give much better result (so replacing pass by a short sleep):
```
    while i2s.playing:
        time.sleep(0.01)
```

Still the first play can be too short and/or metallic.

recantha, 2021-03-20:
Just wanted to confirm that this happens as David says. It's odd. It's the kind of thing you think "should work" but somehow it doesn't *quite*. Impressive stuff, though, what goes on under the hood.

tannewt, 2021-03-22:
I think next step to debugging this is to capture the I2S output stream and then compare it to the original file. Doing that would allow one to characterize how the playback is failing. (Repeated regions of the source for example.)

bhilimon, 2021-03-28:
I've been working on getting audio (wav) working on my Feather RP2040 and I'm also having these similar issues but wildly inconsistent behavior. Mine is coupled with the Adafruit I2S mono amp (https://www.adafruit.com/product/3006). Using the same StreetChicken.wav test file. I'm not sure how far long I2S audio is on this new board but I figured I'd throw in what I'm experiencing if I can help at all.

Is there a preference to what pins we should be using for I2S audio? My setup is as follows:

audio = audiobusio.I2SOut(board.D11, board.D12, board.D25)

It does play audio but I have the following inconsistent/random issues:

- Most of the time there's some static or other noise coming through along with the audio. Pretty much guaranteed on the first play, and maybe 50/50 on future plays. Sometimes the additional noise will cut in or out mid-play. Only sometimes the audio comes through clear for it's entirety. It doesn't seem to be a gain/volume issue. I've been playing around modifying the gain on the amp board but seem to have the same results no matter what. Also swapping between 2 different speakers. I've played with 3.3v and 5v to the amp, that also doesn't seem to matter either.

- Sometimes the playing() function gets "stuck" and always reports True even after the audio file is done playing.

- Sometimes the audio cuts off early. The playing() function seems to change back to False when this happens.

I'll continue trying things with my setup and report back if I make any progress.

tannewt, 2021-03-29:
The pins shouldn't matter because it's all PIO under the hood. It's likely a timing hiccup with the DMA and sample conversion code.

bhilimon, 2021-03-29:
I forgot to reply back with a few notes after playing around yesterday.

- I was trying to play audio while also running some NeoPixel animations. When both were playing the audio distortion would randomly come in and out. Disabling or pausing the NeoPixel animations while audio is playing seems to have completely cleared up the audio distortion except for the initial audio file (next bullet). I don't know if this is a CPU/memory thing or if I just can't do audio and animations at the same time. Either way, I can work around that. 
- The first audio file that is played is still always distorted. It doesn't matter how long it is (.wav). The second time an audio file plays it comes through fine, even if it's a different file. I was thinking it might have been something with loading the file from flash, but doesn't seem to be as subsequent files play fine.
- As a test I moved to a PIO "group" that is not shared with other used pins, that didn't seem to help things.
- The audio.playing() function still appears bugged. I worked around that one for now too.
- I also swapped in my analog amp and switched to audiopwmio instead of audiobusio (i2s) and the distortion on the first file problem followed it. I don't think I tested the playing() function.

Hope it helps.

rsbohn12, 2021-05-27:
Similar results trying to play a RawSample on the Feather RP2040 with MAX98537.

`Adafruit CircuitPython 7.0.0-alpha.2-646-g7df5d74d9 on 2021-05-22; Adafruit Feather RP2040 with rp2040`

    dac = audiobusio.I2SOut(bit_clock=board.D5, word_select=board.D6, data=board.D9)
    clip = make_squarewave(440*Hz, duty_cycle=0.5)
    dac.play(clip)
    time.sleep(0.5)
    dac.stop()

I hear a click at dac.stop() but not audible tone is produced. I measured a 320 kHz signal on the MAX98537 terminal block.

When I connect the same breakout board to the SAMD21 QT Py it does produce audible tones.
I also tried `Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather RP2040 with rp2040` -- still no sound.

dhalbert, 2021-08-12:
This should be fixed by #5079. Please reopen or resubmit if it isn't.

closed 2021-08-12 by dhalbert

adamwolf, 2022-03-04:
I'm on an Itsy Bitsy RP2040.

I'm running the following code, from the example on https://learn.adafruit.com/mp3-playback-rp2040/pico-i2s-mp3, modified to set the pins.

```
import audiocore
import board
import audiobusio

audio = audiobusio.I2SOut(bit_clock = board.D10, word_select = board.D11, data=board.D12)

wave_file = open("StreetChicken.wav", "rb")
wav = audiocore.WaveFile(wave_file)

print("Playing wav file!")
audio.play(wav)
while audio.playing:
    pass
print("Done!")
```

The audio is intermittently crackly, sometimes metallic or spacey, and it sometimes fades in and out at different times after I reset it to have it play again.

I loaded CircuitPython 7.2.0 on it today, and had the same issue, and then loaded the latest alpha off CircuitPython.org, and have the same issues.

```
Adafruit CircuitPython 7.3.0-alpha.0-23-g9265cca75 on 2022-03-04; Adafruit ItsyBitsy RP2040 with rp2040
Board ID:adafruit_itsybitsy_rp2040
```

I left my scope at the shop, so I can't look at the signals until Monday, but I did wiggle the jumper wires between the board and the MAX98357A amp to see if it affected anything, and I didn't noticed any changes.  I added a big fat cap on the power rails of the amp, wondering if some of the crackle might be power issues, and didn't notice any changes either.

Does this sound like the same sort of symptoms as the rest of this thread?

bhilimon, 2022-03-04:
Sounds different from my original issues. I recently updated my project (that I originally was working through these issues on) to CP 7 and the audio issues were fixed as long as I'm running .wav files at 11250 htz. For some reason 22500 htz (.wav) have some weird noise, but the quality different isn't noticeable on my project so I haven't spent any timing poking around at it.

phantomsixthplayer, 2022-06-02:
This also happens on 
```
Adafruit CircuitPython 7.3.0 on 2022-05-23; Adafruit Grand Central M4 Express with samd51p20

UF2 Bootloader v3.14.0 SFHWRO
Model: Grand Central M4 Express
Board-ID: SAMD51P20A-GrandCentral-v0
```

example: https://youtu.be/QqLHkpqCbhw

roddersuk, 2022-11-14:
I'm having the same problem with CircuitPython 7.3.3 on a pico. Anyone found a fix/workaround?

djairjr, 2023-02-10:
Same problem here, with Seeed Xiao RP2040, with Circuitpython 8.
The sound distorts and breaks the code.

dhalbert, 2023-02-10:
Please open a new issue, thanks, with a simple test program.