adafruit/circuitpython #4272: Circuit python crashes when DHT11 on pin 31

electric1236, 2021-02-26:
I am designing a shield for the Metro m4 grand central and was testing a bunch of pins with the dht11 to make sure that all the pins will work with it as I cant connect it to another pin once the shield is made. So all the pins worked as expected (#D4-9 D11, D13-19 D22-45) except D31 just crashes circuit python for some reason, and it goes into safe mode. as soon at the dht11 is connected to pin31 it crashes, it works if it's not connected. the DHT is wired with 5v and then 10k pull up resistor on the data pin just like in the tutorial on adafruit.
[boot_out.zip](https://github.com/adafruit/circuitpython/files/6047220/boot_out.zip)

Im attaching all the files in the circuit python drive.

ajs256, 2021-02-26:
What was the error printed on the REPL when it crashed?

tannewt, 2021-02-26:
@electric1236 Are you pulling up to 5v? The SAMD51 is NOT 5v tolerant so you should be using 3.3v.

electric1236, 2021-02-26:
on the tutorial, it says it should be fine (https://learn.adafruit.com/dht/dht-circuitpython-code). though I'll switch to pulling it up with 3.3v but power the dht11 with 5v. Though I don't think that why it crashed cause it worked fine with every other pin, though ill try it out.

electric1236, 2021-02-26:
No, no error was printed, it just froze then I rebooted it and it said it crashed hard and it was in safe mode and to report it to here

electric1236, 2021-02-27:
I was just testing it again with the 3.3v, when i powered it with 5v but had a 3.3v pull up on the data pin it still had the problem, but when it powered with 3.3v and has a 3.3v pull up it works fine.

also sorry I forgot occasionally it did output an error message "Checksum did not validate. Try again." before crashing when it crashed I heard the noise that it disconnected from my computer and then reconnected.

when it rebooted it says, it also sometimes said something different and to report it here
"Running in safe mode! Not running saved code.

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler."

tannewt, 2021-08-11:
@electric1236 Please retry this with the absolute latest. I wasn't able to reproduce this on a Metro M4 on pin D1 which is also PA22.

closed 2021-08-12 by tannewt