adafruit/circuitpython #4273: AttributeError: 'module' object has no attribute 'I2C'

gregwa1953, 2021-02-26:
This is an issue for Raspberry Pi Pico implementation of CircuitPython.

trying to import board and initialize I2C.  Get back:

AttributeError: 'module' object has no attribute 'I2C'

When trying to do a dir(board) get back:
`>>> dir(board)
['__class__', 'A0', 'A1', 'A2', 'A3', 'GP0', 'GP1'...`
This is with both the Adafruit CircuitPython 6.1.0-rc.1-51-gfb1e0106b on 2021-01-20 version as well as Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11 version.

jerryneedell, 2021-02-26:
There is not a "default" set of pins defined for I2C -- you will have to use busio to configure it
For example, if using GP0 for SDA and GP1 for SDL
```    
i2c = busio.I2C(board.GP1, board.GP10)
```

gregwa1953, 2021-02-26:
I'm using physical pins 11 and 12 GP8 and GP9 I2C0.
I have tried
i2c = busio.I2C(board.GP8, board.GP9) and gotten "invalid pins" error

I'm simply trying to use the "standard" Adafruit displayio_sh1107_simpletest.py example file.

jerryneedell, 2021-02-26:
swap the order SCL is the first argument
```i2c = busio.I2C(board.GP9, board.GP8)```

gregwa1953, 2021-02-26:
That took care of the error.
Demo doesn't work yet, but that's my issue.
Thank you.

closed 2021-02-26 by gregwa1953