adafruit/circuitpython #4281: Pi Pico doesn't handle more than 2 PWM channels well

alessaba, 2021-02-27:
I have a really simple script to interact with a random RGB strip i got. the strip is common anode but it shouldn't matter.

Basically i initialize a regular power pin, and 3 PWM ones (for R,G,B); I leave the power pin turned on and play with the 3 PWM channels, reversing the duty cycle values (2^16 is completely off and 0 completely on).
Problem is, only 2 channels work at a time, the third's duty cycle doesn't update. i tried in the REPL and 2 channels always works completely fine, but when i add the third channel and try to modify the duty cycle, nothing happens.

I tried with MicroPython and what i want works perfectly.

I believe this is a bug, and not intended CircuitPython behaviour, right?

kevinjwalters, 2021-02-27:
This sounds like #4210 - try "absolute latest" version of CircuitPython for the recent fix.

alessaba, 2021-02-27:
tried with [the latest build](https://adafruit-circuit-python.s3.amazonaws.com/bin/raspberry_pi_pico/en_US/adafruit-circuitpython-raspberry_pi_pico-en_US-20210227-24fdda0.uf2), and the problem was solved! Thanks!

closed 2021-02-27 by alessaba