adafruit/circuitpython #4282: CIRCUITPY does not show up on Windows for Maker Pi Pico but ok on Pi Pico

kevinjwalters, 2021-02-27:
I have been using `adafruit-circuitpython-raspberry_pi_pico-en_US-20210215-2a467f1.uf2` successfully for sometime now but I upgraded to a more recent version to test out a PWM fix. I noticed this made `CIRCUITPY` disappear on a Windows 8.1 desktop. Further investigation shows this problem appears to be unique to the Cytron Maker Pi Pico, the bare Pi Pico board does not suffer from this. A Raspberry Pi 4 running their standard os can see the `CIRCUITPY` drive.

I've tested both `adafruit-circuitpython-raspberry_pi_pico-en_US-20210226-2e6e91d.uf2` and `adafruit-circuitpython-raspberry_pi_pico-en_US-20210227-24fdda0.uf2` and these two suffer from this problem.

This seems rather surprising. Is there any code in CircuitPython specific to the Maker Pi Pico?

My bare board says `1.3` on the underside which might be a version number. The underside of Pico board on the Maker Pi Pico is not visible. Both boards appear the same on the topside to my eye in terms of layout and components.

@kiate and @waiweng83 may be interested in this. Would be interesting if this happens on Windows 10...

kevinjwalters, 2021-02-27:
For the problematic combination I also note that device manager shows a [pair](url) of CircuitPython MIDI devices when just the Maker Pi Pico is plugged in. One is ok and one has the yellow problem triangle. The yellow triangle one says:

> This device cannot start. (Code 10)

> The I/O device is configured incorrectly or the configuration parameters to the driver are incorrect.

Both have USB PID of `80F4`. In `Information` under `Properties / Events` it references them as:

GOOD: `\VID_239A&PID_80F4&MI_07\6&797DB01&0&0007`

BAD: `\VID_239A&PID_80F4&MI_05\6&797DB01&0&0005`

dhalbert, 2021-02-27:
This is very odd. I looked at the Cytron board, and though I don't see a schematic, it just appears to add a bunch of stuff to various pins. The USB is untouched.

Try detaching all the Picos and cleaning out the USB devices with the Device Cleanup tool https://www.uwe-sieber.de/misc_tools_e.html. Windows may be confused.

Another thing to do is to erase flash completely: https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/circuitpython#flash-resetting-uf2-3083182-8

waiweng83, 2021-02-27:
@kevinjwalters I tried with both version of circuitpython you mentioned on my Maker Pi Pico and they work just fine on Windows 10. Perhaps you may try out the methods as suggested by @dhalbert and see how it goes.

Thanks.

kevinjwalters, 2021-02-27:
@waiweng83 Thanks, I'll give them a go and report back.

kevinjwalters, 2021-02-27:
All working now after a big device cleanse from Uwe Sieber's tool!

closed 2021-02-27 by kevinjwalters

dhalbert, 2021-02-27:
Great! Winows can do odd things when it detects device errors. It has some heuristics about which drivers to install when based on previous errors but I do not find them obvious.

jepler, 2021-03-15:
This could also have been related to #4377 and fixed for good by #4392

kevinjwalters, 2021-03-16:
I had a go with `adafruit-circuitpython-raspberry_pi_pico-en_US-20210315-05ed179.uf2` and `CIRCUITPY` vanished again. I tried a device cleanse of just my two Pi Picos and that didn't seem to help but I was tired so might have missed something.