adafruit/circuitpython #4284: Using displayio.Tilegrid.transpose_xy will crash the MAGTAG sending it to Safe Mode

jposada202020, 2021-02-27:
This was tested with the following code in a MagTag using CircuitPython 6.1.0

    import displayio
    import terminalio
    
    
    font = terminalio.FONT
    glyph= font.get_glyph(ord('A'))
    letter = displayio.TileGrid(glyph.bitmap, pixel_shader=displayio.Palette(1), default_tile=glyph.tile_index, tile_width=glyph.width, tile_height=glyph.height)
    letter.transpose_xy = True

This works fine in a wioterminal

dhalbert, 2021-02-27:
Could you try this with 6.2.0-beta.2?

FoamyGuy, 2021-02-27:
I tested this on:
```
Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Adafruit MagTag with ESP32S2
```
I'm not getting safe mode, but the device does "hard" crash and reboot.

rsbohn12, 2021-02-27:
Tested on 

    Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit Metro ESP32S2 with ESP32S2

-- hard crash and reboot.

microDev1, 2021-02-27:
I am getting a hard crash and a subsequent reset into safe-mode with the following setup :-

```
Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; microS2 with ESP32S2
```
I ran a backtrace and here are the results :-

```
/home/mdev/code/circuitpython/ports/esp32s2/../../shared-module/displayio/TileGrid.c:111
/home/mdev/code/circuitpython/ports/esp32s2/../../shared-module/displayio/TileGrid.c:307
/home/mdev/code/circuitpython/ports/esp32s2/../../shared-bindings/displayio/TileGrid.c:280
/home/mdev/code/circuitpython/ports/esp32s2/../../py/objfun.c:86
/home/mdev/code/circuitpython/ports/esp32s2/../../py/runtime.c:624
/home/mdev/code/circuitpython/ports/esp32s2/../../py/runtime.c:1169
/home/mdev/code/circuitpython/ports/esp32s2/../../py/vm.c:425
/home/mdev/code/circuitpython/ports/esp32s2/../../py/objfun.c:286
/home/mdev/code/circuitpython/ports/esp32s2/../../py/runtime.c:624
/home/mdev/code/circuitpython/ports/esp32s2/../../py/runtime.c:598
/home/mdev/code/circuitpython/ports/esp32s2/../../lib/utils/pyexec.c:114
/home/mdev/code/circuitpython/ports/esp32s2/../../lib/utils/pyexec.c:524
/home/mdev/code/circuitpython/ports/esp32s2/../../main.c:530
/home/mdev/code/circuitpython/ports/esp32s2/../../main.c:597
/home/mdev/code/circuitpython/ports/esp32s2/supervisor/port.c:330
/home/mdev/code/circuitpython/ports/esp32s2/build-microdev_micro_s2/esp-idf/../../esp-idf/components/esp32s2/cpu_start.c:453
/home/mdev/code/circuitpython/ports/esp32s2/build-microdev_micro_s2/esp-idf/../../esp-idf/components/freertos/xtensa/port.c:143
```

jposada202020, 2021-02-27:
Aditional comments

### For 6.1.0 goes to safe mode

    Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit MagTag with ESP32S2
    >>> import test_transpose
    
    [tio 12:42:33] Disconnected
    [tio 12:42:36] Connected
    Auto-reload is off.
    Running in safe mode! Not running saved code.
    
    You are in safe mode: something unanticipated happened.
    CircuitPython core code crashed hard. Whoops!
    Crash into the HardFault_Handler.
                                     Press any key to enter the REPL. Use CTRL-D to reload.

### For 6.2.0 we have this in the terminal. Hard crash and Reboot
```
    Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Adafruit MagTag with ESP32S2
    >>> import test_transpose
    [tio 12:48:38] Disconnected
    [tio 12:48:41] Connected

```

deshipu, 2021-02-27:
Does it happen when you set the width, height, tile_width and tile_height explicitly to constant numbers?

deshipu, 2021-02-27:
I think I see the problem:  
https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/TileGrid.c#L107

The ``width`` variable is defined as ``int16_t`` but the ``self->pixel_width`` it is getting assigned to is ``uint16_t``.

jposada202020, 2021-02-27:
> Does it happen when you set the width, height, tile_width and tile_height explicitly to constant numbers?

Yes, this code will produce crash and restart.

```
import displayio
import terminalio

font = terminalio.FONT
glyph= font.get_glyph(ord('A'))
letter = displayio.TileGrid(glyph.bitmap, pixel_shader=displayio.Palette(1), default_tile=33, tile_width=6, tile_height=14)
letter.transpose_xy = True
```

deshipu, 2021-02-27:
Can you try #4287 and see if it fixes it?

dhalbert, 2021-03-04:
Still crashes on 6.2.0-beta.3. Works fine on a PyPortal, so it's something specific to the MagTag, perhaps specific to `EPaperDisplay`.

microDev1, 2021-03-04:
Ran the mentioned code on microS2 and pi pico with nothing attached in both cases... crashes on the S2, works fine with the pico.

dhalbert, 2021-03-04:
Reopening as #4287 does not fix this.

dhalbert, 2021-03-04:
I put a printf  at the end of `_update_current_y()` in `TileGrid.c` on a Metro M4. The values for `self->absolute_transform` seem way off. `dx` and `dy` are multiplicative scaling factors, so a `dx` of 117 is way off. The `absolute_transform` structs are shared as pointers: the child points to the parent one, so if it got smashed, bad things would happen. The Metro M4 doesn't crash but that's just an accident. This is not port-specific. The bug still needs to be found but this is pointing toward the root cause.

```python
Adafruit CircuitPython 6.2.0-beta.3-dirty on 2021-03-04; Adafruit Metro M4 Express with samd51j19
>>> import displayio, terminalio
>>> glyph = terminalio.FONT.get_glyph(ord('A'))
>>> letter = displayio.TileGrid(glyph.bitmap, pixel_shader=displayio.Palette(1), default_tile=glyph.tile_index, tile_width=glyph.width, tile_height=glyph.height)
>>> letter.transpose_xy = True
height: 6; self->x y: 0 0
self->absolute_transform->dx dy: 117 5
self.current_area.x1 y1: -10592, 8192; x2 y2 -8954 8222
```

PR #4287 (2021-03-04): Use uint16 for temporary variables in displayio.TileGrid
Should fix #4284
commit 2a3cd5b2 (2021-03-04): Use uint16 for temporary variables in displayio.TileGrid

Should fix #4284
PR #4331 (2021-03-05): Tilegrid transpose fix
Fixes #4284.

Setting `TileGrid.transpose_xy` forces the x,y, height, and width to be recalculated using `_update_current_x()` and `_update_current_y()`. Those routines did not handle the case of `absolute_transform` being `NULL`, and used NULL as a pointer to a transform, so they got random values.

This fix guards against that. It also makes `null_transform`, an identity transform defined for use in `vectorio`, be a global. I used it in the fix.

An alternative fix would be to never use a `NULL` pointer as `absolute_transform`, and use `&null_transform` to signify the  [... cut]
closed 2021-03-05 by dhalbert