adafruit/circuitpython #4291: Template for issues

microDev1, 2021-03-01:
I think this GitHub feature would be nice to have. 
We can also create different templates for bug report, feature request etc.

tannewt, 2021-03-02:
I generally dislike the templates because often I file issues with only a sentence or two of info. (I realize this might be a bad habit.) I also am unsure that it will actually change issue filer behavior.

microDev1, 2021-03-02:
> I generally dislike the templates because often I file issues with only a sentence or two of info. 

This won't be a problem... by default its optional to use the templates.

microDev1, 2021-03-02:
By default its optional to use the template.
I would like the template to be minimal. Something like the following for bug report. 

**Firmware**
```python
Adafruit CircuitPython 6.2.0-beta.2 on 2021-03-01; Raspberry Pi Pico with rp2040
```
**Code/REPL**
```python
import busio, bitbangio
i2c = bitbangio.I2C(board.GP1, board.GP0)
```
**Behavior**
```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TimeoutError: Clock stretch too long
```
**Description** _(optional)_
- Error while using i2c... 
- Only happens when...
- might be related to [#4291]()...

**Additional Info** _(optional)_
Any other information like hardware connection, scope output etc. can be added here.

tannewt, 2021-03-02:
@microDev1 That works for me! Is it possible to add the template via a PR?

microDev1, 2021-03-03:
> Is it possible to add the template via a PR?

Yup! #4316.

ismael-miguel, 2021-03-03:
I don't know if I'm allowed to say anything here, but here it goes...

In my opinion, to the error template, it would be great to add a "Expected behaviour" section.

It might sound stupid because some issues like #4311 where the expected behaviour is to do not trigger adjacent pins.
Others, like my issue #4279, the expected behaviour is that it should throw an exception because the filesystem is readonly.

Others, like #4300, could be that the frequency should be different or similar.

Yes, this could be added to the "Addicional information", but it would probably be drowned out by other things, and possibly missed.

Obviously, this section should be optional.

tannewt, 2021-03-03:
@ismael-miguel That is great feedback! Since there is a PR (#4316), that is a better place for feedback.

closed 2021-03-04 by tannewt