adafruit/circuitpython #4303: [FeatureRequest] configparser to read ini files

bitboy85, 2021-03-02:
As it is possible to use read/write from the internal storage i guess it would be useful to have a library that handles configuration files.

https://docs.python.org/3/library/configparser.html

tannewt, 2021-03-02:
Moved to circuitpython for more visibility. Typically we use python files for config now because they can be imported.

tekktrik, 2021-11-01:
I was actually thinking of doing this as a side project, is this something that there's still interest in adding to the main library (instead of the Community Bundle where I would have added it)?

tannewt, 2021-11-01:
I think the community bundle is fine. It's got good visibility through `circup`.

deshipu, 2023-07-29:
I don't think this is a good first issue, a whole library with proper api and tests seems like something an experienced programmer should make. Should we remove the label?

dhalbert, 2024-03-08:
I think we are set with this with TOML libraries and the built-in `settings.toml` support.

closed 2024-03-08 by dhalbert