adafruit/circuitpython #4304: ESP32S2: cpu.reset_reason is wrong when resetting from button or microcontroller

Neradoc, 2021-03-02:
The ESP32S2 gives `POWER_ON` as a reset reason when pressing the reset button (I expected RESET_PIN).
`microcontroller.reset()` gets a `WATCHDOG` reset reason (expected `SOFTWARE`).

Tested on MagTag (latest) and ttgo (beta 2), printing out `microcontroller.cpu.reset_reason`.

anecdata, 2021-03-02:
Related: https://github.com/adafruit/circuitpython/issues/3858

microDev1, 2021-03-02:
- The reset button is wired to S2's chip enable pin so it literally turns off the chip. That's the reason for `POWER_ON` status.
- I am getting the expected `SOFTWARE` as reset reason when using `microcontroller.reset()`. Can you try with current main.

Neradoc, 2021-03-02:
Ah I was thinking something like that might be going on with the reset button. That's too bad.

I tried latest S3, and compiled current main on the Magtag.
`Adafruit CircuitPython 6.2.0-beta.2-230-g64192b0eb on 2021-03-01; Adafruit MagTag with ESP32S2`
I tried manually calling `microcontroller.reset()` in the REPL.
I also used this code on battery to exclude the possibility of USB messing with it.

I get `WATCHDOG` in both cases. (Well, except on first boot).

```py
import board
import displayio
import microcontroller
import terminalio
import time
from adafruit_display_text import label

display = board.DISPLAY
splash = displayio.Group(max_size=10)
display.show(splash)

text = label.Label(
    terminalio.FONT,
    scale = 3,
    color = (255,255,255),
    x = display.width // 2,
    y = display.height // 2 - 10,
    max_glyphs = 50,
    anchor_point = (0.5, 0.5),
)
splash.append(text)
print(microcontroller.cpu.reset_reason)
text.text = str(microcontroller.cpu.reset_reason)[28:]
display.refresh()

for x in range(20):
    print(microcontroller.cpu.reset_reason)
    time.sleep(1)
microcontroller.reset() # <--- should result in the screen changing to
```

microDev1, 2021-03-02:
Still unable to replicate this. Tried both in `REPL` and `code.py`. 

```python
Adafruit CircuitPython 6.2.0-beta.2-230-g64192b0eb on 2021-03-02; microS2 with ESP32S2
>>> import microcontroller
>>> microcontroller.reset()
# <--- Reset --->
>>> microcontroller.cpu.reset_reason
microcontroller.ResetReason.SOFTWARE
```

Neradoc, 2021-05-15:
@microDev1 do you use the UF2 bootloader ? It seems to be what makes the difference.
I get the same result as you when I rease_flash and install Circuitpython from the bin.

microDev1, 2021-05-15:
> It seems to be what makes the difference.

Yup! I get `WATCHDOG` with UF2 and the expected `SOFTWARE` with non-UF2 build.

dhalbert, 2021-07-19:
This should be fixed now with the latest TinyUF2. Update your TinyUF2 bootloader to at least 0.5.0, following the instructions here: https://learn.adafruit.com/adafruit-magtag/install-uf2-bootloader. Thanks @microDev1 for testing and @hathach for the TinyUF2 fix.

closed 2021-07-19 by dhalbert