adafruit/circuitpython #4305: Document minimum translation requirements

tannewt, 2021-03-02:
We've added a few translations that haven't gotten any translations actually added. To save build time, we're turning those off. As we discussed it, we thought it'd be helpful to have a minimum set of messages to translate before turning a new translation on. My idea was to have any interaction messages translated but error messages are optional/best effort.

Looks like Weblate has a `priority` feature that does what we'd like: https://docs.weblate.org/en/latest/admin/translating.html#strings-prioritization I'll mark things there but suspect I need more privileges to do so.

@hugo @bergdahl @jepler @fede2 Thoughts on this?

bergdahl, 2021-03-02:
That sounds like a really good idea.

tannewt, 2021-03-08:
Ok, here is a first pass at using priority: https://hosted.weblate.org/browse/circuitpython/main/en/?offset=1&q=priority%3A%3D200&sort_by=-priority%2Cposition&checksum=