adafruit/circuitpython #4314: BCM2711 port (for Raspberry Pi 4 support)

tannewt, 2021-03-02:
We should port CircuitPython to the Cortex-A class Raspberry Pis. The main early benefit would be easy HDMI capable displayio.

Prerequisites:

- [ ] Port TinyUSB to the BCM chips. (Should be relatively easy because it uses Synopsis USB IP.)
- [ ] Add internal flash support for the emmc peripheral of the BCM

This should get CP booting and the workflow working. Then it'll be the common steps or implementing standard hardware APIs like I2C and `displayio`.

tannewt, 2021-03-02:
DTS that describes peripherals: https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/bcm2711-rpi-cm4.dts
Peripheral datasheet: https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf
Compute module datasheet: https://datasheets.raspberrypi.org/cm4/cm4-datasheet.pdf

tannewt, 2021-03-02:
Brainstorm from an email with Avram:

> Why do you want to make CP a bootable system for Raspberry Pi?

I think it'd be neat because CP is much simpler than Linux. (It'll never be able to do everything Linux can.) This fits the nostalgia niche of the Pico MicroPython and emulator demos but easily support HDMI devices. It would take the simplicity and accessibility of the CircuitPython workflow and bring it to "the big screen". (Libraries are already available in CPython on Linux.) It'd be a lot like a C64 era experience but with a modern language (Python). It'd complement Linux. It would be a *LOT* of work to do everything Linux does (wifi, ssl, ble, image formats, video playback.) So, coming from Linux it'd feel pretty limited. Coming from CP on a Pico or other micro, it'd feel huge, fast, and power hungry.

I was exploring how you'd edit code and prototyped "BASIC Python" where you edit Python in the same way you would basic. https://github.com/tannewt/basicpython It'd also be a good platform for Eben's simple editor 49er: https://github.com/ebenupton/49er

> Would it be correct to call this an OS?

/me looks up OS on wikipedia. Yes? CircuitPython would manage hardware resources for you. I think of it as a simple, single-task OS.

> How would CP on Raspberry Pi hardware work? What would you see when it was done booting and would you be able to enter data into it directly or would you have to use a computer to write to it as you write to a micro controller?

That's a good question. :-) The simplest place to start would be by supporting the same USB workflow we have today. In that case you'd use a second computer to program it in the same way you do today. This is a bit tricky with existing Pi hardware because they are usually geared towards being a USB host, not a device. So, what Pis would be supported and on what ports would be a bit tricky.

Later on, when we have USB Host support in TinyUSB and CircuitPython, then things could be standalone. This is where I envision using your Pi400 as a modern C64 using BasicPython or 49er. You'd have a basic way of editing and running the core Python code.

A third long term option would be to support the BLE workflow (that is a higher priority than this work.) If we had BLE on one or more Pis, then you could do the same BLE workflow of editing a CP file remotely from a phone or tablet and then seeing the result near-instantly on the TV or monitor of the Pi.

> Would it require a safe shutdown or could you just pull the plug to turn it off?

I don't think so. CircuitPython does very limited file system caching so most changes will have been written out. There may be a hardware limitation I'm unaware of though (maybe SD cards themselves cache.)

> What Raspberry Pi models would this work on? Would it be limited to only those that have USB host mode (Pi Zero and 4)?

Based on my research it'd be limited to Pi models that make the on-chip USB OTG peripheral available on the outside world. (On the Pi 4 for example, the OTG USB is made available on the type C connector that also carries power. The USB A plugs are host-only and connected over PCIe.) Model As and Zeros tend to break the USB OTG out directly but may have inconvenient plugs. However, I could be wrong. It hasn't been implemented yet.

The Compute Module 4 and Pi 400 really make this idea interesting because it allows for board designs that lean into the CircuitPython model, like the Piunora, and make the OTG available on micro USB or USB C (non-power) ports.

> Do you think this will actually happen and, if so, when would you estimate?

Ya, I think this will happen. I have no idea when. CircuitPython's "first five minute" experience would benefit the Raspberry Pi hardware and make simple hardware tasks simple to do.

I can't guess when it'll be done. It is interesting enough that I think I'll get to it at some point, unless others beat me to it (hint hint.) I've been wanting to do a BLE workflow for CircuitPython for a few years now and that really is my priority.

If folks like this idea, then I'd encourage them to reach out to help make it happen. We're always looking for more CircuitPython contributors. My highest priority is always to support others working on CircuitPython so I'm happy to give guidance and get things merged into CircuitPython proper. I'd love to see CircuitPython supported directly on Pis and other single board computers.

2231puppy, 2021-03-05:
Not sure if this is super helpful or not, but the Pi 3A+ also has OTG, I think.

(Also, really REALLY cool idea! happy to test it on my pi 0 w and pi 4 once alpha builds are out)

2231puppy, 2021-03-05:
This is an AWESOME idea! Can't wait to test some alpha builds when they come out!

tannewt, 2021-03-16:
Here is the article based on the questions above: https://www.tomshardware.com/news/circuitpython-bootable-os-raspberry-pi

daveythacher, 2021-04-02:
This will be more involved than you realize.

tannewt, 2021-04-05:
> This will be more involved than you realize.
> 
> This is kind of crazy to me. I have play around with some crazy ideas involving using MPUs as MCUs, Linux, Python, etc. I know there are a few ways to make something like this work, but to be honest I am not seeing it here. Usually something like productivity, cost, etc. helps you net ahead of what you give up.

Please elaborate. How would it "not work"? How would it be "more involved"?

PR #5610 (2021-11-25): Initial broadcom port for Raspberry Pi
This targets the 64-bit CPU Raspberry Pis. The BCM2711 on the Pi 4
and the BCM2837 on the Pi 3 and Zero 2W. There are 64-bit fixes
outside of the ports directory for it.

There are a couple other cleanups that were incidental:
* Use const mcu_pin_obj_t instead of omitting the const. The structs
  themselves are const because they are in ROM.
* Use PTR <-> OBJ conversions in more places. They were found when
  mp_obj_t was set to an integer type rather than pointer.
* Optimize submodule checkout because the Pi submodules are heavy
  and unnecessary for the [... cut]
commit a1052d5f (2021-11-25): Initial broadcom port for Raspberry Pi

This targets the 64-bit CPU Raspberry Pis. The BCM2711 on the Pi 4
and the BCM2837 on the Pi 3 and Zero 2W. There are 64-bit fixes
outside of the ports directory for it.

There are a couple other cleanups that were incidental:
* Use const mcu_pin_obj_t instead of omitting the const. The structs
  themselves are const because they are in ROM.
* Use PTR <-> OBJ conversions in more places. They were found when
  mp_obj_t was set to an integer type rather than pointer.
* Optimize submodule checkout because the Pi submodules are heavy
  and unnecessary for th [... cut]
closed 2021-11-25 by dhalbert