adafruit/circuitpython #4324: ESP32-S2 `watchdog.deinit()` doesn't succeed or raise an error

tylercrumpton, 2021-03-04:
I've been trying to work with the `watchdog` module on my UnexpectedMaker FeatherS2 (esp32s2) running `6.2.0-beta.2`, and have noticed that the `watchdog.deinit()` function doesn't successfully stop the watchdog, nor does it raise an error indicating that `deinit()` is not available for this platform. My example script is:

```python
from watchdog import WatchDogMode, WatchDogTimeout
from microcontroller import watchdog as w

print("Initial mode:", w.mode)
w.timeout = 10  # Set a timeout of 10 seconds
w.mode = WatchDogMode.RAISE
w.feed()
print("Watchdog running mode:", w.mode)
w.deinit()
print("After deinit mode:", w.mode)

while True:
    try:
        while True:
            pass
    except WatchDogTimeout:
        print("The watchdog got hungry!")
```

Which results in an output like:
```
Initial mode: None
Watchdog running mode: watchdog.WatchDogMode.RAISE
After deinit mode: watchdog.WatchDogMode.RAISE
The watchdog got hungry!
The watchdog got hungry!
The watchdog got hungry!
   ... continues every 10 seconds
```

It appears that stopping the watchdog should be supported based on the CircuitPython [wdt HAL implementation for the esp32s2](https://github.com/adafruit/circuitpython/blob/main/ports/esp32s2/common-hal/watchdog/WatchDogTimer.c#L51) and the [ESP-IDF task_wdf component](https://github.com/adafruit/esp-idf/blob/ebe7784258d8c10e9cc334ccc00c3fd270746c8b/components/esp_common/src/task_wdt.c#L247).

I've tried this on both:
`Adafruit CircuitPython 6.2.0-beta.3 on 2021-03-04; FeatherS2 with ESP32S2`, and
`Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; FeatherS2 with ESP32S2`

microDev1, 2021-03-04:
@tylercrumpton Thanks! for reporting this. Can you please test the fix in the linked PR. You can find the build artifacts [here](https://github.com/adafruit/circuitpython/actions/runs/620289184).

tylercrumpton, 2021-03-04:
Done! It all works as expected now, thanks! I tried out both the `RAISE` and `RESET` modes, and they both allow me to `deinit()` and re-setup the watchdog now.

PR #4325 (2021-03-04): ESP32S2: Fix watchdog deinit
Delete subscribed task before watchdog deinit. Fixes #4324.
closed 2021-03-04 by dhalbert