adafruit/circuitpython #4326: ESP32-S2-Saola boards returning code 43 errors on OTG port

aziascreations, 2021-03-04:
Hello, I am currently trying to connect to some ESP32-S2-Saola devboards that are running CircuitPython 6.1.0 via the USB OTG pins, but the device shows up as as "Unknown USB Device (Device Descriptor Request Failed)", and I see the following error in Windows' Device Manager:
```
Windows has stopped this device because it has reported problems. (Code 43)
A request for the USB device descriptor failed.
```

<br>

The boards I tried are manufactured by Espressif and are powered by the ESP32-S2-WROOM and ESP32-S2-WROVER.

I flashed it with the following command taken from the official Adafruit website and using Espressif's tools from their GitHub repository:<br>
`python esptool.py -b esp32s2 --port COMx --after=no_reset write_flash 0x0 <firmware.bin>`

I didn't notice any error during the flashing process, the hash of the flashed firmware was verified properly, and the onboard LED is slowly flashing green which indicates that the main code file has finished executing according to CircuitPython's documentation.

The serial terminal  on the main USB port also doesn't return any error:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x4
load:0x3ffe6104,len:0x19d0
load:0x4004c000,len:0x162c
load:0x40050000,len:0x20fc
entry 0x4004c30c
```

<br>

I've tried the following things to attempt to fix the issue without any success:
* Using 15cm/8cm cables or jumpers to connect the data pins.
* Inserting the USB cable slowly into the host device to connect D+ and D- after the device booted
* Using different breakout boards
* Adding 2 10ohms resistors in series on the D+ and D- line to vaguely match the missing 22ohms resistors
* Testing the 6.2.0-beta3 release of CircuitPython.
* Using different USB cables that have the D+ and D- wires
* Checking that the USB pins from the chip go to the right pins on the booard with a multimeter and through the schematics
* Tying the ID pin to GND
* Swapping the D+ and D- pins
* Using different breadboards on none at all (See #3075)

I don't know if I'm missing something or if my boards have a problem.

tannewt, 2021-03-04:
Please post a picture of the boards you are using with the USB cable connected.

aziascreations, 2021-03-04:
Here are some pictures of the board and some of the wiring attemps I've done.

[image][image][image][image][image]

anecdata, 2021-03-04:
I verified my Saola will mount (Mac though). But take a lok at your command line, I believe `b` is `baud`: `--chip esp32s2` `--baud 115200`

aziascreations, 2021-04-01:
@anecdata Hey, sorry for the late response, you were right about the command, I didn't copy and paste it properly in this issue.
It should have been like this: `python esptool.py -c esp32s2 --port COMx --after=no_reset write_flash 0x0 <firmware.bin>`
However I think this isn't the issue here.

I've been able to get the ESP32S2 to work over the last couple of weeks by plugging it into another USB2.0 Hub that worked with it this time, and I also noticed that can also work on some USB 3.0 ports where it would refuse to work previously depending on the version I used and sometime complete randomness.

<br>

Here is the situation:

After playing around with the 6.1.0 and 6.2.0-beta3 versions through a USB 2.0 Hub, I left the device for a week or two.
None of them worked directly through the USB3.0 port.

I updated Windows from 1803 to 20H2, and when I retried it because I needed to document something, I "mistakenly" plugged it in directly into the same USB3.0 port with the same cable on the same laptop without the USB 2.0 hub and it worked directly, but only if the 6.1.0 and 6.2.0-rc0 versions were flashed onto it, not the 6.2.0-beta3.
I also tried reflashing Saola boards with an ESP32-S2-WROVER and an ESP32-S2-WROOM multiple times with the appropriate `.bin` files to make sure it wasn't just a fluke and it doesn't appear to be.

It's especially weird since the changelogs from 6.2.0-beta3 to 6.2.0-rc0 through 6.2.0-beta4 don't clearly mention any changes related to the ESP32S2's OTG USB port.

dhalbert, 2022-02-18:
@aziascreations Shall we leave this open? It's been a long time, and I wonder if you still have this problem with 7.x.

dhalbert, 2022-03-09:
Closing for now.