adafruit/circuitpython #4330: NUCLEO-F767ZI STORAGE ISSUE

johncblacker, 2021-03-04:
Using cPy 6.1.0 firmware and load to nucleo-f767zi board.  Selecting properties on cPy drive shows total size about 90K which is significantly lower than the 2M available on the board.  Flashed uPy for same board and it shows the correct amount of available storage 2M.  Looked at the STM32F767_fs.ld files in each distro and find that the uPy distro version has another line in the file: "FLASH_APP (rx) : ORIGIN = 0x0800x000, LENGTH = 2016K
I believe this line needs to be inserted into the cPy version of the file in order for the full 2M available to be used.  I tried adding that line then performing a build, but ran into an error in the build (not related to the line I added, BTW).  The build output was: 
 make BOARD=nucleo_f767zi
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
QSTR updated
../../lib/tinyusb/src/common/tusb_fifo.c: In function 'tu_fifo_write':
../../lib/tinyusb/src/common/tusb_fifo.c:149:6: error: assuming signed overflow does not occur when assuming that (X + c) < X is always false [-Werror=strict-overflow]
   if ((p > p + offset) || (p + offset > f->max_pointer_idx))
      ^
cc1: all warnings being treated as errors
../../py/mkrules.mk:55: recipe for target 'build-nucleo_f767zi/lib/tinyusb/src/common/tusb_fifo.o' failed
make: *** [build-nucleo_f767zi/lib/tinyusb/src/common/tusb_fifo.o] Error 1

So, I'm opening this bug report in order for the full 2M of storage for the board to be usable.

tannewt, 2021-03-08:
@hierophect Any idea how much work this is?

johncblacker, 2021-03-09:
Well, since I didn’t develop the code originally, no I don’t.  I did look around in the uPy github repo and see it’s quite a bit different, so I imagine there’s 

quite a bit of work to accomplish it; but, on the other hand, I strongly feel that the original implementation for these boards wasn’t done correctly.  Why

waste a good percentage of the on board memory, especially when the board doesn’t have onboard sd card socket with which to overcome the loss of

the onboard memory.  I understand if Adafruit wants to table this request/issue, but that would leave me and others towards using uPy instead of cPy.

In my opinion cPy was and is an offshoot of uPy and it has much of it’s base foundation in uPy, so why diverge in providing flash memory equal to the 

available ram?

 

Thanks for responding.

jb

hierophect, 2021-03-09:
@johncblacker hi John, the NUCLEO-F767ZI was added by a community member, so it was probably just a mistake in the setup parameters that never got caught. The big F7/H7 Nucleos don't see as much use as the Discovery boards, so this kind of setup problem can persist for a while.

@tannewt this is probably just a typo/parameter problem in either the linker or the flash settings. I can take a look after wrapping up my low power draft today. @k0d might also have ideas, I think he was the one who originally added it.

johncblacker, 2021-03-09:
FWIW, I did modify the boards .ld file and made it look more similar to the uPy version.  Basically added a single line called FLASH_APP, built the firmware onto the board but it didn’t make

any difference, so there’s more to it than that and I don’t know enough about the module structure to go any further myself.  I did find one file that defined the flash size as 96k (0x18000) but

not know what other dependencies there might be I was hesitant to change that value.  The file I looked at is in the supervisor directory, internal_flash.h wherein is has “INTERNAL_FLASH_FILESYSTEM_SIZE

set to 0x18000 (96k) but I don’t know what’d happen if that were changed to 0x200000 (2mb)…

jb

johncblacker, 2021-03-09:
Just for grins, I made a change in the internal_flash.h file for the 767zi changing the size from 0x18000 to 0x200000, rebuilt, flashed to board, checked size and it was still around 80k – so that didn’t work!

jb

k0d, 2021-03-09:
I'll look into this a bit later....

johncblacker, 2021-03-09:
That worked…now when I right mouse click on the cPy drive, I see 2MB.  Did an os.statvfs(‘/’) and here’s what I get:

>>> os.statvfs('/')

(1024, 1024, 2028, 2021, 2021, 0, 0, 0, 0, 255)

 

Looks good to me…how about you?

 

You'll also need to recreate the file system after changing the space for it. If it starts in the same place, then it will just use the existing smaller file system. You can do storage.erase_filesystem() from the REPL to recreate it.

hierophect, 2021-03-09:
The filesystem size cannot literally be 2MB, since Circuitpython still needs at least 256kB of room for the vector tables and firmware used by the python interpreter itself. So setting it to that size is probably not "real", ie you won't actually be able to write to the whole drive, and would encounter irrecoverable crashes if you did.

johncblacker, 2021-03-10:
Fair enough, but looks like my "technique" worked, at least.  Modify the .ld file, then modify the FLASH_FILESYSTEM_SIZE value in the internal_flash.h file basically worked for me, but I haven't tried writing the library bundle on top of it to see what'll happen...may crash as was written above.  If that happens, I'll back the value down some, although 384kB seems a little low in my mind.