adafruit/circuitpython #4332: Can't use a PS/2 Keyboard and the SPI bus at the same time

2231puppy, 2021-03-05:
**Firmware**

```python
Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit ItsyBitsy M4 Express with samd51g19
```

**Code/REPL**

```python
import board, busio, ps2io
kbd = ps2io.Ps2(board.A2, board.A5)
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
```

**Behavior**

```python
code.py output:
Traceback (most recent call last):
  File "code.py", line 3, in <module>
ValueError: MOSI in use

Code done running.
```

**Description**

If the `kbd` line is removed, it works fine.

tannewt, 2021-03-09:
Please try the latest beta too.

NathanY3G, 2021-11-18:
In case it helps to eliminate potential causes, I was unable to reproduce the above issue on a different M4 board running a build of 6.1.0 with the same pin mappings.

```
Adafruit CircuitPython 6.1.0 on 2021-11-18; Adafruit Grand Central M4 Express with samd51p20

>>> import board, busio, ps2io
>>> kbd = ps2io.Ps2(board.A2, board.A5)
>>> spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
>>> 
```