adafruit/circuitpython #4336: RP2040/Pi Pico NVM not working

destinysagent, 2021-03-05:
I'm trying the NVM example on: https://circuitpython.readthedocs.io/en/latest/shared-bindings/nvm/index.html?highlight=NVM

using the line: microcontroller.nvm[0:3] = b\"\xcc\x10\x00\" after importing microcontroller, both in REPL

This gives a syntax error Invalid Sybtax

[image]

boot_out.txt is: Adafruit CircuitPython 6.2.0-beta.3 on 2021-03-04; Raspberry Pi Pico with rp2040

microDev1, 2021-03-05:
This is a docs issue... can you try with `microcontroller.nvm[0:3] = b"\xcc\x10\x00"`.

destinysagent, 2021-03-05:
Yes, that appears to work fine - Many thanks microDev1

dhalbert, 2021-03-05:
Documentation has a stray `\`.

destinysagent, 2021-03-06:
Just for clarification, the document has two stray \ - b\"\xcc\x10\x00\" as opposed to the correct b"\xcc\x10\x00"

PR #4346 (2021-03-06): nvm: fix doc markup
Closes #4336
commit b23c5b2d (2021-03-06): nvm: fix doc markup

closes #4336
closed 2021-03-06 by jepler