adafruit/circuitpython #4339: Make microcontroller.cpu.frequency settable

hierophect, 2021-03-05:
Creating this so that frequency/overclocking/underclocking issues can all go to the same place, as opposed to being specific frequency requests per port.

The idea here is to make `microcontroller.cpu.frequency` adjustable at runtime to allow either overclocking for performance gains (#4176) or underclocking for power savings/noise reduction (discussion in #2284). However, this could end up being a bit tricky in practice, since system clock changes tend to cascade into the peripherals and calculations in the ports may be assuming a known clock frequency. USB clock divisors will also need to be correctly recalculated and applied.

dhalbert, 2021-03-05:
Perhaps we could do this "on next reboot", so that all the initial setup stuff is done with the new clock speed.

hierophect, 2021-03-05:
@dhalbert that would be one way to do it, but it'd also be nice to give it a way of doing this before and after sleep mode. Clocking down has massive power implications for STM32 and I assume the same goes of the other ports.

tannewt, 2021-03-09:
Also note that slower clock does not always mean less power because it takes code longer to execute. My understanding for low power is that you want to process quickly so that you can get to sleep faster.

I agree this should be enabled at some point though for overclocking when power doesn't matter. We could make it settable from boot.py only.

daveythacher, 2021-04-03:
> Also note that slower clock does not always mean less power because it takes code longer to execute. My understanding for low power is that you want to process quickly so that you can get to sleep [...]

Clock generally predicts higher power consumption. It is complicated. It depend on a number of things. For some parts this can be true. You can also do things to lower the power draw.

The difference will likely be small at these clock speeds. The most notable feature to power consumption is voltage. Frequency drives current which drives voltage. Frequency is linear for a given voltage range, but higher frequency needs more voltage. This becomes non-linear.

Again not common or massively significant at these speeds. You will mostly be in linear operation. However there are overhead currents which can cause this to sometimes workout to be correct. However there are also other notions which can cause this to not work out.

If you plot some microcontroller's current you will find that beyond a certain frequency the current power consumption is larger or equal. You will also find that below this frequency the power consumption is larger. Lower frequencies have a larger current draw than higher frequencies proportionally. However it becomes linear.

If you inspect the operating power against the idle power you spot a trend there also. The core roughly responds linearly while the idle power mostly disrupts the lower section of the plot. Lot of fancy games you can play to make it look one way versus another.

Most microcontroller do not compensate against voltage. They run on fixed rails. Some use lower power rails which allows them to lower their power consumption. 

Less time per instruction divided the average current. Resulting in lower power draw. However the power draw does rise against frequency. Once the overhead is averaged out the gains disappear. Then when the voltage is increased they are crushed even more.

The difference only really matters for large sections of code that need to run. The draw however is much more stable with lower frequency. While not as power efficient it actually could still be. The high speed mode could have high burst.

daveythacher, 2021-04-03:
What you need is a clocking manager for each port. Under the covers each module that needs a clock will register a callback to that specific clock. When you change the frequency it will compute the correct settings. Manager will then call each handler notifying it of the new clock value.

Handlers will throw exception if failure occurs for "critical" clocks. Others can fail silently as they are best effort. USB for example is critical clock, while SPI is not. This is a poor example, because USB will likely have dedicated clock tree independent of all others. However the ability to support "critical" clocks could be supported.

If exception is found it will revert configuration back. This failure will be reported back. Internally the correct clock switch procedures will need to be honored. Operations currently pending like DMA/interrupts/SPI/etc. will need to be stalled or wait for completion. These operations could be registered with the clock manager.

This will allow devices like SPI to recalculate the clock dividers to match the configured speed. Some clocks may need to specify a min and max/exact clock for that clock tree. The internal logic manually allocate which clock tree it runs from to avoid conflict. The device will only register with the clock tree it needs. Should the manager fail to find a solution it will except out.

Note this is a heavy handed approach. It is slow and ugly. However it will keep the details away from the user. It will mean more work for the IO modules implemented in C. Rebooting is an option but that will not work for certain applications.

One of the STM32 processors I saw would appear to justify lowering the clock speed. However I am not sure if there is full power management support besides lowering clocks and sleep. Additional stuff like shutdown sections of the chip may not be supported. In reality this probably would be better done with bare metal.

I think there is an argument for STM32 to support this. It is possible, however it would involve a lot of work. This could cause some confusion. You are trying to hide certain things, and the correlation will not be apparent. However it would be friendly at the same time.

Lots of places for this to go wrong. However it is decently centralized and decentralized.

Basically three handlers are need to register with manager. Note this does assume it can block the entire system. I do not know what that means for ESP32.

dewalex, 2021-04-03:
Would it make more sense to first create a settable clock speed for "on next reboot", and then progress to what you're mentioning above?  Many of the applications for lowering the clock speed will not need to be done on the fly.  For instance, what sent me down this rabbit hole was that the stm32f405 jams gps antennas while running at 168mhz.

elijahsgh, 2021-06-06:
This is a couple of months old but I found this over in micropython land:
https://github.com/micropython/micropython/commit/11cf742524202fa5fc062f3e6e3040a82f49b190

Which is pretty much the same topic.

Were there any updates for this feature in CircuitPython?

hierophect, 2021-06-07:
@elijahsgh there's a fair amount of backend work behind the scenes that would be required for this to function across ports, so it's not a very high priority for the core team. If someone would like to suggest an API implementation for it for a single port (to fit a particular application they need it for, like the GPS case on STM32), we could put a not-implemented on the others as a holdover.

tannewt, 2021-06-07:
> This is a couple of months old but I found this over in micropython land:
> [micropython@11cf742](https://github.com/micropython/micropython/commit/11cf742524202fa5fc062f3e6e3040a82f49b190)
> 
> Which is pretty much the same topic.
> 
> Were there any updates for this feature in CircuitPython?

This MP change has no guard rails. Any existing clock setups will inadvertently change.

In CircuitPython, we'd want to make sure and dependent clock and their divisors are update. OR we'd want to limit it to configurable at start up in boot.py.

dewalex, 2021-06-08:
@hierophect Are you saying to raise an issue and ask for the clock speed change to be implemented just for the STM32?  It would be awesome to get a solution for that, my gps project is done and just sitting on the shelf waiting for that final piece of the puzzle.  

@tannewt Would making it configurable in boot.py be a good first step to later implementing the option to change in the main loop?

elijahsgh, 2021-06-09:
> It would be awesome to get a solution for that, my gps project is done and just sitting on the shelf waiting for that final piece of the puzzle.

If you've got hardware waiting on a fix I can share what I did for the rp2040, @dewalex :
 Add to raspberrypi/Makefile under SRC_SDK:
        src/rp2_common/pico_stdlib/stdlib.c \

Add to boards/arduino_nano_rp2040_connect/board.c:
void board_init(void) {
    set_sys_clock_khz(150000000/1000, false);
}

Build:
make BOARD=arduino_nano_rp2040_connect

Reflash :)

In micropython you can change this "on the fly" arbitrarily and I haven't had any trouble (so far :crossed_fingers:) when it comes to i2c/spi/etc. YMMV. Good luck.

tannewt, 2021-06-09:
> @tannewt Would making it configurable in boot.py be a good first step to later implementing the option to change in the main loop?

Yup! That'd be a good start.

dewalex, 2021-06-17:
@elijahsgh Thanks for sending that!  I think I would need to read up more about making a custom build before trying that, so may not be trying that soon.  

@tannewt Where do you think that is on the priority list for the team?

tannewt, 2021-06-22:
> @tannewt Where do you think that is on the priority list for the team?

It's not currently a priority for Adafruit funded folks. As always, we're happy to review PRs and give guidance.

dewalex, 2021-06-22:
@tannewt I understand that.  If someone could, however, just make a solution for the stm32f405 feather to change at boot.py, that would allow for your products (feather gps + feather stm32) to be compatible.  I view that as different from working on changing all of the boards' capabilities just for the feature. I realize there's a lot going on and that still may not be a priority though.  

I know next to no C, but if I manage to solve anything on my own, I'll submit a PR.

dhalbert, 2021-06-22:
> what sent me down this rabbit hole was that the stm32f405 jams gps antennas while running at 168[MHz].

What clock speed does not interfere with GPS? Implementing a general clock-speed change is a big hammer if we could perhaps just tweak the default frequency slightly to avoid some particular harmonics. That would be another issue.

dewalex, 2021-06-22:
@dhalbert I haven't been able to test other frequencies, but I believe that the next lowest "step" is 144MHz for the STM32 for their default divisors, right?  If you have a build with that frequency, I'll be glad to do whatever testing is needed.

mjs513, 2022-04-11:
In lieu of creating another issue figured I would continue the discussion on settable clock frequency.  

On the Teensy 4.1 (4.0 and Sparkfun Micromod) we are able to set the clock frequency programmatically from with the sketch.  I recently ported that over to the Teensy 4.1 port for circuitpython but to get it to work I had to change the shared-bindings which would make it (at least my understanding visible to all ports.  

Basically I added a function to clocks.h for the T4.1 (setfrequency) and then:
1. in processor.h/cpp
```
uint32_t common_hal_mcu_processor_setSysClock(mcu_processor_obj_t *self,
    uint32_t frequency) {
	SystemCoreClock = setarmclock(frequency);
    return SystemCoreClock;
}
```
2.  In shared bindings in processor.h/cpp
```
STATIC mp_obj_t mcu_processor_setSysClock(mp_obj_t self_in, mp_obj_t freq) {
    mcu_processor_obj_t *self = MP_OBJ_TO_PTR(self_in);
	uint32_t value_of_freq = MP_OBJ_SMALL_INT_VALUE(freq);
    common_hal_mcu_processor_setSysClock(self, value_of_freq);
    return mp_const_none;
}

MP_DEFINE_CONST_FUN_OBJ_2(mcu_processor_setSysClock_obj, mcu_processor_setSysClock);

const mp_obj_property_t mcu_processor_freq_obj = {
    .base.type = &mp_type_property,
    .proxy = {(mp_obj_t)&mcu_processor_setSysClock_obj,
			  MP_ROM_NONE,
              MP_ROM_NONE},
};
...
	{ MP_ROM_QSTR(MP_QSTR_setfreq), MP_ROM_PTR(&mcu_processor_setSysClock_obj) }, [to mcu_processor_locals_dict_table]
```

I am not sure how I would do this so its only available to the Teensy ports or MIMXRT10xx.

tannewt, 2022-05-09:
Done in #6279

closed 2022-05-09 by tannewt