adafruit/circuitpython #4345: usb_cdc.serials[1].connected always return False for Raspberry Pi Pico

ismael-miguel, 2021-03-06:
**Firmware**

```python
Adafruit CircuitPython 6.2.0-beta.3 on 2021-03-04; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```python
import usb_cdc
usb_cdc.serials[1].connected
```

**Behavior**

Running the following code ALWAYS returns `False`, even when reading data to/from the Raspberry Pi Pico, on that USB CDC channel.

```python
>>> import usb_cdc
>>> usb_cdc.serials[1].connected
False
```

**Additional Info**

Despite returning `False`, the link is working properly.

I can send and receive data, without any issues.

<hr>

According to this: https://circuitpython.readthedocs.io/en/latest/shared-bindings/usb_cdc/index.html
> True if this Serial is connected to a host. (read-only)

Since it is connected to a host, should return `True`.

And no, the Raspberry Pi Pico is the only device connected to serial, and both ports belong to it:
[image]

dhalbert, 2021-03-06:
I do not see this problem with beta.3, either on Windows or Linux. I do have to be connected to `serials[1]` before it turns to `True`. In the screenshot below, I at first did not have a connection to COM18, but then when I did connect via TeraTerm, you can see it changed from `False` to `True'.

[image]

ismael-miguel, 2021-03-06:
Interesting ...

When I use C# to connect to it, it returns `False`, but I can read and send data without issues:
[image]

But when I use PuTTY, it return `True`:
[image]

dhalbert, 2021-03-06:
I wonder if the C# code you are using is closing the connection after each write or something like that. If you do a read with a timeout in C# so it hangs while you check, does it show as connected during that time?

ismael-miguel, 2021-03-06:
The C# code uses the SerialPort.DataReceived event, and waits for a text line.
(https://docs.microsoft.com/en-us/dotnet/api/system.io.ports.serialport.datareceived?view=netframework-4.7.2)

It opens the connection and waits.
There's nothing written to close the connection.

<hr>

I've tried to stop on the `serial.ReadLine();` method, in C#:
[image]

But it still returns `False`:
[image]

hathach, 2021-03-07:
You need to set line state DTR to 1, it is used by tinyusb to detect terminal connection

dhalbert, 2021-03-07:
Thanks @hathach. Some websearching shows this is a convention, and that terminal programs typically assert it when opening the conenction. So does `pyserial`, by default. This does not mean that DTR/DSR-style flow control is enabled. There is no other easy way to determine that something is connected on the host end. I can add more docmentation to `.connected` to make this clear.

Some more websearching shows that other people have had this problem with C#, which defaults to DTR not being set.

ismael-miguel, 2021-03-07:
Thank you @hathach and @dhalbert.

I can confirm that setting `serial.DtrEnable = true;` in C# solved the issue:
[image]

Would be amazing if this was documented, to avoid possible duplicated issues.

PR #4359 (2021-03-08): Add caveat for usb_cdc.Serial.connected; fix another doc bug
Fixes #4345.
Fixes #4352.

These are documentation changes only.
closed 2021-03-08 by jepler