adafruit/circuitpython #4349: Feather RP2040 crashes when copying files to it (CIRCUITPY not full)

anecdata, 2021-03-07:
**Firmware**

```python
Adafruit CircuitPython 6.2.0-beta.3 on 2021-03-04; Adafruit Feather RP2040 with rp2040

No additional hardware. Feather is connected to host computer via USB.
```

**Code/REPL**

```python
# no code needed to replicate
```

**Behavior**

```python
# no code needed to replicate; just copy libs or fonts until it crashes.
```

**Description**

Feather RP2040 should have 7MB available for CIRCUITPY. When copying a group of files from a Mac (macOS 11.2.1), the Feather will crash, eject, and serial will be lost. Sometimes an error dialog will appear on the Mac:
"The operation can’t be completed because an unexpected error occurred (error code 100006)."
(there are indications this could be related to device full)

The CIRCUITPY drive capacity doesn't seem to be getting reached. When this happens, there should be several MB still fee.

**Additional Info**

To recover, RESET isn't enough, nor is re-flashing with CircuitPython. The board flash must be nuked, then CircrcuitPythoin must be reloaded.

Possibly related to https://github.com/adafruit/circuitpython/issues/4279

hathach, 2021-03-09:
I have just order my feather rp2040, may take several days for it to delivered. Meanwhile, does it always happen every time or occasionally. Also it may be not important, but if possible, could you zip your group of files and uploaded it here. That's way we could also try to mimic the setup as closed as possible.

anecdata, 2021-03-09:
It may be something besides flash, or in addition to flash. I've seen, and on Discord @caternuson mentioned seeing this behavior without file copy activity.

It could be a complicating factor that I'm using the Pico flash nuke UF2 to get the device back. I have not seen that one exists for the Feather RP2040. I don't know of any other way to recover the board when this happens. The Pico version seems to work, and afterward allows power cycle, hard reset, re-entering the bootloader to install CircuitPython, and ultimately a CIRCUITPY drive which works until some action disrupts it again.

Copying the current libraries (0309) from the host computer to the `CIRCUITPY/lib/` folder will consistently trigger ejection of CIRCUITPY and loss of serial for me. Finder reports CIRCUITPY has 7.3MB capacity. The entire library bundle is about 2MB. It varies where in the copy it will fail, but the message is odd, usually something like:

> You can’t copy the item “esp32.mpy” because its name is too long or includes characters that are invalid on the destination volume.

I am seeing cases now where a hard reset will bring back CIRCUITPY and serial, so there are a range of triggers and behaviors, not sure if it's all the same issue. It could be just this board has a flash issue. I only have the one to test. I'll try to narrow it down.

caternuson, 2021-03-09:
The behavior I saw is very simple and 100% repeatable:

- Hold `BOOTSEL`, press `RESET`, get `RPI-RP2` folder
- Copy `adafruit-circuitpython-adafruit_feather_rp2040-en_US-6.2.0-beta.3.uf2` to `RPI-RP2` folder
- Board self rests when copy is complete, `CIRCUITPY` folder shows up, everything works as expected - can access REPL, etc.
- Can create a `code.py` and soft reset via <CTRL><D> at REPL and it works as many times as you want.
- Press `RESET` button, the `CIRCUITPY` folder goes way and never comes back. Any `code.py` also does not run.

There is not even any `dmesg` traffic generated in the last step. It shows the USB disconnect, and then nothing.

jepler, 2021-03-09:
I am having trouble copying "large-ish" files with `Adafruit CircuitPython 6.2.0-beta.3-75-g7d92f5aa5 on 2021-03-09; Adafruit Feather RP2040 with rp2040`.  In my case, I wanted to copy a ~5MB mp3 file for testing.

<details>
<summary>
Kernel messages (click to open)
</summary>

```
[701219.528873] usb 1-2.4.2: new full-speed USB device number 29 using xhci_hcd
[701219.888376] usb 1-2.4.2: New USB device found, idVendor=239a, idProduct=80f2, bcdDevice= 1.00
[701219.888380] usb 1-2.4.2: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[701219.888382] usb 1-2.4.2: Product: Feather RP2040
[701219.888384] usb 1-2.4.2: Manufacturer: Adafruit
[701219.888385] usb 1-2.4.2: SerialNumber: 5454730383505904
[701219.967470] cdc_acm 1-2.4.2:1.0: ttyACM2: USB ACM device
[701219.985454] cdc_acm 1-2.4.2:1.2: ttyACM3: USB ACM device
[701220.003450] usb-storage 1-2.4.2:1.4: USB Mass Storage device detected
[701220.003601] scsi host20: usb-storage 1-2.4.2:1.4
[701220.031829] input: Adafruit Feather RP2040 Consumer Control as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-2/1-2.4/1-2.4.2/1-2.4.2:1.5/0003:239A:80F2.004D/input/input295
[701220.089036] input: Adafruit Feather RP2040 as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-2/1-2.4/1-2.4.2/1-2.4.2:1.5/0003:239A:80F2.004D/input/input296
[701220.089146] input: Adafruit Feather RP2040 Keyboard as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-2/1-2.4/1-2.4.2/1-2.4.2:1.5/0003:239A:80F2.004D/input/input297
[701220.089321] input: Adafruit Feather RP2040 Mouse as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-2/1-2.4/1-2.4.2/1-2.4.2:1.5/0003:239A:80F2.004D/input/input298
[701220.089421] hid-generic 0003:239A:80F2.004D: input,hidraw3: USB HID v1.11 Gamepad [Adafruit Feather RP2040] on usb-0000:02:00.0-2.4.2/input5
[701221.026978] scsi host20: scsi scan: INQUIRY result too short (5), using 36
[701221.026985] scsi 20:0:0:0: Direct-Access     Adafruit Feather RP2040   1.0  PQ: 0 ANSI: 2
[701221.027325] sd 20:0:0:0: Attached scsi generic sg2 type 0
[701221.027684] sd 20:0:0:0: [sdn] 14337 512-byte logical blocks: (7.34 MB/7.00 MiB)
[701221.027860] sd 20:0:0:0: [sdn] Write Protect is off
[701221.027861] sd 20:0:0:0: [sdn] Mode Sense: 03 00 00 00
[701221.028001] sd 20:0:0:0: [sdn] No Caching mode page found
[701221.028003] sd 20:0:0:0: [sdn] Assuming drive cache: write through
[701221.063478]  sdn: sdn1
[701221.065222] sd 20:0:0:0: [sdn] Attached SCSI removable disk
[701221.415556] FAT-fs (sdn1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[701409.790900] cdc_acm 1-2.4.2:1.0: failed to set dtr/rts
[701426.018595] usb 1-2.4.2: reset full-speed USB device number 29 using xhci_hcd
[701426.126950] usb 1-2.4.2: device descriptor read/64, error -32
[701426.342945] usb 1-2.4.2: device descriptor read/64, error -32
[701426.546951] usb 1-2.4.2: reset full-speed USB device number 29 using xhci_hcd
[701426.662940] usb 1-2.4.2: device descriptor read/64, error -32
[701426.882953] usb 1-2.4.2: device descriptor read/64, error -32
[701427.086959] usb 1-2.4.2: reset full-speed USB device number 29 using xhci_hcd
[701427.114561] usb 1-2.4.2: Device not responding to setup address.
[701427.350596] usb 1-2.4.2: Device not responding to setup address.
[701427.562908] usb 1-2.4.2: device not accepting address 29, error -71
[701427.654908] usb 1-2.4.2: reset full-speed USB device number 29 using xhci_hcd
[701427.682643] usb 1-2.4.2: Device not responding to setup address.
[701427.918674] usb 1-2.4.2: Device not responding to setup address.
[701428.126912] usb 1-2.4.2: device not accepting address 29, error -71
[701428.139149] usb 1-2.4.2: USB disconnect, device number 29
[701428.146913] blk_update_request: I/O error, dev sdn, sector 1933 op 0x1:(WRITE) flags 0x4800 phys_seg 30 prio class 0
[701428.146980] blk_update_request: I/O error, dev sdn, sector 2173 op 0x1:(WRITE) flags 0x4800 phys_seg 30 prio class 0
[701428.146991] blk_update_request: I/O error, dev sdn, sector 4 op 0x1:(WRITE) flags 0x100000 phys_seg 23 prio class 0
[701428.146995] Buffer I/O error on dev sdn1, logical block 3, lost async page write
[701428.146997] Buffer I/O error on dev sdn1, logical block 4, lost async page write
[701428.146999] Buffer I/O error on dev sdn1, logical block 5, lost async page write
[701428.147001] Buffer I/O error on dev sdn1, logical block 6, lost async page write
[701428.147002] Buffer I/O error on dev sdn1, logical block 7, lost async page write
[701428.147005] Buffer I/O error on dev sdn1, logical block 8, lost async page write
[701428.147006] Buffer I/O error on dev sdn1, logical block 9, lost async page write
[701428.147007] Buffer I/O error on dev sdn1, logical block 10, lost async page write
[701428.147009] Buffer I/O error on dev sdn1, logical block 11, lost async page write
[701428.147010] Buffer I/O error on dev sdn1, logical block 12, lost async page write
[701428.147016] blk_update_request: I/O error, dev sdn, sector 2413 op 0x1:(WRITE) flags 0x4800 phys_seg 30 prio class 0
[701428.147037] blk_update_request: I/O error, dev sdn, sector 2653 op 0x1:(WRITE) flags 0x4800 phys_seg 30 prio class 0
[701428.147057] blk_update_request: I/O error, dev sdn, sector 31 op 0x1:(WRITE) flags 0x100000 phys_seg 1 prio class 0
[701428.147070] blk_update_request: I/O error, dev sdn, sector 2893 op 0x1:(WRITE) flags 0x4800 phys_seg 30 prio class 0
[... cut ...]
In some previous cases it was reported that being at the repl (instead of in a situation where auto-reload would take place) improved flaky USB copying, but I did not find that this workaround helped me in this case.

Note that `storage.erase_filesystem()` is broken, again, which may be fixed by #4371. It is _expected_ that re-flashing circuitpython wouldn't format the filesystem, but `erase_filesystem` should. Until that's fixed, yes, you need to use "nuke.uf2" to clean up the mess, followed by a re-flash of circuitpython.

hathach, 2021-03-09:
@caternuson your issue is serious, but a bit different, probably another issue I guess. This one is triggered when copying large file. @jepler can you reproduce the issue with pico with smaller file ? I only have pico for now to test with.

tannewt, 2021-03-09:
Note that beta.3 has an old TinyUSB commit. Please try the absolute newest builds and confirm here that you still see issues. Otherwise, it's an issue that was already fixed.

caternuson, 2021-03-09:
FWIW, I get no change in behavior.
```
Adafruit CircuitPython 6.2.0-beta.3-75-g7d92f5aa5 on 2021-03-09; Adafruit Feather RP2040 with rp2040
>>>
```

dhalbert, 2021-03-09:
@caternuson try 709cee0 builds or later after #4371, now merged, finishes building.

caternuson, 2021-03-09:
@dhalbert Thanks. Not working in a different way. Should I start a new thread? I feel like I'm hijacking this one.

anecdata, 2021-03-09:
Large copies still fail with
`2021-03-09T21:52:17.000Z 1.1 MB adafruit-circuitpython-adafruit_feather_rp2040-en_US-20210309-2c3002b.uf2`
(which is an hour newer than https://github.com/adafruit/circuitpython/commit/709cee0aafd3dee493765e7ddc12c08cb75dea19)

bhilimon, 2021-03-10:
> 
> 
> @dhalbert Thanks. Not working in a different way. Should I start a new thread? I feel like I'm hijacking this one.

Support sent me here. I'm also having this same issue. I've posted more info in this forum thread the last few days. I keep trying newer code but still no luck.

https://forums.adafruit.com/viewforum.php?f=57

dhalbert, 2021-03-10:
For the Pico, `storage.erase_filesystem()` did not work in 6.2.0-beta.3 nor in `adafruit-circuitpython-raspberry_pi_pico-en_US-20210309-d650849.uf2`. I also used the `flash_nuke.uf2`, reloaded CPy, and then created an empty new file in the filesytem that did not cause any trouble. Again, `storage.erase_filesystem()` did not work with either version, and the board did not reboot to CPy after losing the connection. Unplugging and replugging showed the new file was still there.

By comparison, `adafruit-circuitpython-adafruit_feather_rp2040-en_US-20210309-d650849.uf2` did erase itself with `storage.erase_filesystem()`, but `adafruit-circuitpython-adafruit_feather_rp2040-6.3.0-beta.2.uf2` did not work. I tested each a couple of times to make sure.

I am at a loss to explain the differences between the boards. @tannewt do you know of some difference in the treatment of the flash?

tannewt, 2021-03-10:
Ya, the pico has different flash init code than adafruit feather does currently. It gets put in a "continuous read" state that may need to be exited. Let's retire this issue in favor of more specific issues. This one is a bit of a hodgepodge.

anecdata, 2021-03-12:
Large copies behaving as expected now. Fixed by #4377 closing.

closed 2021-03-12 by anecdata