adafruit/circuitpython #4353: STM32F411CE Blackpill with Flash size

jerryneedell, 2021-03-07:
The mpconfigboard.mk file for the Blackpill species  different Flash device that the one linked to  on the product page. the link is for the GD256Q16C https://www.adafruit.com/product/4763

I put one of these on the boards and naively just created a build with from the tip of main.
Not surprisingly, the board boots to safe mode and the OS is unhappy with the new device....

I changed the mpconfigboard.mk to specify the GD256Q16C
```
SPI_FLASH_FILESYSTEM = 1
EXTERNAL_FLASH_DEVICE_COUNT = 1 
EXTERNAL_FLASH_DEVICES = GD256Q16C
```

and when rebuilt all apears to be well. I see a 2Mbyte CIRCUITPY Drive
and I and able to read and write from/to it.

This really is not a "bug" in that I assume it would work if the default FLASH device were used. 
Should the Product page be changed to link to the larger flash or should the default Flash be changed to use the 2Mbyte part.

jerryneedell, 2021-03-07:
I also verified that just adding the GD25Q16C to the list works as well.
```
diff --git a/ports/stm/boards/stm32f411ce_blackpill_with_flash/mpconfigboard.mk b/ports/stm/boards/stm32f411ce_blackpill_with_flash/mpconfigboard.mk
index 9a95a3539..166b56b6e 100644
--- a/ports/stm/boards/stm32f411ce_blackpill_with_flash/mpconfigboard.mk
+++ b/ports/stm/boards/stm32f411ce_blackpill_with_flash/mpconfigboard.mk
@@ -4,9 +4,9 @@ USB_PRODUCT = "stm32f411ce blackpill with flash"
 USB_MANUFACTURER = "Unknown"
 
 SPI_FLASH_FILESYSTEM = 1
-EXTERNAL_FLASH_DEVICE_COUNT = 2
+EXTERNAL_FLASH_DEVICE_COUNT = 3
 #See supervisor/shared/external_flash/devices.h for options
-EXTERNAL_FLASH_DEVICES = W25Q64FV,W25Q64JV_IQ
+EXTERNAL_FLASH_DEVICES = GD25Q16C,W25Q64FV,W25Q64JV_IQ
 LONGINT_IMPL = MPZ
 
 INTERNAL_FLASH_FILESYSTEM = 0

```
This is covered in the guide to building CP https://learn.adafruit.com/building-circuitpython/choosing-a-different-spi-flash-chip  so creating a custom build works fine.  Perhaps just a note on the Product page would be helpful but I'll close this as it is not really a CP bug.

jepler, 2021-03-07:
FWIW, I think we _should_ fix the port so that when the user solders the flash chip Adafruit proposes, CP works.

closed 2021-03-07 by jepler