adafruit/circuitpython #4361: 6.2.0 beta 4 release notes

dhalbert, 2021-03-08:
This release, **6.2.0-beta.4**, is the fifth beta release of CircuitPython **6.2.0**. [[[ADD HIGHLIGHTS]]]. See **Port status** below for details on port stability for each port, and **Known issues** for known problems. If you find any issues with this release, please [file an issue](https://github.com/adafruit/circuitpython/issues/new).

## Download from circuitpython.org
Firmware downloads are available from the  [downloads page](https://circuitpython.org/downloads)  on [circuitpython.org](https://circuitpython.org). The site makes it easy to select the correct file and language for your board.

## Installation
To install follow the instructions in our new [Welcome to CircuitPython!](https://learn.adafruit.com/welcome-to-circuitpython/installing-circuitpython) guide. To install the latest libraries, see [this page](https://learn.adafruit.com/welcome-to-circuitpython/circuitpython-libraries) in that guide.

Try [the latest version of the Mu editor](https://learn.adafruit.com/welcome-to-circuitpython/installing-mu-editor) for creating and editing your CircuitPython programs and for easy access to the CircuitPython serial connection (the REPL).

## Port status
CircuitPython has a number of "ports" that are the core implementation for a variety of microcontroller families. Stability varies on a per-port basis. As of this release, `atmel-samd`, `cxd56` (Spresense),  `nrf`, and `stm` for the F4 family are **stable**. `esp32s2`, `raspberrypi`, and `stm` for other STM chip families are being actively improved but may be missing functionality and have bugs. `litex` and `mimxrt10xx` are in an alpha state and *will* have bugs and missing functionality.

## Changes since 6.2.0-beta.3

### Fixes and enhancements
- Save ~1500 bytes of firmware space via slight changes in vm execution with a minor speed penalty. #4344. Thanks @jepler.
- Refactor displayio set_pixel for use in bitmap_tools. #4357. Thanks @kmatch98.
- Update TinyUSB. #4327. Thanks @hathach.
- Fix incorrect 16-bit signedness in `displayio.TileGrid`. #4287. Thanks @deshipu.
- Fix `TileGrid.transpose_xy` crash. #4331. Thanks @dhalbert.
- Code and documentation cleanups. #4335. #4337, #4338. Thanks @tyomitch.

### Board- and port-specific changes
- **ESP32-S2**
  - Fix watchdog `deinit()`. #4325. Thanks @microDev1.
- **RP2040**
  - Fix use of `tud_disconnect()`. #4327. Thanks @hathach.
- **SAMD21**
- **Spresense**
- **STM32 BlackPill**
  - Add GD25Q16C as a possible flash chip. #4356. Thanks @jerryneedell.
- **Teensy 4.0 and 4.1**
  - Enable EVE. #4342. Thanks @jamesbowman.

### Build and infrastructure changes
- Shrink some builds, using optimization flags. #4333. Thanks @dhalbert.
- Avoid secondary build failure when a test fails. #4340. Thanks @jepler and thanks @microDev1 for the original inspiration PR. 

### Documentation
- Fix typos in `microcontroller.nvm` documentation. #4346. Thanks #jepler.
- Translation additions and improvements. Thanks:
  - @bergdahl (Swedish) xxx
  - @rodrigoargumedo (Spanish) xxx
  - @jposada202020 (Spanish)
  - @wtuemura (Portuguese - Brazil) xxx
  - @MisterD66 (German) xxx
  - @hugodahl (French) xxx
  - @dglaude (French) xxxx
  - @hexthat (Chinese - Pinyin) xxx
  - Luca De Filippo (Italian) xxx

## New boards since 6.2.0-beta.3

Full commit log is [here](https://github.com/adafruit/circuitpython/compare/6.2.0-beta.3..6.2.0-beta.4).

## Breaking changes since 5.x
- `i2cslave` is now `i2cperipheral` and the class in it is changed as well.
- The `stop` kwarg has been removed from `I2C.writeto()`. If no stop is desired, then use `writeto_then_readfrom`.
- The default speed of busio.I2C and board.I2C is now 100khz, not 400khz as before. Use `busio.I2C` to set the speed explicitly.
- `_bleio.ConnectionError` has been removed. Code will now raise the native `ConnectionError` instead.

## Known issues
- ESP32-S2: Crash when repeatedly creating and destroying busio.I2C object on ESP32-S2. #3846.
- RP2040: A few I2C devices do not work on the RP2040 with `busio.I2C`. Use `bitbangio.I2C` instead, and use `bitbangio.I2C(..., probe=False)` if necessary. See #4082 and #4235.
- See https://github.com/adafruit/circuitpython/issues for other issues.

## Thanks
Thank you to all who used, tested, and [contributed since 6.2.0-beta.3](https://github.com/adafruit/circuitpython/graphs/contributors?from=2021-03-03&to=2021-XX-XX&type=c), including ______ and many others on GitHub and Discord. Join us on the [Discord chat](https://adafru.it/discord) to collaborate.

## Documentation
Documentation is available [in readthedocs.io](https://circuitpython.readthedocs.io/en/latest/).

This release is based on [MicroPython 1.9.4 @25ae98f](https://github.com/micropython/micropython/commit/25ae98f07cb3c4488cb955403dfe56b8bb8db6f0). Support upstream MicroPython by purchasing [a PyBoard](https://store.micropython.org/#/store) (from Adafruit [here](https://www.adafruit.com/product/2390)) or [sponsoring MicroPython on GitHub](https://github.com/sponsors/micropython).

## Translations
One important feature of CircuitPython is translated control and error messages. With the help of fellow open source project [Weblate](https://weblate.org/), we're making it even easier to add or improve translations. Sign in with an existing account such as Github, Google or Facebook and [start contributing through a simple web interface](https://hosted.weblate.org/engage/circuitpython/). No forks or pull requests needed!

## Troubleshooting
Check out [this guide](https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting) for info on common problems with CircuitPython. If you are still having issues, then post to the [Adafruit Support Forums](https://forums.adafruit.com/viewforum.php?f=60) and join [Discord](https://adafru.it/discord).

## Assets
_Builds are no longer stored as assets on this release page, because there are too many of them. Please see the **Download from circuitpython.org** section above._

dhalbert, 2021-03-08:
(Explanation - after accidentally pressing "Publish Release" instead of "Save Draft", I am going to keep draft release notes in an issue, so that can't happen again. Two other advantages are that the editing interface has a menu bar, unlike the release notes editor page, and it's public. I'll create an issue for each upcoming release and then close it on release.)

microDev1, 2021-03-08:
> I'll create an issue for each upcoming release and then close it on release.

This is a good change. 👍
Should these upcoming release notes be pinned?

dhalbert, 2021-03-08:
Good idea - pinned.

ajs256, 2021-03-11:
Suggestion: The last release to include builds as release assets was for 4.x, maybe we don't need the explanation that assets aren't on the release page anymore.

ajs256, 2021-03-12:
Now that there are issue templates, the "file an issue link" should link to [`/issues/new/choose`](https://github.com/adafruit/circuitpython/issues/new/choose). The current link opens a blank issue, instead of the template list.

dhalbert, 2021-03-12:
> Now that there are issue templates, the "file an issue link" should link to [`/issues/new/choose`](https://github.com/adafruit/circuitpython/issues/new/choose). The current link opens a blank issue, instead of the template list.

Fixed -thanks!

dhalbert, 2021-03-18:
6.2.0-beta.4 released.

closed 2021-03-18 by dhalbert