adafruit/circuitpython #4363: Add ESP32-S3 support

tannewt, 2021-03-08:
We should evolve the ESP32-S2 port to support ESP32-S3 as well. @unexpectedmaker has started looking at this.

UnexpectedMaker, 2021-03-08:
It would be super helpful if we could move to IDF4.3 as the base for all of the esp32 sooner than later as that is the minimum I need to work on the S3 integration.

tannewt, 2021-03-08:
There is a pending PR for 4.3 here: https://github.com/adafruit/circuitpython/pull/4195

askpatrickw, 2021-03-09:
I had a great chat with @UnexpectedMaker about this update to the IDF, he's going to post more detailed notes, but a couple of things:

1.The paths updates I provided (now ~ a month old) are incorrect in many instances, wherever possible the path should have an esp32s2 folder in the path.
2. In order to support S3, ESP32 as well as the S2, from the same port, the build system will need to have a way to indicate the target board. @tannewt should advise on that, and we should probably break that out as a separate issue. q: Maybe this can go in the board definition?
3. When the second item is worked on, the conditional paths should also be fixed so either the esp32s2 folder is conditionally esp32s3, or esp32 or that there are conditional sections for each target micro-controller's includes

Seon is going to provide a sample of the path changes and if my day allows, I'll give a swing at those changes tomorrow.

@microDev1 @skieast, I'm also happy to bow out if someone else has time and would like to step-in.

Keep in mind, this also fixes building on the my M1 Mac Mini...

microDev1, 2021-03-09:
Yes, we can have the board definitions contain the `IDF_TARGET` definition. 
I think we should have something similar to what other ports like [`stm`](https://github.com/adafruit/circuitpython/tree/main/ports/stm) have for building different micros which have shared apis.
The nice thing here is that apis are pretty similar for `esp32`, `S2`, `S3` and the `C3` as well.

microDev1, 2021-03-09:
> Do you mean how $(MCU_SERIES_LOWER) is used in the include paths (and other places)?

Yup... so something like the following in Makefile:

```python
INC += -isystem esp-idf/components/esp32s2/include
```

```python
INC += -isystem esp-idf/components/$(IDF_TARGET)/include
```

askpatrickw, 2021-03-09:
I'm just curious if I understand this..

So if we added `IDF_TARGET = esp32s2` to each `boards/<boardname>/mpconfigboard.mk`
we could then use the INC as you show above anywhere there is a mcu specific include and then the S3 or any other idf support mcu could be supported.

There was [this thread about INC paths in the PR](https://github.com/adafruit/circuitpython/pull/4195#issuecomment-780173306) which I still think is relevant. A standar way the includes are handled in this port will now need to be chosen as well (I think)

tannewt, 2021-03-09:
> I'm just curious if I understand this..
> 
> So if we added `IDF_TARGET = esp32s2` to each `boards/<boardname>/mpconfigboard.mk`
> we could then use the INC as you show above anywhere there is a mcu specific include and then the S3 or any other idf support mcu could be supported.
> 
> There was [this thread about INC paths in the PR](https://github.com/adafruit/circuitpython/pull/4195#issuecomment-780173306) which I still think is relevant. A standard way the includes are handled in this port will now need to be chosen as well (I think)

This sounds good to me. You can use it in the long include paths I prefer too. Though I'm open to changing how we do includes for ESP.

askpatrickw, 2021-03-09:
Please see the PR #4195  I made some of these changes to head us down this path.

@tannewt if you show me an example of how to specify a variable in an include, I'm willing to update all the includes to the long paths an remove the length INC section.

UnexpectedMaker, 2021-03-09:
Great stuff @askpatrickw - I was going to mention you need to remove the ..../include/... stuff now, but you did that already!

askpatrickw, 2021-06-30:
@unexpectedmaker, ESP-IDF4.3 was merged so you should be clear to look at this again.

UnexpectedMaker, 2021-06-30:
> @UnexpectedMaker, ESP-IDF4.3 was merged so you should be clear to look at this again.

I need to wait fore the next S3 silicon to get into my hands... the latest stuff Espressif are working on for S3 is in IDF4.4 and they have not back ported everything to 4.3 and there was a breaking firmware change that makes my S3 dev kit board incompatible :(

I have the initial beta silicon, they are on beta 3 internally but not releasing that and so whatever the next one is, is what I am waiting for.

I'm in the S3 silicon queue - no ETA yet, but I believe it's close.

closed 2021-12-03 by dhalbert