adafruit/circuitpython #4364: multiple serial ports mounted from usb_cdc breaks REPL auto connect scripts (6.2.0 beta 3)

maholli, 2021-03-09:
Awesome new `usb_cdc` capability creates multiple serial ports when connecting to PC/Mac/Linux machine. The REPL port number is not consistently highest/lowest number which causes issues for scripts written to auto-connect to CP boards. @dhalbert suggested this issue be created:

Some of the scripts my students and I use:

Windows Batch:
```batch
@echo off
python -c "import serial.tools.list_ports as ls; p=ls.comports(); [print(i[0]) for i in p]" > Output
SET /p SERPORT=<Output
START putty.exe -serial %SERPORT% 115200
DEL Output
```

AutoHotKey
```AutoHotKey
#c::
COMPort := Object()
COMPortName := Object()
Num:=0
Loop, HKLM, HARDWARE\DEVICEMAP\SERIALCOMM\
{
    RegRead, OutputVar
    If not InStr("%OutputVar%", "COM3")
      Run "putty.exe" "-serial" "%OutputVar%" "115200"
      Exit
    COMPort%Num%:=OutputVar
    COM_result:=COMPort%Num%
    Num+=1
}
Return
```

Python one-liner
```python
python -c "import serial.tools.list_ports as ls; p=ls.comports(); [print(i[0]) for i in p]
```

dhalbert, 2021-03-09:
Thank you for being a guinea pig for this. It turns out that there is enough information provided by pyserial to figure out which port is right. You need to use the `.interface` attribute on the result of `comports()`:
```python
import serial.tools.list_ports
for port in serial.tools.list_ports.comports():
    if port.interface == "CircuitPython CDC control":
        print(port.device)
```
The key is the `port.interface`, which is the interface name. For the REPL port, it is `CircuitPython CDC control`; for the secondary port, it is `CircuitPython CDC2 control`.  Or it could be `CircuitPython CDC data` and `CircuitPython CDC2 data`. The key difference is `CDC` vs `CDC2`.

However, note that this is broken in the current `pyserial` on MacOS: it returns the same interface name twice. Fortunately, @Neradoc has fixed the problem and submitted a PR to `pyrserial`. Since the code is mostly standalone, it could be pulled out of `pyserial` for now. Or we could fork `pyserial` temporarily. See https://github.com/pyserial/pyserial/pull/566.

We should write a self-contained script that returns the appropriate device for use in scripts like yours. I am also planning to add this capability to the Mu editor.

dhalbert, 2021-03-12:
Fixed by #4384.

closed 2021-03-12 by dhalbert