adafruit/circuitpython #4368: PulseIn in ESP32S2 not working as expected in pulseio.PulseIN

jposada202020, 2021-03-09:
```python
Adafruit CircuitPython 6.1.0-beta.2 on 2020-12-03; FeatherS2 with ESP32S2
```

**Code/REPL**

`
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/examples/dht_simpletest.py
`

**Behavior**

in the Pulseio.PulseIn library https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/7c81737eff2b2fcaf399dfc7afee61d12867ca58/adafruit_dht.py#L82
    For the Metro ESP32S2 if I use hi-sig=True: Only the first measure will be good, as the pulses will start with a low and then high, however, after the first measure all the pulses will start at high, and our flag hi_sig=True will not be longer be valid
    80 pulses: [24, 54, 24, .....
    Temp: 71.6 F / 22.0 C Humidity: 37%
    81 pulses: [54, 24, 54, .....

    For the featherS2, it is the opposite situation so the DHT library will not work.

See https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/53

**Description**

- Error while using Pulseio.PulseIn for the featherS2
- Happens when using Adafruit_DHT library,
- Related with https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/53
- Behavior could be reproduced. Probably not the same Root cause but similar behavior as #4300 

**Additional Info**

When using Pulsio.Pulsein fot the ESP32S2, you will use PulseIn to get values for the DHT11/22 sensors using CircuitPython 

While reading, for the Metro Express ESP32S2 the first pulse read is a Low so the first lecture is a checksum fail, after this initial failure the rest of the first pulse readings are High (in DHT therms). The DHT library is expecting the first pulse as a high and work like a charm. No problems here.
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/7c81737eff2b2fcaf399dfc7afee61d12867ca58/adafruit_dht.py#L82

        binary = 0
        hi_sig = False
        for bit_inx in range(start, stop):
            if hi_sig:

It was noted by @dgriswo that for the featherS2 this behavior was the opposite. For the FeatherS2 the first pulse in the first reading is a High, getting a valid lecture, and after this all the pulses are high, opposite to the Metro ESP32S2. To solve this you could change the DHT library in the code showed above and rewrite the value for hi_sig = True. This solves the issue for the FeatherS2.  However, in doing this now the METRO ESP32S2 will not work.

Investigating the issue I tried to understand why the two board where generating two different things. My train of thought was, maybe if I destroy and create the object every reading, I will get the same lecture (Low Pulse in DHT therms), however, this was not the case, according to my findings, you will have a low pulse or high pulse and you cannot predict this. you will have sometimes a low and sometimes a high, For me it was around 50% each. I am guessing that once the PulseIn object is created, and after the initial reading, the behavior stabilizes.
Removing [this](url) line resolves the issue.

microDev1, 2021-03-11:
> @microDev1 want to take a look at this?

I can but it will be a while before I get to this.
I did ran a test and for me setting `hi_sig=True` worked... I am running:

```python
Adafruit CircuitPython 6.2.0-beta.2 on 2021-03-02; microS2 with ESP32S2
```

microDev1, 2021-03-11:
@microDev1 want to take a look at this?

I can but it will be a while before I get to this.
I did run a test and for me setting hi_sig=True worked... I am running:

Adafruit CircuitPython 6.2.0-beta.2 on 2021-03-02; microS2 with ESP32S2

microDev1, 2021-03-11:
> @microDev1 want to take a look at this?

I can but it will be a while before I get to this.
I did run a test and for me setting `hi_sig=True` worked... I am running:

```python
Adafruit CircuitPython 6.2.0-beta.2 on 2021-03-02; microS2 with ESP32S2
```

jposada202020, 2021-03-12:
@microDev1 Thank you for taking a look at this

mwisslead, 2022-04-27:
I think this is caused by a buffer overflow in PulseIn.c which I think I have a fix for. I was getting checksum errors with a dht22 after the first reading but rock solid after the fix. Should I submit a PR?

closed 2023-02-20 by jposada202020