adafruit/circuitpython #4372: RP2040 Feather Clock Stretch too long error with I2C Singleton

makermelissa, 2021-03-09:
See https://forums.adafruit.com/viewtopic.php?f=60&t=176447&p=859863 for more details.

Using `i2c = board.I2C()` causes a "Clock Stretch too long" error but using
`i2c = busio.I2C(board.SCL, board.SDA)` fixes it.

dhalbert, 2021-03-09:
@makermelissa Are there other controllers that would be similar to the SSD1327, say like the SSD1306? I can order one but would like to test sooner?

makermelissa, 2021-03-09:
It seems to be getting caught on `i2c = board.I2C()`, so as long as something with pull-ups is attached via I2C, you should be able to reproduce the error. I would think the SSD1306 would be fine to test with.

kevinjwalters, 2021-03-10:
Are `board.SCL` and `board.SDA` defined on RP2040 / Pi Pico?

SAK917, 2021-03-12:
I am also getting this error using the Feather RP2040 with an SH1107 128x64 OLED Featherboard:

Adafruit CircuitPython 6.2.0-beta.3 on 2021-03-04; Adafruit Feather RP2040 with rp2040
>import board
>import displayio
>displayio.release_displays()
>i2c = board.I2C()

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TimeoutError: Clock stretch too long

I get the same result using the most recent CP build for the RP2040:
Adafruit CircuitPython 6.2.0-beta.3-125-g0e451a168 on 2021-03-11; Adafruit Feather RP2040 with rp2040

This code works on both a Feather M4 Express and a QT Py Haxpress  when connected to the same display.

PR #4390 (2021-03-12): correct clock stretch timeout for board.I2C()
Fixes #4372.

The clock stretch timeout for `board.I2C()` was given as zero, but the default is 255. This timeout is only used by `bitbangio.I2C()`, and didn't matter up to now for other ports. But the RP2040 port uses bitbangio for zero-byte writes, so it used the zero clock stretch timeout.

Anyone can review. This is simple. Tested.
closed 2021-03-12 by dhalbert