adafruit/circuitpython #4373: Intermittent "Connection Error: Unknown Failure 205" error on Adafruit MagTag when connecting to WiFi

LewsTherinTelamon, 2021-03-09:
**Firmware**

```python
Adafruit CircuitPython 6.2.0-beta.3 on 2021-03-04; Adafruit MagTag
```

**Code/REPL**
Wifi test from guide
```python
import ipaddress
import ssl
import wifi
import socketpool
import adafruit_requests
 
# URLs to fetch from
TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"
JSON_QUOTES_URL = "https://www.adafruit.com/api/quotes.php"
JSON_STARS_URL = "https://api.github.com/repos/adafruit/circuitpython"
 
# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise
 
print("ESP32-S2 WebClient Test")
 
print("My MAC addr:", [hex(i) for i in wifi.radio.mac_address])
 
print("Available WiFi networks:")
for network in wifi.radio.start_scanning_networks():
    print("\t%s\t\tRSSI: %d\tChannel: %d" % (str(network.ssid, "utf-8"),
            network.rssi, network.channel))
wifi.radio.stop_scanning_networks()
 
print("Connecting to %s"%secrets["ssid"])
wifi.radio.connect(secrets["ssid"], secrets["password"])
print("Connected to %s!"%secrets["ssid"])
print("My IP address is", wifi.radio.ipv4_address)
 
ipv4 = ipaddress.ip_address("8.8.4.4")
print("Ping google.com: %f ms" % (wifi.radio.ping(ipv4)*1000))
 
pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())
 
print("Fetching text from", TEXT_URL)
response = requests.get(TEXT_URL)
print("-" * 40)
print(response.text)
print("-" * 40)
 
print("Fetching json from", JSON_QUOTES_URL)
response = requests.get(JSON_QUOTES_URL)
print("-" * 40)
print(response.json())
print("-" * 40)
 
print()
 
print("Fetching and parsing json from", JSON_STARS_URL)
response = requests.get(JSON_STARS_URL)
print("-" * 40)
print("CircuitPython GitHub Stars", response.json()["stargazers_count"])
print("-" * 40)
 
print("done")
```

**Behavior**

Intermittent failure to connect to wifi AP. Behavior began _after_ connecting to a second network and then trying to connect to the first network again. Confirmed "secrets.py" is correct as Wifi sometimes connects.  Connection attempt seems to take 2-3 times longer than standard when the error occurs and then the error below appears in the REPL.

```python
Traceback (most recent call last):
  File "code.py", line 30, in <module>
ConnectionError: Unknown Failure 205
```

**Additional Info**

Not sure whether resetting my router fixed the issue temporarily or it cleared up on its own briefly.  Error occurs when using the simpletest above and when using
```magtag.network.get_local_time()```

anecdata, 2021-03-10:
205 unfortunately is not a very helpful [error code](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/wifi.html#wi-fi-reason-code): "Espressif-specific Wi-Fi reason-code: the connection to the AP has failed."

Does the intermittent error persist after a hard reset of the MagTag?

LewsTherinTelamon, 2021-03-10:
Hard reset being either unplug/replug or a pressing of the reset button? Persists after both.

mzseast, 2021-03-12:
I reported the same issue in the adafruit forums ("Magtag - ConnectionError: Unknown failure 205") a few weeks ago.

In my case, when running the magtag wifi test program, or any other sample programs (from the adafruit learn circuitpython example), it can see several access points in my area. However when attempting to connect to my local wifi network, all I get is the 205 error after a 10-15 second delay. Contents of "secrets" file are correct. I have NEVER been able to connect to my network.

I've tried all 6.2.x versions of magtab circuitpython - no difference.

I have several ESP32 devkit boards (programmed & running test programs from arduino), and they can all connect to my network without problems - problem seems to be ESP32-S2 (and/or circuitpython) specific????

Note that I'm new to circuit/micro-python...

anecdata, 2021-03-12:
If you haven't already, try `connect` in a loop with try/except. After connecting the first time, it's a little more resilient to re-establishing the connection if it fails.

Again, if you haven't already, try to connect to an alternate Access Point, and see if proximity to the AP makes a difference (a scan in CircuitPython will show RSSI for each AP; unfortunately a connection is probably needed to see the RSSI of the device from the AP perspective).

mzseast, 2021-03-13:
I've spent some time experimenting with location & orientation of the magtag and have FINALLY got it to connect to my wifi router.
It looks like it could possibly be signal corruption due to parts layout on the magtag pcb causing interference to the wifi signal.
Results of my testing in the adafruit forums ("Magtag - ConnectionError: Unknown failure 205").

HolgBogarth, 2021-04-23:
I am experiencing this same issue using circuitpython.  The further away from my AP I get, the more 205 errors I seem to receive.  The other ESP32 boards I've own do not have this issue.

anecdata, 2021-04-23:
If there is more than one AP broadcasting the same SSID, you can select a specific AP to connect to by its BSSID, and / or, you can select a specific channel to use to avoid channel congestion or other sources of interference:
https://circuitpython.readthedocs.io/en/latest/shared-bindings/wifi/index.html#wifi.Radio.connect

dhalbert, 2021-04-23:
Forum discussion is here: https://forums.adafruit.com/viewtopic.php?f=19&t=175553

tavdog, 2021-05-02:
I'm seeing this problem as well.  It was working fine and then one day it won't connect.  The AP is shown on wifi scan but it won't connect to it. Sometimes it says "No network with that ssid" and other times it's "Unknown failure 205".  If I bring the magtag within a few feet of the AP it will connect.

vrtisworks, 2021-12-20:
Thought I would pass along my experience in case it provides a clue.. I have a Magtag running 

Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit MagTag with ESP32S2
Board ID:adafruit_magtag_2.9_grayscale

It was running the Weather code from the Adafruit examples fine for a few weeks.  And then one day it started with the Unknown Failure 205 error.   During the course of testing, I deleted everything and reloaded all the libraries and code.  Still have the problem.

I have tried both and ASUS RT-AC56U and a Trendnet TEW-810DR, located in different parts of the house. The Trendnet only has the Magtag connected to it.  Results were "about" the same,  It would work sometimes, and not work sometimes.

I added this code:
# - Set the clock (and make sure we can connect to the AP
print("Connecting and getting time")
need_connect=True
connect_tries=01
while need_connect:
    try:
        magtag.network.get_local_time()
        need_connect=False
    except:
        connect_tries+=1
        print("Connect failed: Tries:{}".format(connect_tries))
        time.sleep(10)

This will at least keep trying to connect to the AP every 10 seconds.  Usually it will make the connection in less than 3 tries.  But I have seen it go as high as 10 times before it was successful.

rpavlik, 2022-04-03:
Just had all the ESP32-S2 devices from Adafruit in my home start acting like my router doesn't exist, very strange.  ("No network with that name" and/or error 205). I suspect it's router related, since they're different ages and usage profiles, but I didn't change anything router related... Maybe it's the new neighbors.

ericzundel, 2022-10-16:
I was searching around for what to do for Unknown Error 205 on discord and kept finding this issue.  Here's what I did to work around it on my mesh network (Google Nest WiFi):
1) Use the app WiFi Analyzer on my Android phone to scan and create a dump.
2) Look through the generated .XML for the BSSIDs and signal strength of all my access points.  I chose the one with the highest (closest to zero) level in decibels.
3) Add the BSSID as follows to the radio connect command:

```
            wifi.radio.connect(os.getenv('WIFI_SSID'), os.getenv('WIFI_PASSWORD'), bssid=bytes([0x90,0x9D,0x7D,0x99,0x99,0x99]))
```
No, that really isn't by BSSID but you get the idea.  Now, it seems to connect every time. Which to me begs the question, "Can this priority be automated?"  

Note, I can also force the error to happen by specifying the BSSID of an AP that is very weak.

anecdata, 2022-10-16:
Espressif [docs]( https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s2/api-guides/wifi.html#esp32-s2-wi-fi-station-connecting-when-multiple-aps-are-found) seem to imply that when there is more than one AP with the same SSID, it will [re]connect to the best one.

rpavlik, 2022-10-18:
PicoW isn't espressif, so the preceding comment is probably misplaced.

anecdata, 2022-10-18:
Correct, not relevant to `raspberrypi, please ignore.

grbd, 2022-12-28:
As a suggestion if there's an option in the wifi router for "Channel bandwidth"
Try setting it to "40Mhz" instead of "20/40Mhz"

This seemed to fix problems I was having when connecting to the lower channel numbers of 7 and below

dhalbert, 2025-01-31:
I'm going to close this since there seem to be multiple causes for this and the mentioned versions are quite old. New reports of similar behavior are welcome.

closed 2025-01-31 by dhalbert