adafruit/circuitpython #4377: Feather RP2040 CIRCUITPY folder disappears after reset

caternuson, 2021-03-10:
Starting new issue as requested from here:
https://github.com/adafruit/circuitpython/issues/4349#issuecomment-795742928

Pretty simple way to re-create:

- Hold `BOOTSEL`, press `RESET`, get `RPI-RP2` folder
- Copy firmware UF2* to `RPI-RP2` folder
- Board self resets when copy is complete, `CIRCUITPY` folder shows up, everything works as expected - can access REPL, etc.
- Can create a `code.py` and soft reset via `<CTRL><D>` at REPL and it works as many times as you want.
- Press `RESET` button, the `CIRCUITPY` folder goes away and never comes back. Any `code.py` also does not run.

*I originally used the 6.2.0-beta.3 firmware, but have also tried various Absolute Newest builds. All behave the same.

Here's some dmesg traffic FWIW.

After holding `BOOTSEL` and pressing `RESET`:
```
[Wed Mar 10 10:06:14 2021] usb 2-1.6: new full-speed USB device number 57 using ehci-pci
[Wed Mar 10 10:06:14 2021] usb 2-1.6: New USB device found, idVendor=2e8a, idProduct=0003, bcdDevice= 1.00
[Wed Mar 10 10:06:14 2021] usb 2-1.6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Mar 10 10:06:14 2021] usb 2-1.6: Product: RP2 Boot
[Wed Mar 10 10:06:14 2021] usb 2-1.6: Manufacturer: Raspberry Pi
[Wed Mar 10 10:06:14 2021] usb 2-1.6: SerialNumber: E0C912952D54
[Wed Mar 10 10:06:14 2021] usb-storage 2-1.6:1.0: USB Mass Storage device detected
[Wed Mar 10 10:06:14 2021] scsi host7: usb-storage 2-1.6:1.0
[Wed Mar 10 10:06:15 2021] scsi 7:0:0:0: Direct-Access     RPI      RP2              2    PQ: 0 ANSI: 2
[Wed Mar 10 10:06:15 2021] sd 7:0:0:0: Attached scsi generic sg3 type 0
[Wed Mar 10 10:06:15 2021] sd 7:0:0:0: [sdc] 262144 512-byte logical blocks: (134 MB/128 MiB)
[Wed Mar 10 10:06:15 2021] sd 7:0:0:0: [sdc] Write Protect is off
[Wed Mar 10 10:06:15 2021] sd 7:0:0:0: [sdc] Mode Sense: 03 00 00 00
[Wed Mar 10 10:06:15 2021] sd 7:0:0:0: [sdc] No Caching mode page found
[Wed Mar 10 10:06:15 2021] sd 7:0:0:0: [sdc] Assuming drive cache: write through
[Wed Mar 10 10:06:15 2021]  sdc: sdc1
[Wed Mar 10 10:06:15 2021] sd 7:0:0:0: [sdc] Attached SCSI removable disk
```

After copying firmware UF2 to `RPI-RP2` folder:
```
[Wed Mar 10 10:07:03 2021] usb 2-1.6: USB disconnect, device number 57
[Wed Mar 10 10:07:03 2021] blk_update_request: I/O error, dev sdc, sector 260 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[Wed Mar 10 10:07:03 2021] Buffer I/O error on dev sdc1, logical block 259, lost async page write
[Wed Mar 10 10:07:03 2021] FAT-fs (sdc1): unable to read boot sector to mark fs as dirty
[Wed Mar 10 10:07:04 2021] usb 2-1.6: new full-speed USB device number 58 using ehci-pci
[Wed Mar 10 10:07:04 2021] usb 2-1.6: New USB device found, idVendor=239a, idProduct=80f2, bcdDevice= 1.00
[Wed Mar 10 10:07:04 2021] usb 2-1.6: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[Wed Mar 10 10:07:04 2021] usb 2-1.6: Product: Feather RP2040
[Wed Mar 10 10:07:04 2021] usb 2-1.6: Manufacturer: Adafruit
[Wed Mar 10 10:07:04 2021] usb 2-1.6: SerialNumber: 545473038350C913
[Wed Mar 10 10:07:04 2021] cdc_acm 2-1.6:1.0: ttyACM0: USB ACM device
[Wed Mar 10 10:07:04 2021] cdc_acm 2-1.6:1.2: ttyACM1: USB ACM device
[Wed Mar 10 10:07:04 2021] usb-storage 2-1.6:1.4: USB Mass Storage device detected
[Wed Mar 10 10:07:04 2021] scsi host7: usb-storage 2-1.6:1.4
[Wed Mar 10 10:07:04 2021] input: Adafruit Feather RP2040 Consumer Control as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.5/0003:239A:80F2.0082/input/input193
[Wed Mar 10 10:07:04 2021] input: Adafruit Feather RP2040 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.5/0003:239A:80F2.0082/input/input194
[Wed Mar 10 10:07:04 2021] input: Adafruit Feather RP2040 Keyboard as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.5/0003:239A:80F2.0082/input/input195
[Wed Mar 10 10:07:04 2021] input: Adafruit Feather RP2040 Mouse as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.5/0003:239A:80F2.0082/input/input196
[Wed Mar 10 10:07:04 2021] hid-generic 0003:239A:80F2.0082: input,hidraw7: USB HID v1.11 Gamepad [Adafruit Feather RP2040] on usb-0000:00:1d.0-1.6/input5
[Wed Mar 10 10:07:05 2021] scsi host7: scsi scan: INQUIRY result too short (5), using 36
[Wed Mar 10 10:07:05 2021] scsi 7:0:0:0: Direct-Access     Adafruit Feather RP2040   1.0  PQ: 0 ANSI: 2
[Wed Mar 10 10:07:05 2021] sd 7:0:0:0: Attached scsi generic sg3 type 0
[Wed Mar 10 10:07:05 2021] sd 7:0:0:0: [sdc] 14337 512-byte logical blocks: (7.34 MB/7.00 MiB)
[Wed Mar 10 10:07:05 2021] sd 7:0:0:0: [sdc] Write Protect is off
[Wed Mar 10 10:07:05 2021] sd 7:0:0:0: [sdc] Mode Sense: 03 00 00 00
[Wed Mar 10 10:07:05 2021] sd 7:0:0:0: [sdc] No Caching mode page found
[Wed Mar 10 10:07:05 2021] sd 7:0:0:0: [sdc] Assuming drive cache: write through
[Wed Mar 10 10:07:05 2021]  sdc: sdc1
[Wed Mar 10 10:07:05 2021] sd 7:0:0:0: [sdc] Attached SCSI removable disk
[Wed Mar 10 10:07:06 2021] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```

At this point everything is as expected. I've got a `CIRCUITPY` folder. The LED blink code.py is running. I can connect to serial and drop into REPL:
```python
Traceback (most recent call last):
  File "code.py", line 10, in <module>
KeyboardInterrupt: 

Code done running.
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.3-106-gbe9e045ee on 2021-03-10; Adafruit Feather RP2040 with rp2040
>>> print("hello")
hello
>>> 
```

But then, press the `RESET` button:
```
[Wed Mar 10 10:17:11 2021] usb 2-1.6: USB disconnect, device number 58
```
and that's it. No more messages, `CIRCUITPY` folder does not show up, and the code.py LED blinker is not running.

bhilimon, 2021-03-10:
Also having the same issue. https://forums.adafruit.com/viewtopic.php?f=57&t=176407&p=859903

jedgarpark, 2021-03-11:
I'm having this issue as well, and noticed the same behavior if you pull power.

FoamyGuy, 2021-03-11:
I'm seeing the same thing on my two Feather RP2040s. Plugged in via USB Hub vs. direct into PC did not make a difference.

I see the same behavior mentioned when pressing reset button, or unplug / replugging power.

tannewt, 2021-03-11:
Do any of you have the same issue on Pico with the same version of CP?

FoamyGuy, 2021-03-11:
I do not have the same issue on a Pico. I tried with the newest UF2 from S3 for Pico and Feather RP2040. The Pico seems to behave normally. I can unplug and replug it and `CIRCUITPY` continues to come back normally.

bhilimon, 2021-03-11:
> 
> 
> Do any of you have the same issue on Pico with the same version of CP?

Sorry, I don't have a Pico or any other RP2040 board to test with.

caternuson, 2021-03-11:
Pico works as expected.

Tested with `6.2.0-beta.3` and `15e97f1, which was "latest" at time of testing. For Pico, tested by unplugging/plugging USB cable. `CIRCUITPY` folder comes back and `code.py` runs each time.

KingerNorth, 2021-03-11:
I just tested based on this issue and I found that it the Feather RP2040 from the latest production run (I got 2 of them yesterday delivered by DHL to me in Canada) worked as they should. I followed the steps as laid out by caternuson at the start of the post. 

1) Attached the Feather RP204 to my Windows 10 PC
2) Held BOOTSEL and pressed RESET, go the RPI-RP2 folder
3) Dragged the latest UF2 file from circuitpython.org into the folder
4) The CIRCUITPY folder appeared
5) Opened MU and edited the file for a basic LED blink
6) Save and it ran fine
7) Interrupted with <CTRL><D>
8) Typed in code for a basic print at the >>> prompt
9) Pressed RESET and restarted the Feather RP2040 and the CIRCUITPY folder appeared and the code.py was running

KingerNorth, 2021-03-11:
I did further test and I EJECTED the CIRCUITPY USB drive from my PC in the Windows File Explorer window, I then unplugged the USB cable from the Feather RP2040. I waited a few seconds then plugged the USB cable back in and the CIRCUITPY drive appeared and the code.py restarted properly.

todbot, 2021-03-11:
Just as another datapoint: the two RP2040 Feathers I received yesterday exhibit this problem.  The steps I performed:

- Plug in device. It appears dead: no lights, no USB activity to host. Device draws 7 mA
- BOOTSEL reset to RPI-RP2 bootloader. Device draws 21 mA
- Install CircuitPython UF2. Latest tried was `adafruit-circuitpython-adafruit_feather_rp2040-en_US-20210311-3b948b2.uf2`)
- Programming successful. Green neopixel, CIRCUITPY drive, both CDC UARTs, everything else all present. Can install modules and edit `code.py`. Device draws 26 mA
- Power cycle via plug/unplug 
- Device appears dead. No USB, no LED. Device draws 7 mA
- Repeat firmware installation :)

dhalbert, 2021-03-11:
Could you all include which OS's you have this problem on, and if you try it on another OS, does the problem change or go away?

bhilimon, 2021-03-11:
> 
> 
> Could you all include which OS's you have this problem on, and if you try it on another OS, does the problem change or go away?

I've tried a few Windows 10 machines an also Linux (my regular Pi 4 w/normal Pi OS). I got to the same point on all machines I've tried installing/copying from.

todbot, 2021-03-11:
> Could you all include which OS's you have this problem on, and if you try it on another OS, does the problem change or go away?

The issue occurs for me on both MacOS 11.2.2 Big Sur x86 and Windows 10 Pro x86

FoamyGuy, 2021-03-11:
Linux:
[image]

and Windows 7. 

Same behavior for me on both.

I have noticed that `code.py` does seem to persist. I saved basic blinky to it, and that same script was still there and ran just fine after subsequent re-pasting of the UF2.

caternuson, 2021-03-11:
Ubuntu 20.04.2 LTS and Windows 10

KingerNorth, 2021-03-11:
> Could you all include which OS's you have this problem on, and if you try it on another OS, does the problem change or go away?

I was using my Windows 10 PC. I'll try it later on my Macbook Pro.

jerryneedell, 2021-03-12:
FYI -- I received a feather_rp2040 today and  I am(so far) unable to reproduce this issue on a Raspberry Pi 400.
CP is the current tip of main - g8d3217cbf

caternuson, 2021-03-12:
I just tried this with the 2nd Feather RP2040 I ordered and it is ***not*** doing this. It behaves normally. Both Feathers are from the same work order. Neither have headers soldered. They are being used straight from bag.

closed 2021-03-12 by jepler

tannewt, 2021-03-12:
Hi folks, please try this with builds after aec03a4. I found we were clocking the flash faster than it's rated for. I think this caused it to sometimes work if the clocks started up without overshooting their speed. If it did, it'd try to load code off the flash and it'd be corrupt. The corrupt code would cause a hard fault and things would stop. #4392 lowers the speed to 40mhz-ish which should be much more reliable even if the starting clocks overshoot.

todbot, 2021-03-12:
Verified that this change fixes the problem on the two Feather 2040s I have.

tannewt, 2021-03-12:
FYI @aallan

bhilimon, 2021-03-12:
Seems to have fixed it. Both of my boards are running normally now. Thanks!

caternuson, 2021-03-13:
I'm late to this so just reporting it works here also. I used `b413535` FWIW. Both boards work as expected.

Thanks for the fix!

mathias, 2021-08-22:
Is it possible that this problem could be back, on a Macropad running 7.0.0-alpha.6 ? 

I’m seeing a very similar problem, except I can’t get the board back by reset — I have to use the nuke uf2 to clear it and start over with flashing. I described what I’m seeing here: https://forums.adafruit.com/viewtopic.php?f=60&t=182523&p=886785#p886785 before I found this Issue.

dhalbert, 2021-08-23:
@mathias I answered in the forums.

dhalbert, 2025-02-24:
@AnkurG99 Please open a thread in https://forums.adafruit.com or post in our discord https://adafru.it/discord. Please describe what you mean by "it removes everything" where you bring that up.

PR #4392 (2021-03-12): Lower default flash speed.
/ 6 leads to ~40mhz. GD25C 2M and 4M have a max 0x03 read speed of 60mhz.
If the divisor is / 4 then the speed is just over 60mhz.

Fixes #4377
commit aec03a40 (2021-03-12): Lower default flash speed.

/ 6 leads to ~40mhz. 2M and 4M have a max 0x03 read speed of 60mhz.
If the divisor is / 4 then the speed is just over 60mhz.

Fixes #4377