adafruit/circuitpython #4382: Circuitpython canio module on STM32F407 and STM32F412 boards

Shubhankargo, 2021-03-11:
Hi

I'm new to Circuitpython. I am trying to build a CAN bus based application on circuit python. Can anyone tell me if the canio module is available on the on STM32F407 and STM32F412 discovery boards? Or will I have to purchase a STM32F405 feather board from Adafruit?

Thanks.

hierophect, 2021-03-11:
@Shubhankargo, both of those chips have CAN peripherals and _should_ work with our current module, but we don't seem to have them enabled. @jepler can we turn this on for all F4 chips, or at least for the F407? The F407 should definitely work, the F412 should _probably_ work, but it could have some weird register differences as an "access" level F4 chip.

jepler, 2021-03-11:
You can check the list of supported modules on the circuitpython.org downloads page for the specific board, such as https://circuitpython.org/board/stm32f4_discovery/.  The canio module is not available at this time for "STM32F407 Discovery kit" or "STM32F412 Discovery kit".  We would be happy to accept a pull request to enable it, if it is supported by the underlying hardware.

closed 2021-03-11 by jepler

Shubhankargo, 2021-03-12:
Thank you for your response guys.

The download pages of neither of those boards (chips) show the 'canio' module as enabled. Are you saying that I can add the bin file for the STM32F07 and run the module like normal?

Sorry I'm a bit of a noob and not too well versed with pull requests.

microDev1, 2021-03-12:
@Shubhankargo The module list will be update once a new release is published.
For now you can test the latest build by grabbing it from [S3](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/).

hierophect, 2021-03-12:
@Shubhankargo note that I did not add the F412, it's not a direct superset of the F405 like the F407 is and I don't have the cycles to test it right now.

Shubhankargo, 2021-03-21:
Thanks for the help guys.
So since the STM32F07 is a strict super set of the STM32F05, should load the circuitpython .bin file of the STM32F05 Feather directly into the F07 Discovery board, or just copy the canio module into the lib after loading the STM32F07 .bin file?

hierophect, 2021-03-21:
@Shubhankargo I'm not sure I understand what you mean. You should not use the Feather binary since it will have totally different pin definitions and oscillator settings. CanIO is a native module and does not have a .mpy file in the libraries folder of the filesystem, so I'm not sure what you mean there either.

If you want to use CAN on the F407 Discovery now that support has been merged, you should visit the [s3](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/) link and click through the `stm32f4_discovery/` to get the latest build in your preferred language, or the [build artifacts for the PR](https://github.com/adafruit/circuitpython/pull/4385/checks) for a zip file of builds from that specific commit for the F4 discovery.

Shubhankargo, 2022-02-08:
Apologies for not getting back earlier on this. The solution worked perfectly. Thank you @hierophect .

PR #4385 (2021-03-11): STM32: add all F405 modules to F407
The F407 is a strict superset of the F405 silicon (it just adds a camera and ethernet to the chip) and should support all the same modules (except RGBMatrix since that requires Protomatter changes). Basically just needed a new copy-paste of the F405 peripherals folder to the F407. In practice this adds SDIO, FrameBufferIO, and CANIO to the STM32F4 Discovery board.

Resolves #4382