adafruit/circuitpython #4394: socketpool.Socket is missing getaddrinfo

johncblacker, 2021-03-13:
Documentation includes:
socket.getaddrinfo(host: str, port: int) → Tuple[int, int, int, str, str]
Gets the address information for a hostname and port

Returns the appropriate family, socket type, socket protocol and address information to call socket.socket() and socket.connect() with, as a tuple.
But it has not been implemented on the esp32-s2 port.  Just wanted to insert this as a reminder.

anecdata, 2021-03-13:
See[ RTD](https://circuitpython.readthedocs.io/en/latest/shared-bindings/socketpool/index.html#socketpool.getaddrinfo):
```
>>> import wifi
>>> import socketpool
>>> from secrets import secrets
>>> 
>>> wifi.radio.connect(secrets['ssid'], secrets['password'])
>>> 
>>> pool = socketpool.SocketPool(wifi.radio)
>>> pool.getaddrinfo("adafruit.com", 443)
[(0, 0, 0, '', ('104.20.38.240', 443))]
>>> 
```

PaulskPt, 2022-02-08:
My MAGTAG date and time app was running OK for months. Recently the script started crashing.
In this moment I am getting OSError -2 on a MAGTAG.
The MAGTAG is flashed with:
```Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit MagTag with ESP32S2```
```Board ID:adafruit_magtag_2.9_grayscale```

Script crashes here with OSError -2:

In file adafruit_requests.py, line 536 
       ```addr_info = self._socket_pool.getaddrinfo(
            host, port, 0, self._socket_pool.SOCK_STREAM
        )[0]```

whereas:
self._socket_pool = ```<SocketPool>```

where also:
ping google.com (8.8.8.8) timed-out
ping google.com (8.8.4.4) timed-out
ping router in LAN 9.0 mSec
WiFi connected to LAN access-point
IP address received: 192.168.1.101

proto = 'https:'
host = 'io.adafruit.com'
port = 443
timeout = 60
addr_info = 0

This is the boot info the ESP32s2 is sending via TX:

[image]

tannewt, 2022-02-08:
@PaulskPt please open a new issue. I'm not sure your issue is related to this one.

PaulskPt, 2022-02-08:
OK

PaulskPt, 2022-02-09:
The script ran already uninterrupted from yesterday 19:46 local until this moment in the next morning.