adafruit/circuitpython #4398: Is there a premade digitizer module

ArielWolle, 2021-03-13:
Title

Or do I have to build one from the ground up, if so how is the report structed (buffer size and order of (x,y,ifPressed,etc))

dhalbert, 2021-03-13:
The information you need is in https://github.com/adafruit/circuitpython/blob/550eab2ddda780adb939693a51c8c07c4a540c36/tools/hid_report_descriptors.py#L192. The report size can be derived from there but is also listed here: https://github.com/adafruit/circuitpython/blob/550eab2ddda780adb939693a51c8c07c4a540c36/tools/hid_report_descriptors.py#L28.

ArielWolle, 2021-03-13:
Any good documentation online on these reports? Because to me it looks like something crazy. Also thank you so much your so fast with these responses :)

dhalbert, 2021-03-13:
https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/ is a good intro.

This is really more a support question, and would be better asked in https://forums.adafruit.com and/or our discord https://adafru.it/discord. So I'm closing this for now. Feel free to ask more about descriptors in either of those places.

closed 2021-03-13 by dhalbert