adafruit/circuitpython #4400: Circuitpython code.py not work anymore

important0, 2021-03-13:
My OS Arch linux
Item ItsyBitsy M0 Express

## What do I did ?
https://circuitpython.org/board/itsybitsy_m0_express/
1) Update > adafruit-circuitpython-itsybitsy_m0_express-en_US-6.2.0-beta.3.uf2
2) Update > update-bootloader-itsybitsy_m0-v3.13.0.uf2

↓

Inside ItsyBitsy M0 Express

https://learn.adafruit.com/welcome-to-circuitpython/creating-and-editing-code

code.py
```
import board
import digitalio
import time
 
led = digitalio.DigitalInOut(board.D13)
led.direction = digitalio.Direction.OUTPUT
 
while True:
    led.value = True
    time.sleep(0.5)
    led.value = False
    time.sleep(0.5)
```

↓
What happen ?
there is notting to work at all

↓

## Troubleshooting
https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting

"MY" CircuitPython RGB Status Light

Green → Orange?

↓

Then 2times flash blue&green

## what do I did ?
Hardreset circuitpython
https://github.com/tuupola/circuitpython/blob/master/docs/wipy/tutorial/reset.rst

```
>>> import os
>>> os.mkfs('/flash')
```
Not working

## what do I want ?

I want
・factry reset my ItsyBitsy M0 Express
・work for code.py

thank you for reading until end.

anecdata, 2021-03-13:
There should be an exception printed to the serial monitor. What is displayed?
https://learn.adafruit.com/welcome-to-circuitpython/kattni-connecting-to-the-serial-console

ajs256, 2021-03-13:
> Hardreset circuitpython
https://github.com/tuupola/circuitpython/blob/master/docs/wipy/tutorial/reset.rst

That link is from an out-of-date fork. https://circuitpython.readthedocs.io will always be the latest.

ajs256, 2021-03-15:
To erase the CircuitPython filesystem, use these commands at the REPL:
```py
>>> import storage
>>> storage.erase_filesystem()
```

dhalbert, 2021-03-31:
Closing for now. Best places to ask for support like this are https://adafru.it/discord and https://forums.adafruit.com.

closed 2021-03-31 by dhalbert