adafruit/circuitpython #4404: Firmware configuration file

bitboy85, 2021-03-14:
To customize the board i think a firmware configuration file could come handy.
The file should stay in the botsel part of the storage and should allow some basic configuration of the device.
The configuration should be applied on each firmware update so the configuration will always be the same even if the defaults in the firmware itself changes.

It could look like this
`
[GENERAL]
FileSystem="FS Label"

[USB]
USB_VID = 0x239A
USB_PID = 0x804C
USB_PRODUCT = "PyRuler"
USB_MANUFACTURER = "Adafruit Industries LLC"

[ENDPOINTS]
0=CDC
1=Storage
2=Keyboard
3=BootKeyboard
4=Gamepad
5=MIDI
`

Reference:
https://github.com/adafruit/circuitpython/issues/4191
https://github.com/adafruit/circuitpython/issues/1136
https://learn.adafruit.com/welcome-to-circuitpython/the-circuitpy-drive
https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython/customizing-the-board-files

deshipu, 2021-03-14:
Maybe it would be best to keep this information in the bootloader? https://github.com/Microsoft/uf2/blob/master/cf2.md

jepler, 2021-03-15:
See also #1015 about making USB descriptors configurable. I think we favored the idea of doing it in Python code from boot.py over having a "configuration file" in another language.

Note that the filesystem label is set just once when the storage is initially formatted (or re-formatted via storage.erase_filesystem()) rather than each time the device is plugged in.

tannewt, 2021-03-16:
Why do you want this? `boot.py` is meant to store settings too.

bitboy85, 2021-03-16:
If it is possible to edit those settings within boot.py i'm fine with it. I thought, the configuration needs to be available at the time circuit python is installed/updated.
The idea behind the bot-sel storge was that this configuration cannot be accidentally deleted and it would be easier to deploy because you just need to copy the configuration an the firmwarefile to the same directory.

tannewt, 2021-03-16:
The UF2 file is copied to a fake drive so no other files can be copied with it. `boot.py` runs before USB starts, so it can change what the USB device is configured as.

s-ol, 2022-04-05:
Configuring this in `boot.py` definitely makes sense. As far as I can tell, what is missing is just a python API for setting/overriding `USB_VID`, `USB_PID`, `USB_PRODUCT`, `USB_MANUFACTURER`; probably in the `supervisor` module.

It's interesting that `usb_build_device_descriptor` already has parameters for setting `USB_VID` and `USB_PID` at runtime, but it is only ever called once (and with static arguments).

These values could be non-const variables that can be updated (in RAM only) with something like e.g.:
```
import supervisor
supervisor.set_usb_device_descriptor(
  vid: Optional[int] = 0xdead,
  pid: Optional[int] = 0xbeef,
  manufacturer: Optional[str] = "ACME Inc.",
  product: Optional[str] = "Foobar",
)
```

PR #6247 (2022-09-21): Add supervisor.set_usb_identification(manufacturer, product, vid, pid) function
This adds `supervisor.set_usb_identification(manufacturer: Optional[str] = None, product: Optional[str] = None, vid: int = -1, pid: int = -1)`

This function can be called to override the build-type settings for USB_VID, USB_PID, USB_MANUFACTURER_NAME, USB_PRODUCT_NAME for the next boot process from `boot.py`.

Closes #4404
closed 2022-09-21 by dhalbert