adafruit/circuitpython #4409: alarm for RP2040

MisterD66, 2021-03-15:
I would love to see some kind of powersaving / sleep functions for the RP2040 / Raspberry Pi Pico.

I think that is one of the key features missing for CircuitPython on the new Pi Pico board are some Sleep funktions!
For me the reason choosing a Pi Pico over a ESP32 is the lower powerconsumption for projects needing no WiFi!

mperino, 2021-04-14:
Any update on this?  @tannewt and @dhalbert have a lot of discussion in a lot of different issues, so apologies if I have missed the solution.  I see mention of alarm being in 6.1 Beta, but I don't yet see it in 6.2 released:
The alarm module doesn't seem to be in the base CircuitPython 6.2, nor do I find a standalone lib..
code.py output:
Traceback (most recent call last):
  File "code.py", line 5, in <module>
ImportError: no module named 'alarm'

I've also checked  https://learn.adafruit.com/deep-sleep-with-circuitpython/alarms-and-sleep
Which also seems to indicate that it was in 6.1 beta.. 
Thanks in Advance.

ladyada, 2021-04-14:
there is no support yet for it on the RP2040, only ESP32-S2 at this time

mperino, 2021-04-14:
Thank you @ladyada,   Any chance of getting it higher on the priority list.  We are trying to use Pico + Sensors + Adafruit I/O for an Innovation Project for the FIRST Innovation Challenge...

ladyada, 2021-04-14:
if someone can submit the PR for support, that will be the best thing to speed it up, we have no ETA on when the team would get to it otherwise!

tannewt, 2021-04-14:
@hierophect has started looking at lowering power on the RP2040 I believe.

mperino, 2021-04-15:
https://github.com/micropython/micropython/pull/7122
PR 7122 created.  Thank you all!

dhalbert, 2021-04-15:
> [micropython#7122](https://github.com/micropython/micropython/pull/7122)
> PR 7122 created. Thank you all!

A pull request contains code that you want to add or change. A PR against `micropython` is not want you want to do. Please close that PR.

dhalbert, 2021-04-15:
The ESP32-S2 has the WiFi capability you need for adafruit.io, and can also use sensors. Is there some strong reason to choose the Pico over some other boards?

MisterD66, 2021-04-15:
in my opinion there is only one reason to chose the Pico over an ESP - the lower powerconsumption for projects that need no wifi or bluetooth

[image]

MisterD66, 2021-04-15:
Also i think you are kind of advertising bords wrong!

instead of the list of supported modules there should be a list of "missing features".

for a beginner it is not easy to know what kind of modules you would need for your project and choose the correct board!

i jumped on the RP2040 hype right at the beginning and also used CP for the first time. I chose a small GSM Datalogger as project to learn CP. It was written nowhere that the implementation was not completetd or beta...

first i2c was not working yet for the display
then UART was not working for the gsm modem
then there was no support for single wire protokolls like the DHT22

all of those are fixed now but i had to stop my projekt for weeks inbetween

now i am waiting for lowpowermodes to be implemented...

For comparison: Arduino support was added just some days ago - but it is feature complete with the first release.

I loved the fast developement cycle with CP but i am considering turning my back on it and returning to Arduino...

dhalbert, 2021-04-15:
@MisterD66 GitHub issues are not the place to discuss this kind of thing. I will say that there are some misunderstandings that can be addressed. Let's redirect this discussion to discord, https://adafru.it/discord, or the forums, https://forums.adafruit.com/viewforum.php?f=60. The original issue stands, but I am limiting further discussion.

dhalbert, 2021-04-15:
@mperino We can discuss your requirements further in discord or the forums.

hierophect, 2021-04-15:
I'm currently developing the full Alarm module for the RP2040 and expect to have a PR in soon. It is based on the recent updates I've done to the internal API in #4606, so it probably won't make it in before those changes do.

ladyada, 2021-04-15:
> The request is to add alarm and deep sleep to CircuitPython for RPI2040 aka Pico.
> Creating this FR here (not a PR) based on the feedback from @ladyada and @dhalbert .
> 
> We intend to use RPI2040 for a FIRST innovation project (Go RoBucs Team 4160!) where power consumption is a main factor, as is lowest cost. There already is support for alarm and deep sleep on ESP32 in 6.x.b but those are more expensive, and draw more power.
> 
> The current support on ESP32 with 6.1 Beta is illustrated here:
> https://learn.adafruit.com/deep-sleep-with-circuitpython/alarms-and-sleep
> and
> https://circuitpython.readthedocs.io/en/latest/shared-bindings/alarm/index.html
> 
> To give a "Big Picture" of what we are trying to do:
> 
> The functionality in alarm would work well for us, since our project is based on the following Proof of Concept (which needs a LOT of refinement):
> https://github.com/mperino/GardenSensor
> 
> We intend to explore LORAWAN instead of WiFi as well, but the code:
> https://github.com/mperino/GardenSensor/blob/main/Pico_Garden.py
> Is the code that the kids are learning to greatly improve power management on.
> 
> The students are also designing an "all in one" board to integrate Airlift or LORAWAN radio, a DHT sensor, Stemma Capacitive, Solar Panel (measure light level), and battery. With a socket to mount a Pico.

hierophect, 2021-07-30:
Resolved by #4816, whoops

closed 2021-07-30 by hierophect